bblanchon/ArduinoJson #1631: Advises/Solutions needed for solving ArduinoJson MsgPack communication issue

HappySamuel, 2021-08-17:
Hi

I have been using ArduinoJson since version 5, and i like it. Now i want to migrate into ArduinoJson 6 and use it msgpack, as the total bytes of msgpack is smaller than JsonString of the same Json Object. However i faced some issues when using it, and i have been clueless for days on debugging it.

I am using ArduinoJson's msgpack serialization and deserialization on bluepill board. It can run normally at the beginning, the encoding and decoding of msgpack into json works fine, as well as msgs exchange between pc and bluepill board works fine too. I can std::cout those msgs with no issue. 

Sample msgs communication is as below:
[image]

However, after running some cycle, the board will hang, even though it only have read, decode, encode and write. The way i identify the board whether hang or not, is by blinking in-built LED (10Hz). If the LED become either solid OFF or solid ON, means the board hang. At this moment of time, if the board is being plugged (USB) into Windows PC, it will show USB device not recognized. If reset button of the bluepill board is being pressed, the board will be normal again and can be used to run again. But still, it will hang after some cycles.

Code in bluepill board:
[scargo_V1.2.2_test6.zip](https://github.com/bblanchon/ArduinoJson/files/6997196/scargo_V1.2.2_test6.zip)

Code in PC:
[low_level_msgpack.zip](https://github.com/bblanchon/ArduinoJson/files/6997258/low_level_msgpack.zip)

Note: (1) The code on PC side is able to auto-reconnect if the reset button on the hang bluepill board is being pressed.
 

Need some helps / advises / solutions on solving communication issue. 

Bluepill board Specs:
MCU: stm32f103c8t6
SRAM: 20kB
FLASH: 64 kB

Best,
Samuel

bblanchon, 2021-08-17:
Hi Samuel,

From what I can see, your program stores pointers to strings in the `JsonDocument`, but the pointers dangle because the `JsonDocument` is destructed.
Instead, you should copy the strings.
[The ArduinoJson Troubleshooter covers this topic](https://arduinojson.org/v6/troubleshooter/#runtime/deserialization/ok/crash/no-usual-suspect). 

Best Regards,
Benoit

HappySamuel, 2021-08-18:
Hi @bblanchon 

Thank you for the fast reply. For the stores pointers to strings in JsonDocument that you mentioned, can help me highlight which section of code? I don't quite get what you mean.

At the mean time, i will go to ArduinoJson Troubleshooter to get some guidance.

Best,
Samuel

bblanchon, 2021-08-18:
I was refering to the following statement:

    cmd.system = root["SYS#"].as<const char*>();

I wrongly assumed that `system` was a `const char*`, but I just realized that it’s a `String`, so it should be OK.

Did you manage to get some guidance from the troublehsooter?
Can you locate the line that causes the crash (preferably using ArduinoTrace)?

HappySamuel, 2021-08-18:
Hi @bblanchon 

I followed the suggestion in [](https://arduinojson.org/v6/troubleshooter/#runtime/deserialization/ok/crash/no-usual-suspect) by changing `as<const char*>()` to `as<String>()` . I'm going to try it out and see if this solved my issue.

For the troubleshooter guides, i found that there's no conversion of JsonObject from the StaticJsonDocument before extracting the value.
```
    StaticJsonDocument<1024> read_doc;
    auto error = deserializeMsgPack(read_doc, SerialPC);
    JsonObject readJson = read_doc.as<JsonObject>();
```
 And it's directly extracting value from StaticJsonDocument doc.
```
  StaticJsonDocument<512> doc;
  deserializeJson(doc, file);
  username = doc["username"];  // ✅ stores a copy
```
Would like to know why it can being used directly?

I also read [](https://arduinojson.org/v6/example/config/) , and found this line `config.port = doc["port"] | 2731;`
Does this line of code means as following?
If there's no doc["port"], config.port = 2731
If there's doc["port"], config port = doc["port"]

Lastly, i never use ArduinoTrace before, so i think i will need to learn how to use ArduinoTrace to locate the line that causes the crash.

Best,
Samuel

bblanchon, 2021-08-19:
Hi Samuel,

We need to know where the program crashes. 
[ArduinoTrace](https://github.com/bblanchon/ArduinoTrace) is very straightforward: add as many calls to `TRACE()` as you need and run the program.

Best regards
Benoit

HappySamuel, 2021-08-21:
Hi @bblanchon 

I have tried TRACE() of [ArduinoTrace](https://github.com/bblanchon/ArduinoTrace) , however i found that it couldn't help me to trace where does the program crashed. It's due to the following configuration:
1. There're 2 USBs available on my board: 
-   Direct USB (CDC Serial) for fast communication, that can give me 50Hz - 100Hz bi-directional communication
-   Uart USB (Serial1, baudrate:115200) can only give me 10-20Hz bi-directional communication
2. Using Uart USB to perform bi-directional communication, while having Direct USB as output of TRACE() can observe the code on runtime. But the code won't crashed, maybe due to running slower?
3. Using Direct USB to perform bi-directional communication, while having Uart USB as output of TRACE() couldn't observe anything, and the code will crashed.

Best,
Samuel

bblanchon, 2021-08-22:
Hi Samuel,

If you cannot narrow down to the exact line that causes the crash, you must simplify your program to isolate the issue.
The goal is to create an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) with 50 lines at most.

Best regards,
Benoit

HappySamuel, 2021-08-24:
Hi @bblanchon 

I have found out the issue that troubles me. For using Direct USB, it allows carrying out fast communication (50-100Hz), but the allowable bytes of data that to be sent to the board is lesser than Uart USB (communication 10-20Hz). According to my test, as long as i constrains the data_write (PC -> bluepill) to be less than 120 bytes, the communication can be established and stable. For the return byte msg, data_read (PC <- bluepill) can be twice / triple size of the data_write, i guess is because the PC (i7) is stronger? As bluepill board is just using cortex M3 mcu, therefore weaker than PC? Not sure if my assumption is correct or not.

Anyway, thank you for your advises and tool (ArduinoTrace).

Best,
Samuel

closed 2021-08-24 by HappySamuel

bblanchon, 2021-08-24:
You're welcome, Samuel; I'm glad you solve your problem!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: