bblanchon/ArduinoJson #1632: Adding CR/LF to end of Serialization

dpetican, 2021-08-18:
This may seem like its pointless but I do a lot of printing json to a serial port for debug.   Each json document ends up right after the other rather than on separate lines.  I've gotten around this by creating a macro that prepends some text which includes a newline before the json.  Note that since this is debug output it is never deserialized.

Macros work but it would be more elegant if I could incorporate this directly in ArduinoJson.  I'm wondering if a custom writer class would be the way to go.  If so where would I look to duplicate the built-in writer class and extend it? debugOutput is defined as a global bool somewhere.  Migrating to v6 code soon.  Thanks.

As a feature request would it be hard to add support to the serialize class to pre/postpend text?  I know this is very edge case.

Macro:
```
#define DEBUG_JSON(pretext, object) if(debugOutput){\
    DEBUG_PORT.println(pretext);\
    object.printTo(DEBUG_PORT);\
    }
```

And used like this:
`DEBUG_JSON(F("\n<<<SERIAL=="), doc);`

Or this depending on direction:
`DEBUG_JSON(F("\n>>>SERIAL=="), doc);`

bblanchon, 2021-08-19:
Hi @dpetican,

There is no need to add such feature to the library, because it can easily be added in the user code.
With ArduinoJson 6, you can create a `serializeJsonLine()` like so:

    void serializeJsonLine(const JsonDocument& doc, Print& dst) {
      serializeJson(doc, dst);
      dst.println();
    }

Best Regards,
Benoit

dpetican, 2021-08-20:
> This may seem like its pointless but I do a lot of printing json to a serial port for debug.

Yes it was pointless.  Sorry.  Its just my penchant for doing preprocessor macros that gets in the way of proper solutions sometimes.

closed 2021-08-20 by dpetican