bblanchon/ArduinoJson #1633: Check temperature value from filtered deserialised weatherapp example

haeleng, 2021-08-19:
Hi,

I'am learning C++ for ArduinoJson for my hobby.
I'm not so familiar with objects because I used to program in the late 80's with Pascal
I have your book but still I'm getting stuck
I use a Arduino Mega with Visual Studio 2017

Question is about the example code below.
I like to check if the temperature is above 18 degree celcius.
How do I get the value of temp from the doc object?
thanks

	// The filter: it contains "true" for each value we want to keep
	StaticJsonDocument<200> filter;
	filter["list"][0]["dt"] = true;
	filter["list"][0]["main"]["temp"] = true;

	// Deserialize the document
	StaticJsonDocument<400> doc;
	deserializeJson(doc, input_json, DeserializationOption::Filter(filter));
	//deserializeJson(doc, input_json);

	serializeJsonPretty(doc, Serial);

haeleng, 2021-08-19:
Hi Benoid,

What about the code below

	// The filter: it contains "true" for each value we want to keep
	StaticJsonDocument<200> filter;
	filter["list"][0]["dt"] = true;
	filter["list"][0]["main"]["temp"] = true;

	// Deserialize the document
	StaticJsonDocument<400> doc;
	deserializeJson(doc, input_json, DeserializationOption::Filter(filter));
	//deserializeJson(doc, input_json);// werkt zonder filtering

	JsonObject object = doc.as<JsonObject>();

	JsonVariant Timeval = object["list"][0]["dt"];
	if (Timeval.isNull()) {
		Serial.println(F("Error: Time missing"));
	}
	JsonVariant Tempval = object["list"][0]["main"]["temp"];
	if (Tempval.isNull()) {
		Serial.println(F("Error: Temp missing"));
	}

	Serial.print(F("Time= "));
	Serial.println(Timeval.as<long>());
	Serial.print(F("Temp= "));
	Serial.println(Tempval.as<float>());
	if (Tempval.as<float>() > 4) Serial.println("above");
	else Serial.println("below");

	serializeJsonPretty(doc, Serial);

bblanchon, 2021-08-20:
Hi @haeleng,

You can write:

```c++
if (doc["list"][0]["main"]["temp"] > 18)
```

Best regards,
Benoit

closed 2021-09-06 by bblanchon