bblanchon/ArduinoJson #1637: Feature Request: Support chucked data streams to be better support HTTP1.1 clients

GeorgeKerwood, 2021-08-24:
Firstly, many congratulation on, and thanks for, your fantastic project! It's been a great help to me and I much enjoyed the book which has improved invaluable alongside. 

I would humbly request that you consider supporting the chunked transfer encoding of http 1.1 client streams. I have recently completed a project deploying [WiFiNina](https://github.com/arduino-libraries/WiFiNINA) on an [Arduino Nano 33 IoT](https://store.arduino.cc/usa/nano-33-iot) which integrates a [NINA W-102](https://www.u-blox.com/en/product/nina-w10-series-open-cpu) WiFi. Since the library does not support http 1.0 degradation, a very large part of the work effort was preparing various REST API responses to "load" the JSON content to buffers prior to deserialization.

Many Thanks,

George

bblanchon, 2021-08-25:
Hi George,

I will add a new decorator in the [StreamUtils](https://github.com/bblanchon/ArduinoStreamUtils) library, but I'm very busy at the moment, so I cannot give any ETA.

When you say "the library does not support http 1.0 degradation", which library are you talking about?

Best regards,
Benoit

GeorgeKerwood, 2021-08-25:
Salut!

Thanks for the reply. Great, no particular hurry, my project is concluded for the time being I'd just thought it would make a useful addition.

"The library" in this case was [WiFiNina](https://github.com/arduino-libraries/WiFiNINA). Unless of course it was just that I couldn't discover how to do so. A further caveat had been that [Postman](https://www.postman.com/), which I had been using to prepare and test my requests, also does not support http 1.0 so the debugging/testing on the embedded side would of been considerable longer with that limitation.

Your hard work is apricated, Best Regards,
George