bblanchon/ArduinoJson #1641: Bonjour Benoit, Not a bug. But how to?

nodoubtman, 2021-08-28:
Hello Benoit, I'm Marc from Montreal,QC.

I have a question:

I have a file in spiffs: 
contains records of radio stations web:

structures:

`{
  "stations": [
    {
      "name": "SLEEP RADIO",
      "url": "http://149.56.234.138:8169/stream",
      "id": 260
    },
    {
      "name": "BEATLES-128k",
      "url": "http://listen.181fm.com/181-beatles_128k.mp3",
      "id": 261
    },
    {
      "name": "AMBI NATURE RADIO",
      "url": "http://94.23.252.14:8067/stream",
      "id": 262
    },
    {
      "name": "Beethoven Channel",
      "url": "http://212.83.149.66:8297/stream",
      "id": 263
    },
    {
      "name": "Heart 80s",
      "url": "http://media-ice.musicradio.com/Heart80s",
      "id": 264
    },
    {
      "name": "BIRD SONG",
      "url": "http://streaming.radio.co/s5c5da6a36/listen",
      "id": 265
    },
    {
      "name": "Classical",
      "url": "http://51.15.152.81:8290/stream",
      "id": 266
    },
    {
      "name": "Chill Lounge Florida",
      "url": "http://us5.internet-radio.com:8283/",
      "id": 267
    },
    {
      "name": "Ambient Radio",
      "url": "http://uk2.internet-radio.com:31491/",
      "id": 268
    },
    {
      "name": "VIRGIN RADIO",
      "url": "http://139.162.245.57:8115/stream",
      "id": 269
    },
    {
      "name": "I Love Smooth Jazz",
      "url": "http://us5.internet-radio.com:8096/",
      "id": 270
    },
    {
      "name": "Keith Jarrett - Over The Rainbow",
      "url": "http://us5.internet-radio.com:8201/",
      "id": 271
    },
    {
      "name": "Joy Radio",
      "url": "http://uk5.internet-radio.com:8174/",
      "id": 272
    },
    {
      "name": "Smooth Jazz CD 101.9 New York",
      "url": "http://us3.internet-radio.com:8485/",
      "id": 273
    },
    {
      "name": "CLASSIC ROCK RADIO HD",
      "url": "http://us5.internet-radio.com:8267/",
      "id": 274
    },
    {
      "name": "Classic Rock Florida HD",
      "url": "http://us4.internet-radio.com:8258/",
      "id": 275
    },
    {
      "name": "Smooth Jazz Florida",
      "url": "http://us4.internet-radio.com:8266/",
      "id": 276
    },
    {
      "name": "Bach - C Major Prelude",
      "url": "http://us3.internet-radio.com:8232/",
      "id": 277
    },
    {
      "name": "Classic Hits Global HD",
      "url": "http://us2.internet-radio.com:8075/",
      "id": 278
    },
    {
      "name": "Box UK Radio danceradiouk",
      "url": "http://uk7.internet-radio.com:8226/",
      "id": 279
    },
    {
      "name": "INTRA NATURE RADIO",
      "url": "http://104.251.118.50:8793/stream",
      "id": 280
    },
    {
      "name": "NATURE RADIO SLEEP",
      "url": "http://larry.torontocast.com:1540/;",
      "id": 281
    },
    {
      "name": "100 GREATEST CLASSICAL",
      "url": "http://62.171.171.202:8140/radio.mp3",
      "id": 282
    },
    {
      "name": "RELAXING RADIO 365",
      "url": "http://s10.voscast.com:9574/",
      "id": 283
    },

    {
      "name": "RELAXING SOUNDS",
      "url": "http://centauri.shoutca.st:8063/live",
      "id": 284
    }

  ]
}`

I just wanted to know if there's a way to insert a new station as it:

    {
      "name": "NOUVELLE STATION",
      "url": "https... nouvelle station
      "id": 285
    }

Thank you.
Marc :)

bblanchon, 2021-08-28:
Hi Marc,

Here is to add the new station:

```c++
JsonObject station = doc["stations"].createNestedObject();
stations["name"] = "NOUVELLE STATION";
stations["url"] = "https... nouvelle station";
stations["id"] = 285;
```

Ensure that the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) is large enough; otherwise, the new station would be missing.

See also:

* [Serialization Tutorial](https://arduinojson.org/v6/doc/serialization/)
* [JsonVariant::createNestedObject()](https://arduinojson.org/v6/api/jsonvariant/createnestedobject/)
* [Why is the output incomplete?](https://arduinojson.org/v6/issues/incomplete-output/)

Marc, this is a fairly simple problem; I'm surprised you didn't find the answer in the documentation.
Please let me know if there is a way I can improve it.

Best regards,
Benoit

nodoubtman, 2021-08-28:
OK, Benoit. J'avoue que j'ai mal posé ma question, je sais c'est simple de faire un serializeJson avec l'assistant. Merci.

Mais comment je peux ajouter la nouvelle station dans le fichiers spiffs, sans tout chamboulé dans la mémoire.

Merci bien.

J'espère que cette fois-ci la question est bien posée :)
Bonne journée.

A+
Marc.

nodoubtman, 2021-08-28:
oups!.... j'aurais pas du la fermer

bblanchon, 2021-08-29:
Hi @nodoubtman,

So I guess your question is "how to append an object to a file?".

There are two options, depending on the way you shape the content of the file.

**Option 1**:  The file contains an array (that's your case).  
In this case, you must deserialize the file, append the object and serialize back, overriding the file.

**Option 2**: The file contains line-separated objects (i.e., [JSONLines](https://jsonlines.org/) or [ndjson](http://ndjson.org/)).
In this case, you can open the file in append mode and serialize only the new element.
To read all objects of this file, you need to call `deserializeJson()` repeatedly.

Best regards,
Benoit

nodoubtman, 2021-08-29:
Merci Benoit tu es le meilleur :))

Bonne journée.
Marc.

bblanchon, 2021-08-30:
Merci, Marc, j'ai corrigé la page.
N'oublie pas de [mettre une étoile](https://github.com/bblanchon/ArduinoJson/stargazers) pour soutenir ArduinoJson 😉 
Bonne chance pour ton projet !

closed 2021-08-30 by bblanchon