bblanchon/ArduinoJson #1642: Invalid use of incomplete type

apeiris, 2021-08-28:


apeiris, 2021-08-28:
.pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Strings/StringAdapter.hpp:16:37: error: invalid use of incomplete type 'class ArduinoJson6183_D1::StringAdapter<const char (*)[8], false, void>'

in version 6.17.3 it works fine but not in 6.18.3

apeiris, 2021-08-28:
Re-opening, Closed accidently.

bblanchon, 2021-08-29:
Hi @apeiris,

I'm sorry, I don't have enough information to help you.
Please provide the complete compiler output and the relevant lines from your program.

Best regards,
Benoit

apeiris, 2021-08-29:
Hi Benoit,

Thanks for quick response. Following 1)is the complete output and the 2) program.
Please note that Version 16.17.3 compiles without any error.

Best regards
Tony
1)--------------------------------------------------------------------------
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.ef99cd7
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - tool-openocd-esp32 2.1000.20201202 (10.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <webthing-arduino> 0.11.7+sha.88d49df
|   |-- <ArduinoJson> 6.18.3
|   |-- <ESPAsyncWebServer-esphome> 1.3.0
|   |   |-- <AsyncTCP-esphome> 1.2.1
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ArduinoJson> 6.18.3
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <AsyncTCP-esphome> 1.2.1
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <ArduinoJson> 6.17.3
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <AsyncTCP-esphome> 1.2.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.3
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <AsyncTCP-esphome> 1.2.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.18.3
|-- <ESP32 AnalogWrite> 0.2.0
|-- <Adafruit AHTX0> 2.0.1
|   |-- <Adafruit BusIO> 1.9.1
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|   |-- <Adafruit Unified Sensor> 1.1.4
|-- <Adafruit AHTX0> 2.0.0
|   |-- <Adafruit BusIO> 1.9.1
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|   |-- <Adafruit Unified Sensor> 1.1.4
|-- <PubSubClient> 2.8.0
|-- <StreamUtils> 1.6.1
|   |-- <EEPROM> 1.0.3
|-- <Adafruit BusIO> 1.9.1
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Wire> 1.0.1
[... 867 lines cut ...]

void do_fade(const JsonVariant &input)
{
    JsonObject inputObj = input.as<JsonObject>();
    long long int duration = inputObj["duration"];
    long long int brightness = inputObj["brightness"];

    delay(duration);

    ThingDataValue value = {.integer = brightness};
    propertyLampLevel.setValue(value);
    int level = map(brightness, 0, 100, 255, 0);
    analogWrite(lampPin, level);

    ThingDataValue val;
    val.number = 102;
    ThingEventObject *ev = new ThingEventObject("overheated", NUMBER, val);
    ESP_LOGI(TAG, "Queued Event ->%s", ev->name.c_str());
    deviceLamp.queueEventObject(ev);
}

ThingActionObject *action_generator(DynamicJsonDocument *input)
{
    return new ThingActionObject("fade", input, do_fade, nullptr);
}

bblanchon, 2021-08-30:
Relevant part of the output:

```
.../ArduinoJson/Document/JsonDocument.hpp: In instantiation of 'VariantRef JsonDocument::getOrAddMember(TChar*) [with TChar = const char [8]]':
.../ArduinoJson/Document/JsonDocument.hpp:107:54:   required from 'ArrayRef JsonDocument::createNestedArray(TChar*) [with TChar = const char [8]]'
src\main.cpp:301:57:   required from here
.../ArduinoJson/Document/JsonDocument.hpp:247:63: error: invalid use of incomplete type 'class StringAdapter<const char (*)[8], false, void>'
```

And here is `src\main.cpp:301`:

```c++
JsonArray devices = doc.createNestedArray(&"devices");
```

You must remove the ampersand, like so:

```c++
JsonArray devices = doc.createNestedArray("devices");
```

There is a similar error on line 376.

It worked by accident (quite a miracle, honestly) on 6.17.3.

apeiris, 2021-08-30:
@bblanchon , All is good. Thanks ,

bblanchon, 2021-08-31:
You're welcome, @apeiris.
Thank you for using ArduinoJson and good luck with your project!

closed 2021-08-31 by bblanchon