bblanchon/ArduinoJson #1644: Nesting object from SD card

rasppliances, 2021-08-29:
Hello,

I am reading a JSON document from an SD card on an ESP8266.  I want to send the data read in over serial to an Adafruit MacroPad RP2040.  But there's a twist.  I want the data from the sd card to go into an element of a JsonDocument, then send the JsonDocument.

Say for instance the file on my SD card is:

{
  "name":"Fred",
  "wife": {
    "name":"Wilma"
  }
}

I want to read that from the SD card and then send something like this over serial:

{
  "fileName":"fred.json",
  "fileData":
  {
    "name":"Fred",
    "wife": {
      "name":"Wilma"
    }
  }
}

Is there a way to build this out without having to iterate through the contents of the SD card?

Thanks
Rick

bblanchon, 2021-08-30:
Hi Rick,

You can insert the first document into the second, like so:

```c++
doc2["fileData"] = doc1;
```

This line will perform a deep copy.
It's currently not possible to avoid the copy (see issue #1343).

Best regards,
Benoit

bblanchon, 2021-08-30:
Alternatively, if you don't need to parse the content of the SD card, you can load the file into a string, and call [serialized()](https://arduinojson.org/v6/api/misc/serialized/).

rasppliances, 2021-08-30:
Thanks for answering.  I think I might have had other issues that made me think that this was causing an issue.  I'll have to try again later today.

closed 2021-09-06 by bblanchon