bblanchon/ArduinoJson #1645: ArduinoJson conflicts with WiFiNINA on Arduino WiFi Rev2

swabygw, 2021-08-29:
Hello - I have upgraded from the Arduino UNO to the Rev2 so that I can run a web server on the microcontroller card.  The web server on the Rev2 works quite well, actually.  However, upon adding in the ArduinoJson.h #include, it makes my WiFiServer declaration, on line 30 of the image, into an error, as shown in the image.  Removing the ArduinoJson.h #include fixes the issue, but I need to use ArduinoJson, too.  Any idea why they're conflicting, or how to resolve it?  (sorry for the question, but the error is a bit advanced for me)

[image]

bblanchon, 2021-08-30:
Hi @swabygw,

It looks like you're using Visual Micro, am I right?
Is it an actual compilation error, or just an Intellisense error as in #1181?

Best regards,
Benoit

swabygw, 2021-08-31:
Incredible.  That was it - it was just an Intellisense error.  Sorry for the false alarm.

closed 2021-08-31 by swabygw

bblanchon, 2021-08-31:
No problem 😉 
Did you manage to mute the error?

swabygw, 2021-09-01:
Nope - learned to live with it.  It still compiles and uploads successfully in vMicro, despite the Intellisense error.