bblanchon/ArduinoJson #1648: Unable to compile in Visual Studio

prosd, 2021-09-01:
First I want to thank you for the great work!

I downloaded the single header to install in a Visual Studio c++ project.
Compilation of the library fails at line #5042 with error C2227 "**left of '->read' must point to class/struct/union/generic type**"

Extract from ArduinoJson header file where build fails:
`Template <typename TSource, typename Enable = void>
struct Reader {
 public:
  Reader(TSource& source) : _source(&source) {}
  int read() {
    return _source->read();       ///// Fails at this line
  }
  size_t readBytes(char* buffer, size_t length) {
    return _source->readBytes(buffer, length);
  }
 private:
  TSource* _source;
};`

Thank you for assisting!

bblanchon, 2021-09-02:
Hi @prosd,

[ArduinoJson compiles fine on Visual Studio 2010 to 2019](https://ci.appveyor.com/project/bblanchon/arduinojson/branch/6.x); there is probably something wrong with your program.
What's the type of the second argument passed to [deserializeJson()](https://arduinojson.org/v6/api/json/deserializejson/)?

Best regards,
Benoit

prosd, 2021-09-02:
Hello Benoit,

You are completely right! Using "ifstream::open" instead of "fopen_s" has resolved my issue.

Sorry to have troubled you for this matter. Nevertheless thank you for your time and effort.

Best regards, 
Jan

closed 2021-09-02 by prosd

bblanchon, 2021-09-03:
You're welcome. Thank you for using ArduinoJson.
Good luck with your project, Jan!