bblanchon/ArduinoJson #1652: Delete a nested key

mithunrockey, 2021-09-12:
Hi Benoit,

First of all, thank you for this great library.

Wondering is there a way to delete a nested key somewhere?
foe example, if I want to remove the "A" from this JSON?
{"H":{"K":"","A":"","M":0}}

Can .remove() do this?

thanks!

bblanchon, 2021-09-13:
Hi @mithunrockey,

```c++
doc["H"].remove("A");
```

Best regards,
Benoit

mithunrockey, 2021-09-16:
Hi @bblanchon,

Sorry, I already tried the same but got like below for  {"H":{"K":"","A":"","M":0}}

DynamicJsonBuffer jsonBuffer(200);
JsonObject& root = jsonBuffer.parseObject(payload);

if (!root.success())
{
    Serial.println("JSON parsing failed!");
}
else
{
    root["H"].remove("A");
}

Produced the following error.

In member function 'void MyArduino::arduino_process_serial_data(char*, int)':
error: 'class ArduinoJson::Internals::JsonObjectSubscript<const char*>' has no member named 'remove'
   root["H"].remove("A");
             ^~~~~~
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

Should be something wrong I'm doing?

thanks!

bblanchon, 2021-09-17:
Hi @mithunrockey,

I see that you're using ArduinoJson 5.
Indeed, this version didn't allow deleting a nested key like I showed above.
Instead, you must extract a [`JsonObject`](https://arduinojson.org/v5/api/jsonobject/) reference and call [`remove()`](https://arduinojson.org/v5/api/jsonobject/remove/), like so:

```c++
JsonObject& hobj = root["H"];
hobj.remove("A");
```

I really recommend that you switch to v6 which is superior to v5 on almost every point.

See also:

* [Migration guide](https://arduinojson.org/v6/doc/upgrade/)
* [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/)

Best regards,
Benoit

mithunrockey, 2021-09-17:
Thanks Benoit!

Yes, I'm using the version 5. The project I'm working with started 5 years back and I think is nearly there. Will surely move to v6 soon as I can, as there will be a bit for me to do for this migration.

Thanks again for your kind help and this lovely, helpful lib!

bblanchon, 2021-09-18:
You're welcome, @mithunrockey.
Thank you for using ArduinoJson!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

closed 2021-09-18 by bblanchon