bblanchon/ArduinoJson #1653: Serialize it in JSON Format and send it by RF24 library !

oussama415, 2021-09-14:
Hi, I have a problem when i use ArduinoJson with the RF24 library
I want to serialize it in JSONFormat and after that sending it with RF24 !!!!

When i copy a sample code in my RF24 code, before sending by RF24, the function serialise return nothing 
{} : what it return !!!

Can you help me please 👍

bblanchon, 2021-09-15:
Hi @oussama415,

Which sample program are you referring to?

Best regards,
Benoit

oussama415, 2021-09-15:
Hi @bblanchon Thanks for your fast reply 
This is a simple code for transmitting

```
#include <RF24Network.h>
#include <RF24.h>
#include <SPI.h>
#include <ArduinoJson.h>

RF24 radio(7, 8);               // nRF24L01 (CE,CSN) (7,8)
RF24Network network(radio);      // Include the radio in the network
const uint16_t this_node = 00;   // Address of this node in Octal format 
const uint16_t node01 = 01;    

void setup()
{
  Serial.begin(115200);
  SPI.begin();
  radio.begin();
  network.begin(90, this_node);  
  radio.setDataRate(RF24_2MBPS);
}
void loop()
{
  network.update();
  StaticJsonDocument<300> doc;
  doc["var1"] = 50;

  RF24Adapter radioForArduinoJson(radio); // Not sending 
  serializeJson(doc, radioForArduinoJson); // I get nothing like this : {}

//---------------------- Sending normally char type----------------------------//
  //  RF24NetworkHeader header1(node01);     // (Address where the data is going)
  //  network.write(header1, &json1, sizeof(json1)); 
//---------------------- Sending normally char type----------------------------//

}
```

i have nothing in return for the line :  serializeJson(doc, radioForArduinoJson); 
i want to serialize in JSON Format and send it by radio 

I see you have already make a class for RF24Adapter, i don't know when exactly i copy it !!! 

This is the class : 

```
class RF24Adapter {
 public:
  RF24Adapter(RF24 &radio) : _radio(&radio) {}

  int read() {
    char buf[1];
    _radio->read(buf, 1);
    return buf[0];
  }

  size_t readBytes(char *buffer, size_t length) {
    _radio->read(buffer, static_cast<uint8_t>(length));
    return length;
  }

  size_t write(uint8_t c) {
    return _radio->write(&c, 1) ? 1 : 0;
  }

  size_t write(const uint8_t *buffer, size_t length) {
    return _radio->write(buffer, static_cast<uint8_t>(length)) ? length : 0;
  }

 private:
  RF24 *_radio;
};
```

Thanks for your help !

bblanchon, 2021-09-16:
Hi @oussama415,

I have no experience with RF24Network and I cannot spot any obvious mistake in the code, so I cannot help you.
Here are a few ideas to troubleshoot this issue:

1. Get ArduinoJson out of the way:
   1. make sure you can transmit plain text (e.g "hello world")
   2. try to send the string "{\"var1\":50}" to simulate the work of the library
2. Restore ArduinoJson
   1. try with a temporary buffer, or a `String`
   2. if it works, you can retry with `RF24Adapter`
3. Ask for help
    1. If nothing works, [please contact the RF24Network team](https://github.com/nRF24/RF24Network/issues)
    2. If you get no answer, try [stackoverflow](https://stackoverflow.com/) or [the Arduino Forum](https://forum.arduino.cc/)

Please let me know how you solve your issue, so other users can find the answer in this conversation.
If there is enough material, I will add a dedicated page on arduinojson.org

Best regards,
Benoit

bblanchon, 2021-10-01:
Any progress, @oussama415?

oussama415, 2021-10-01:
No progress, the solution is to make a message with JSON format a with a
char  and send it by RF24   but the problem is the maximum  we can send but
RF24 is 32 bytes !!!

bblanchon, 2021-10-02:
I didn't know RF24 was limited to 32 bytes.
That makes it a very bad candidate for JSON.

I recommend that you send binary structures instead.
If you don't know what I'm talking about, please watch the first part of [this video](https://youtu.be/hAB4TdX8dwM).