bblanchon/ArduinoJson #1654: Is possibl to keep a global JsonDoc in PSRAM esp32 ?

johnnytolengo, 2021-09-23:
I am traying to keep a global Json object in PSRAM for logging texts or any data, but it seems that's not possible.

Using:
```
struct SpiRamAllocator {
    void* allocate(size_t size) {
      return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
    }
    void deallocate(void* pointer) {
      heap_caps_free(pointer);
    }
  };
  using SpiRamJsonDocument = BasicJsonDocument<SpiRamAllocator>;
```

Within the class:
```
class myClass{
  public:
 myClass();
 void test();
 SpiRamJsonDocument jsonLogs;
}

myClass::myClass(): jsonLogs(10000){  //initialize 
}

myClass::test(){
  jsonLogs["hellotest"] = 12;
  serializeJsonPretty(jsonLogs, Serial);   //THAT RETURNS ALWAYS EMPTY
}
```

The document jsonLogs returns always empty and after checking the PSRam memory space I see that no memory is taken at all.
My questions:  

- is that possible to keep an object in Ram? 
- if yes, is that compromising the Heap while the object is in Ram?
- if no, what's an alternative for that?

Regards,

bblanchon, 2021-09-24:
Hi @johnnytolengo,

Using a global `JsonDocument` is a very bad idea in general, please see [I found a memory leak in the library!](https://arduinojson.org/v6/issues/memory-leak/).
Then, a `JsonDocument` is a poor choice to store logs and (binary) data; remember that ArduinoJson is a *serialization* library, not a container library.

Instead, I recommend that you use containers from the STL, like `std::vector` or `std::deque`.
You can configure STL containers with allocator classes (just like `JsonDocument`), but you can also configure your ESP32 to [make the PSRAM accessible from the regular allocation functions](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#provide-external-ram-via-malloc).

Regarding your current issue, it looks like `heap_caps_malloc()` returned `NULL`.
Can you check the result of `doc.capacity()`? It should be `10000`. If it's `0`, it confirms that `heap_caps_malloc()` returned `NULL`.
In that case, it probably means that your ESP32 cannot use the external RAM for some reason.
Please let me know how you solve this issue so I can add this scenario in the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/).

Best regards,
Benoit

johnnytolengo, 2021-09-24:
I understand, I have to take some way to retain the data somewhere before/after I use ArduinoJson.
thank you for the detailed explanation, that makes sense

bblanchon, 2021-09-25:
@johnnytolengo, can you check `doc.capacity()` so we can understand why the document was empty? (see my previous message)

johnnytolengo, 2021-09-25:
seems that the psram memory is not allocated, I think that the reason is because object is created before the psram is enabled. 

output:
```
capacity:0 
{}
```

BTW my solution was hold the fixed-lenght data in a struct array in psram, that worked, the downside is that the allocated memory is always taken but it doesn't hurt when you have available memory.

bblanchon, 2021-09-25:
> I think that the reason is because object is created before the psram is enabled.

So, if you declare the `JsonDocument` in the `setup()` function, the result of `capacity()` gets back to normal?

johnnytolengo, 2021-09-25:
in the setup() is working good. 
But in my example the JsonDocument is used as a Global variable inside the class, and contructing it in the Class constructor.

bblanchon, 2021-09-26:
That's the information I was looking for; thanks!
I guess we can close this issue now.