bblanchon/ArduinoJson #1655: How to iterate through nested objects ?

GeorgeFlorian, 2021-09-24:
I have the following JSON :
```
{
    "network_settings": {
        "connection": "wifi",
        "ip_type": "dhcp",
        "ssid": "Jorj_2.4",
        "ip_address": "",
        "gateway": "",
        "subnet": "",
        "dns": ""
    },
    "input": {
        "ip_1": "Not Set",
        "port_1": "Not Set",
        "ip_2": "Not Set",
        "port_2": "Not Set"
    },
    "output": {
        "timer_1": "0",
        "timer_2": "0"
    },
    "relay_state": {
        "state1": "On",
        "state2": "Off"
    },
    "wiegand": {
        "database_url": "Not Set",
        "pulse_width": "90",
        "pulse_gap": "90"
    },
    "rfid": {
        "ip_rfid": "Not Set",
        "port_rfid": "Not Set"
    }
}
```
And I want to clear all the values of the nested objects (network_settings, input, etc), but retain all the keys.

I though of iterating through them and make them `""`:
```
              StaticJsonDocument<768> read_config = readSettings("/config.json");
              JsonObject data = read_config.as<JsonObject>();

              for(JsonPair i : data){
                for(JsonPair k : data[i.key()]){
                  k.value().as<String>() = "";
                }
              }
```

But I get an error:
```
this range-based 'for' statement requires a suitable "begin" function and none was foundC/C++(2291)
```

bblanchon, 2021-09-25:
Hi @GeorgeFlorian,

Like `JsonDocument`, there are two ways to iterate over a `JsonVariant`: as an array or as an object.
To disambiguate, we need to cast the `JsonVariant` to a `JsonObject` like so:

```c++
for(JsonPair i : data){
  for(JsonPair k : data.value().as<JsonObject>()){
    //
  }
}
```

I added this scenario in the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/#compiletime/range-based-for-requires-begin).

BTW, `k.value().as<String>() = "";` doesn't change the value in the `JsonDocument` but in the returned `String`.
Instead, you must write `k.value().set("");`

Best regards,
Benoit

GeorgeFlorian, 2021-09-28:
@bblanchon Thank you. Works like a charm.

GeorgeFlorian, 2021-09-28:
@bblanchon Thank you for your prompt reply.

It doesn't work.

From:
```cpp
  for (JsonPair i : data)
  {
    for (JsonPair k : data.value().as<JsonObject>())
    {
      k.value().set("");
    }
  }
```
I get `class "ArduinoJson6184_91::ObjectRef" has no member "value"C/C++(135)`

and from
```cpp
  for (JsonPair i : data)
  {
    for (JsonPair k : data[i].value().as<JsonObject>())
    {
      k.value().set("");
    }
  }
```
I get `no operator "[]" matches these operands -- operand types are: ArduinoJson::JsonObject [ ArduinoJson::JsonPair ]C/C++(349)`

I am using the same logic in a JS script:
```javascript
for (let i in json_data) {
  for (let key in json_data[i]) {}
}
```

bblanchon, 2021-09-28:
I'm sorry, I meant

```c++
for(JsonPair i : data) {
  for(JsonPair k : i.value().as<JsonObject>()) {
    //
  }
}
```

GeorgeFlorian, 2021-09-28:
> I'm sorry, I meant
> 
> ```c++
> for(JsonPair i : data) {
>   for(JsonPair k : i.value().as<JsonObject>()) {
>     //
>   }
> }
> ```

Thank you.
Sorry to bother you.

closed 2021-09-28 by GeorgeFlorian