bblanchon/ArduinoJson #1656: JSON ever changing in size

victor7376, 2021-09-27:
I'm trying to figure out some coding for a project I'm wanting to make.

The Json that I'm receiving from a url also includes another url that you are required to go to after processing the first url.

Example:
1st URL = Json which includes the 2nd url
2nd url = New json info which also includes the 3rd url
3rd url = New json info which includes the 4th url
etc etc etc.

The json from the urls is always changing in size, so using the assistant doesn't help in this case.

The url I'm looking at is from this guide: https://chaturbate.com/apps/api/docs/index.html which shows this example for code https://chaturbate.com/apps/api/docs/rest.html

Any help would be grateful.

bblanchon, 2021-09-28:
Hi @victor7376,

Either you use a different `JsonDocument` for each request or one that is big enough for the largest response.

Please see: [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/).

Best regards,
Benoit

victor7376, 2021-09-28:
Ok, thank you.  I understand the capacity part, but how would I change the url that gets called first and replace it with the new url that is included in the json response from the first call?

Cheers

bblanchon, 2021-09-30:
ArduinoJson handles the JSON deserialization, not the HTTP request.
To change the URL, you must update the part of your program that makes the HTTP requests.
You can refer to [JsonHttpClient.ino](https://arduinojson.org/v6/example/http-client/) for an example.