bblanchon/ArduinoJson #1658: Is it possible to add new elements into JsonArray and store it to same memory SPIFFS.

Bharath-M-R, 2021-09-30:
Hi,

I have stored a JsonArray in file system of ESP8266. I have to open it and parse it and add few elements to the parsed array and again write to the same memory in file system.

Is it possible do so?

1. I am using ESP8266-01 module
2. platform - Arduino IDE.

thanks

bblanchon, 2021-09-30:
Hi @Bharath-M-R,

Please read [How to append a JSON object to a file?](https://arduinojson.org/v6/how-to/append-to-a-file/).

Best regards,
Benoit

Bharath-M-R, 2021-10-01:
Hi @bblanchon ,

Thanks for your response. I have gone through the link you have shared. But in my case I am not adding a single json object to the array. 
My Json structure is like,
{"DEVICE_ID":["DEV01","DEV02","DEV03","DEV04"]}   and I want to append one more Device ID  "DEV05" to this JsonArray.

I have tried ,
JsonArray.add("DEV05");
It's getting added to the JsonArray but when I write this object to file system and read again it's appears as same as {"DEVICE_ID":["DEV01","DEV02","DEV03","DEV04"]}

Hope you understand my point

Thanks

closed 2021-10-04 by Bharath-M-R