bblanchon/ArduinoJson #1659: Teensy4.0 section type conflict with static_asset_manifest_json_0

gannaramu, 2021-10-01:
I am using aWOT library with teensy 4.0 and having an issue when I include the ArduinoJson for a parse response function which looks like this: Apparently this code compiles for teensy3.6
```
bool parse_response(uint8_t *buffer)
{
  DeserializationError error = deserializeJson(doc, buffer);
  serializeJsonPretty(doc, Serial);

  if (error)
  {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return false;
  }

  return true;
}
```

```
In file included from lib\ArduinoJson\src/ArduinoJson/Numbers/FloatTraits.hpp:14:0,
                 from lib\ArduinoJson\src/ArduinoJson/Numbers/parseNumber.hpp:7,
                 from lib\ArduinoJson\src/ArduinoJson/Variant/VariantImpl.hpp:10,
                 from lib\ArduinoJson\src/ArduinoJson.hpp:32,
                 from lib\ArduinoJson\src/ArduinoJson.h:9,
                 from src\main.cpp:6:
lib\ArduinoJson\src/ArduinoJson/Deserialization/DeserializationError.hpp:98:5: error: messages causes a section type conflict with static_asset_manifest_json_0  
     ARDUINOJSON_DEFINE_STATIC_ARRAY(
     ^
In file included from src\main.cpp:4:0:
src\StaticFiles.h:2:5: note: 'static_asset_manifest_json_0' was declared here
   P(static_asset_manifest_json_0) = {
     ^
src\aWOT.h:74:44: note: in definition of macro 'P'
 #define P(name) static const unsigned char name[] PROGMEM
```
Compiler Information:
```
Processing teensy40 (platform: teensy; board: teensy40; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy40.html
PLATFORM: Teensy (4.11.0) > Teensy 4.0
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 1.94MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES:
 - framework-arduinoteensy 1.153.0 (1.53)
 - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1) 
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 90 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.18.4
|-- <Ethernet>
|   |-- <SPI> 1.0       
|-- <SPI> 1.0
Building in release mode
```
The [StaticFiles.json](https://github.com/gannaramu/Mini-SSS3/blob/main/Mini-SSS3-Firmware/src/StaticFiles.h)
The Code is at: https://github.com/gannaramu/Mini-SSS3/tree/main/Mini-SSS3-Firmware

Any help is appreciated

bblanchon, 2021-10-01:
Hi @gannaramu,

> error: X causes a section type conflict with Y

This looks a lot like #1518, so it's probably due to a bug in the Teensy 4 core.
As a workaround, you can disable the support for Flash string by setting [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/) to `0`.

Best regards,
Benoit

gannaramu, 2021-10-01:
Thanks Benoit, I was able to get it to work with this workaround. https://forum.pjrc.com/threads/68343-Teensy4-0-section-type-conflict-with-static_asset_manifest_json_0?p=289918&viewfull=1#post289918

closed 2021-10-01 by gannaramu