bblanchon/ArduinoJson #1660: Brackets in openweatherdata

tonbor, 2021-10-04:
https://arduinojson.org/v6/example/filter/ will not work, the brackets makes that the temp etc. values can not been extracted from the (smaller) json doc . Indeed this is a common problem with the arduinojson getting openweather data. Please solve this.

bblanchon, 2021-10-05:
Hi @tonbor,

I just tested the example on UNO and ESP32, and I can confirm it works on both.
Can you give more details about your problem?

Best regards,
Benoit

tonbor, 2021-10-05:
Just add  // Print the result
  serializeJsonPretty(doc, Serial);
  String temp = doc["list"]["main"]["temp"];
  Serial.print("Temp = ");
  Serial.println(temp);
}

void loop() {
  // not used in this example
}
This is the result: {
  "list": [
    {
      "dt": 1581498000,
      "main": {
        "temp": 3.23
      }
    },
    {
      "dt": 1581508800,
      "main": {
        "temp": 6.09
      }
    }
  ]
}Temp = null

bblanchon, 2021-10-06:
The expression is incorrect; it should be:

```c++
doc["list"][0]["main"]["temp"];
```

bblanchon, 2021-10-07:
You can read the [Deserialization Tutorial](https://arduinojson.org/v6/doc/deserialization/).
Also, the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) generates the code for you.
What else do you need?

closed 2021-10-09 by bblanchon