bblanchon/ArduinoJson #1669: SOLVED - serializeJson returns an output with trailing garbage

JulianNF, 2021-10-31:
Hello, firstly: thanks @bblanchon for all the work on arduinoJson! 🙏

# Issue description:
I followed the v6 tutorial for serializing JSON, but my output, though containing valid JSON, is trailed with varying amounts of garbage.

# Apparent Cause:
It seems that the result of serializing DOES get null terminated, BUT the result of `measureJson()` does not account for the null terminator. If we blindly apply the length returned by `measureJson()` to the `serializeJson()` call, we end up with a correctly formed JSON output, but any references when then make to the output contain a bunch of trailing garbage.

I haven't tested it, but I presume that the same happens with `measureJsonPretty()` and `serializeJsonPretty()`.

Note that I'm using arduinoJson to generate and send data via MQTT (pubSubClient library). I've checked thoroughly through traces and the issue is definitely happening with the output of `serializeJson()`, and not the pubSubClient library.

# Solution:
Adding one byte to the length returned by `measureJson()`:

``` cpp
void publishStatusMessage() {
	const int capacity = JSON_OBJECT_SIZE(3);
	StaticJsonDocument<capacity> jsonDoc;

	jsonDoc["aProperty"] = variableName;
	jsonDoc["someProperty"] = otherVariable;
	jsonDoc["anotherProperty"] = thirdVariable;

	size_t jsonLength = measureJson(jsonDoc) + 1; // Account for null-terminator, else trailing garbage is returned
	char output[jsonLength];
	serializeJson(jsonDoc, output, sizeof(output));

	mqttClient.publish(showerStatusTopic, output);
}
```

# Proposal:
@bblanchon Could you add the important detail of needing to account for the null terminator in the documentation? Thanks!

## Before
[image]

## After
[image]

bblanchon, 2021-10-31:
Hi @JulianNF,

I'm sorry you had this issue, but [the documentation](https://arduinojson.org/v6/api/json/measurejson/) already says that the NUL terminator is not included in the result.
Moreover, I think it's already implied by the function's name: it *measures* the *length* of the JSON document, and the NUL terminator is not part of the JSON document.
Also, it follows the usual C/C++ conventions: `strlen()`, `std::string::size()`, `String::size()`... they all exclude the NUL terminator.
So, honestly, I don't see what else I could do to prevent users from falling into this trap.

BTW, using a variable-length array like that is the recipe for a stack overflow.
I **strongly recommend** using a fixed-size, as shown in [the documentation](https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/).
Not only will your program be safer, but it will also be faster and lighter.

Best regards,
Benoit

bblanchon, 2021-10-31:
PS: I think I could add this scenario to the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/), can you copy and paste your report? Thanks.

closed 2021-11-09 by bblanchon