bblanchon/ArduinoJson #1672: Error serializeJson(doc, file) on SAMD51

sergioxdev, 2021-11-15:
Hi Guys

I'm testing ArduinoJson on an Adafruit Metro M4 Express AirLift

I use libraries
#include "SdFat.h"
#include "Adafruit_SPIFlash.h"

when I try to call:
serializeJson (doc, file);
to write the file on the SPIFlash connected to SAMD51
an error is returned to me.

I am attaching the example code to be able to emulate the problem:
`#include <SPI.h>
#include "SdFat.h"
#include "Adafruit_SPIFlash.h"
#include <ArduinoJson.h>

// Uncomment to run example with custom SPI and SS e.g with FRAM breakout
// #define CUSTOM_CS   A5
// #define CUSTOM_SPI  SPI
#if defined(CUSTOM_CS) && defined(CUSTOM_SPI)
  Adafruit_FlashTransport_SPI flashTransport(CUSTOM_CS, CUSTOM_SPI);

#elif defined(ARDUINO_ARCH_ESP32)
  Adafruit_FlashTransport_ESP32 flashTransport;
#else
  #if defined(EXTERNAL_FLASH_USE_QSPI)
    Adafruit_FlashTransport_QSPI flashTransport;
  #elif defined(EXTERNAL_FLASH_USE_SPI)
    Adafruit_FlashTransport_SPI flashTransport(EXTERNAL_FLASH_USE_CS, EXTERNAL_FLASH_USE_SPI);

  #else
    #error No QSPI/SPI flash are defined on your board variant.h !
  #endif
#endif

Adafruit_SPIFlash flash(&flashTransport);
FatFileSystem fatfs;
File myFile;

struct Config {
  char text[32];
};
#define n_row_config 1

const String base_config_[n_row_config][5] = {  
  {"1","text", "", "char", "32"},
};

char *filename_config = "/config.json";
Config config_;

void setup() {
  Serial.begin(115200);
  while (!Serial) {
    delay(1); // wait for serial port to connect. Needed for native USB port only
  }

  Serial.print("Initializing Filesystem on external flash...");
  
  flash.begin();

  if ( !fatfs.begin(&flash) ) {
    Serial.println("Error: filesystem is not existed. Please try SdFat_format example to make one.");
    while(1) yield();
  }

  Serial.println("initialization done.");

  myFile = fatfs.open("test.txt", FILE_WRITE);

  // if the file opened okay, write to it:
  if (myFile) {
    Serial.print("Writing to test.txt...");
    myFile.println("testing 1, 2, 3.");
    // close the file:
    myFile.close();
    Serial.println("done.");
  } else {
    // if the file didn't open, print an error:
    Serial.println("error opening test.txt");
  }

  // re-open the file for reading:
  myFile = fatfs.open("test.txt");
  if (myFile) {
    Serial.println("test.txt:");

    // read from the file until there's nothing else in it:
    while (myFile.available()) {
      Serial.write(myFile.read());
    }
    // close the file:
    myFile.close();
  } else {
    Serial.println("Error opening test.txt");
  }
  //-----------------------------------------------------
  if (flash.begin()) {
    Serial.println("Mounted file system");
    removeFile(filename_config);
    if (loadConfiguration(filename_config, config_)) {
      Serial.println("loadConfiguration: OK");
    }
    else {
      if (saveConfiguration(filename_config, config_)) {
        Serial.println("saveConfiguration: OK");
        if (loadConfiguration(filename_config, config_)) {
          Serial.println("loadConfiguration: OK");
        }
        else {
          Serial.println("loadConfiguration: NO");  
        }
      }
      else {
        Serial.println("saveConfiguration: NO");
      }
    }
  }
  
  //-----------------------------------------------------
}

void loop() {
  // nothing happens after setup
}

bool loadConfiguration(char *filename, Config &config) {
  Serial.println("Check file Exist");
  if (fatfs.exists(filename)) {
    Serial.println("Reading config file");
    File file = fatfs.open(filename, FILE_WRITE);
    if (file && file.size()) {
      size_t size_ =  file.size();
      size_ = (size_-(((n_row_config*6)-1)+(2*3))+2);
      const size_t capacity = JSON_OBJECT_SIZE(10) + 190;
      DynamicJsonDocument doc(capacity);
      DeserializationError error = deserializeJson(doc, file);
      if (error)
        Serial.println("Failed to read file, using default configuration");

      for (int row = 0; row < n_row_config; row++)
      {
        bool bool_tmp_;
        int num_tmp_ = base_config_[row][0].toInt();
        String name_tmp_ = base_config_[row][1];
        String value_tmp_ = base_config_[row][2];

        if(doc[name_tmp_]!="")
          value_tmp_ = doc[name_tmp_].as<String>();
        if(base_config_[row][3]=="char")
        {
          switch(num_tmp_)
          {
            case 1:
              strlcpy(config_.text, value_tmp_.c_str(), sizeof(config_.text));
              break;
          }
        }
      }
      Serial.println("Reading Completed");
      file.close();
      return 1;
    }
    file.close();
    Serial.println("Failed to load json config");
    return 0;   
  }
  Serial.println("Failed to load json config");
  return 0;
}

bool saveConfiguration(char *filename, Config &config) {
  File writeFile = fatfs.open(filename, FILE_WRITE);
  if (writeFile) {
    writeFile.println("_");
    writeFile.close();
    Serial.println("Successful file creation");
  } else {
    Serial.println("File creation Error");
  }
  if (fatfs.exists(filename)) {
    File file = fatfs.open(filename, FILE_WRITE);
    if (file && file.size()) {
      int size_ =  file.size();
      size_ = (size_-(((n_row_config*6)-1)+(2*3))+2);
      
      const size_t capacity = JSON_OBJECT_SIZE(n_row_config) + size_;     
      DynamicJsonDocument doc(capacity);
        
      for (int row = 0; row < n_row_config; row++)
      {
        char db_tmp[sizeof(base_config_[row][0])];
        base_config_[row][0].toCharArray(db_tmp, sizeof(base_config_[row][0]));

        int num_tmp_ = base_config_[row][0].toInt();
        String name_tmp_ = base_config_[row][1];
        String value_tmp_ = base_config_[row][2];

        if(base_config_[row][3]=="char")
        {
          switch(num_tmp_)
          {
            case 1:
              doc[name_tmp_] = config_.text;
              break;          
          }          
        }
        db_tmp[sizeof(name_tmp_)];
        name_tmp_.toCharArray(db_tmp, sizeof(name_tmp_));
      }

      bool error = serializeJson(doc, file);
      if (error)
      {
        Serial.println("Failed to write file - BLOCK HERE");
        return 0;
      }
      else
        Serial.println("Save Completed");
      file.close();
      return 1;
    }
    Serial.println("Failed to save json config");
    return 0;   
  }
  Serial.println("Failed to save json config");
  return 0;
}

void printFile(const char *filename) {
  Serial.println("Check file Exist");
  if (fatfs.exists(filename)) {
    Serial.println("Reading config file");
    File file = fatfs.open(filename, FILE_WRITE);
    if (file && file.size()) {
      while (file.available()) {
        Serial.print((char)file.read());
      }
    }
    file.close();
  }
}

void removeFile(char *filename) {
  fatfs.remove(filename);
  Serial.println("File Remove");
}
`
any suggestion is accepted

bblanchon, 2021-11-16:
Hi @sergioxdev,

> an error is returned to me.

What error is returned? Can you be more explicit?

This is probably unrelated but I noticed an error in `printFile()`: you use `FILE_WRITE` instead of `FILE_READ`.

BTW, did you try the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter)?

Best regards,
Benoit

sergioxdev, 2021-11-16:
hello @bblanchon

i am doing a series of tests, initially i used the DynamicJson,
now I'm testing with the StaticJson version,

I'll update you as soon as I understand the problem

sergioxdev, 2021-11-16:
Hi @bblanchon

I found the error in the functions:
- loadConfiguration()
- printFile()

when you access the file:
File file = fatfs.open (filename, FILE_WRITE);
must be replaced with
File file = fatfs.open (filename);

Now works

closed 2021-11-16 by sergioxdev