bblanchon/ArduinoJson #1673: HSPI pins and Datalogging help on ESP32

HarborDesigns, 2021-11-16:
**Iissue**  
I am trying to implement data logging on micro SD card using ESP32. I want to store my data in JSON format on SD card. I am using HSPI pins which is different from VSPI pins. Would the library work for HSPI pins of ESP32. If yes then how can I modify the example [code by RENZO MISCHIANTI on this link](https://www.mischianti.org/2020/01/26/manage-json-file-with-arduino-and-esp8266/). 

I can read and write a text file on SD card but when my code starts to write, it crashes ESP32 and makes it restart. The last print message which I get on serial window is this: "Start write......Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled."

**Environment**  
Arduino IDE 1.8.13]

**Reproduction**  
My code is mostly same as the link except the SPI pin definition for HSPI pins and the way I am setting up and initiating SPI protocol 
```
#include "FS.h"
#include "SD.h"
#include "SPI.h"
#include <ArduinoJson.h>

#define SCK  14
#define MISO  12
#define MOSI  13
#define CS  15

const char *filename = "/test.jso";  // <- SD library uses 8.3 filenames
 
File myFileSDCart;
 
JsonObject getJSonFromFile(DynamicJsonDocument *doc, String filename, bool forceCleanONJsonError = true ) {
    // open the file for reading:
    myFileSDCart = SD.open(filename);
    if (myFileSDCart) {
        // read from the file until there's nothing else in it:
//          if (myFileSDCart.available()) {
//              firstWrite = false;
//          }
 
        DeserializationError error = deserializeJson(*doc, myFileSDCart);
        if (error) {
            // if the file didn't open, print an error:
            Serial.print(F("Error parsing JSON "));
            Serial.println(error.c_str());
 
            if (forceCleanONJsonError){
                return doc->to<JsonObject>();
            }
        }
 
        // close the file:
        myFileSDCart.close();
 
        return doc->as<JsonObject>();
    } else {
        // if the file didn't open, print an error:
        Serial.print(F("Error opening (or file not exists) "));
        Serial.println(filename);
 
        Serial.println(F("Empty json created"));
        return doc->to<JsonObject>();
    }
 
}
 

bool saveJSonToAFile(DynamicJsonDocument *doc, String filename) {
    SD.remove(filename);
 
    // open the file. note that only one file can be open at a time,
    // so you have to close this one before opening another.
    Serial.println(F("Open file in write mode"));
    myFileSDCart = SD.open(filename, FILE_WRITE);
    if (myFileSDCart) {
        Serial.print(F("Filename --> "));
        Serial.println(filename);
 
        Serial.print(F("Start write..."));
 
        serializeJson(*doc, myFileSDCart);
 
        Serial.print(F("..."));
        // close the file:
        myFileSDCart.close();
        Serial.println(F("done."));
 
        return true;
    } else {
        // if the file didn't open, print an error:
        Serial.print(F("Error opening "));
        Serial.println(filename);
 
        return false;
    }
}
 
// Prints the content of a file to the Serial
void printFile(const char *filename) {
    // Open file for reading
    File file = SD.open(filename);
    if (!file) {
        Serial.println(F("Failed to read file"));
        return;
    }
 
    // Extract each characters by one by one
    while (file.available()) {
        Serial.print((char) file.read());
    }
    Serial.println();
 
    // Close the file
    file.close();
}
void setup(){
  Serial.begin(115200);
  SPIClass spi = SPIClass(HSPI);
  spi.begin(SCK, MISO, MOSI, CS);

  if (!SD.begin(CS,spi,80000000)) {
    Serial.println("Card Mount Failed");
    return;
  }

  Serial.println(F("SD library initialized"));
  Serial.println(F("Delete original file if exists!"));
  SD.remove(filename);

}

void loop() {

    DynamicJsonDocument doc(1024);
 
    JsonObject obj;
    obj = getJSonFromFile(&doc, filename);
 
    obj[F("millis")] = millis();
 
    JsonArray data;
    // Check if exist the array
    if (!obj.containsKey(F("data"))) {
        Serial.println(F("Not find data array! Crete one!"));
        data = obj.createNestedArray(F("data"));
    } else {
        Serial.println(F("Find data array!"));
        data = obj[F("data")];
    }
 
    // create an object to add to the array
    JsonObject objArrayData = data.createNestedObject();
 
    objArrayData["prevNumOfElem"] = data.size();
    objArrayData["newNumOfElem"] = data.size() + 1;
 
 
    boolean isSaved = saveJSonToAFile(&doc, filename);
 
    if (isSaved){
        Serial.println("File saved!");
    }else{
        Serial.println("Error on save File!");
    }
 
    // Print test file
    Serial.println(F("Print test file..."));
    printFile(filename);
 
    delay(5000);
}
```

**Program output**  
If relevant, include the program output.

Expected output:
It should create a json file with data structure
```
{
  "millis": 10735,
  "data": [
    {
      "prevNumOfElem": 1,
      "newNumOfElem": 2
    },
    {
      "prevNumOfElem": 2,
      "newNumOfElem": 3
    },
    {
      "prevNumOfElem": 3,
      "newNumOfElem": 4
    }
  ]
}
```

Actual output:

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
SD library initialized
Delete original file if exists!
Error opening (or file not exists) /test.jso
Empty json created
Not find data array! Crete one!
Open file in write mode
Filename --> /test.jso
Start write......Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

 Core  1 register dump:
 PC      : 0x40089e81  PS      : 0x00060730  A0      : 0x800d5fb3  A1      : 0x3ffb24a0  
A2      : 0x00000002  A3      : 0xffffffff  A4      : 0x00000004  A5      : 0x00000040  
A6      : 0x00000020  A7      : 0x8003ffff  A8      : 0x000000f0  A9      : 0x0000004f  
A10     : 0x3ffb24a0  A11     : 0x00000001  A12     : 0x00000200  A13     : 0x00000298  
A14     : 0x3ffb2778  A15     : 0x00000022  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000042  LBEG    : 0x400d652c  LEND    : 0x400d6584  LCOUNT  : 0x00000000  

```

HarborDesigns, 2021-11-17:
I have raise suspicion on ArduinoESP32 board library as I am seeing something not normal. I have doubt that SD card CS line is linked to MTDO of JTAG which could trick ESP32 in exception and reset it by itself. I am sharing the [link of the issue which I opened in ArduinoESP32 library](https://github.com/espressif/arduino-esp32/issues/5897#issue-1055583031).

Please share your comments if you have any helpful suggestion.

bblanchon, 2021-11-17:
> Example shown in this link works for me, which does all kinds of operation on SD card including writing and reading char string.
> I am not sure if it is because of the way SPI function is called since I am not using default VSPI pins on my ESP32

This response is ambiguous: do you run the example with VSPI or HSPI pins?

Also, what did you learn from EspExceptionDecoder?

HarborDesigns, 2021-11-17:
I think I am halfway there to implement read and write for JSON. Part of my problem was the GPIO12 and part of it was initiation of HSPI in my code.

I basically still have to power cycle if the GPIO12 function was changed during the operation. For eg. If I have to flash code but I am using GPIO12 for SD_MISO then I have to power cycle. Upon new power cycle pull down the GPIO12 with 10K resistor and keep it low until code is flashed. After code flashing if you want to use SD_MISO you still need to power cycle for second time as the current mode was using GPIO12 as MTDI. During second power cycle you can keep the GPIO12 as floating to start the SD_MISO if you initiate it in early setup stage.

I think in future I need to use a MOSFET or toggle button to change the functionality of GPIO12.

The second part of my problem was my code where I was not initiating and checking my SD card properly in my code. Since I am using HSPI pins and not default VSPI pins of ESP32 I had to check the card mounting status using this code:

> SPIClass spi = SPIClass(HSPI);
> spi.begin(SCK, MISO, MOSI, CS);
> if (!SD.begin(CS,spi,80000000)) { //instead of what library suggest which is 'if(!SD.begin())'
> Serial.println("Card Mount Failed");
> return;
> }

Till now I tested your [example code to configure the file](https://arduinojson.org/v6/example/config/) which successfully worked but the other example file to write [JSON data (by RENZO MISCHIANTI)](https://www.mischianti.org/2020/01/26/manage-json-file-with-arduino-and-esp8266/) to my SD card still causes ESP32 to crash. 

**Can you take a look at other example code and let me know what could have been wrong if I use HSPI pins on ESP32?**

bblanchon, 2021-11-18:
It's difficult to detect a bug when there is so much code and so little information: you should be able to reproduce this crash with just a few lines.
Why don't you start from a working example (or from scratch) and then adapt it to your case?
Alternatively, can you try to serialize to a temporary string and write this string to the SD file?

HarborDesigns, 2021-11-18:
@bblanchon I think the crash could be because I am allocating dynamic memory and other JSON function inside main loop. I did a quick small test. 

When I use this code inside setup then it will work but it crashes ESP32 when I use it in loop function.

```
File file = SD.open(filename, FILE_WRITE);
  if (!file) {
    Serial.println(F("Failed to create file"));
    return;
  }   
DynamicJsonDocument doc(512);
doc["Mode"] = "1";
doc["Timestamp"] = "16:26:43.832";
doc["Debug Message"] = "Normal";
JsonArray CH1_data = doc.createNestedArray("CH1_data");
CH1_data[0] = 123;
CH1_data[1] = 456;
CH1_data[2] = 789;
// Serialize JSON to file
if (serializeJson(doc, file) == 0) {
    Serial.println(F("Failed to write to file"));
  }
 // Close the file
  file.close();
// Print test file
Serial.println(F("Print test file..."));
printFile(filename);
```

1) Do you know why I can't implement same code inside main loop?
2) How do I add more values to 'CH1_data' after the first three has written. I tried using `CH1_data.add(11111);
CH1_data.add(22222);` but it doesn't work. Is there a way to add the next value to the next array item. I expected .add to store the value in next array address (at CH1_data[3] & CH1_data[4]  location )but it still printed old JSON data.
```
Print test file...
{"Mode":"1","Timestamp":"16:26:43.832","Debug Message":"Normal","CH1_data":[123,456,789]}
Print test file...
{"Mode":"1","Timestamp":"16:26:43.832","Debug Message":"Normal","CH1_data":[123,456,789]}
```

3) Can you share one example to allocate dynamic memory on SD card based on memory available on SD card. I have sensor which spits out 1000 samples per second and I have it already stored in an array temporarily. I want to add 1000 values to the same array on SD card

bblanchon, 2021-11-19:
The code above should work inside the main loop; I don't know why it crashes. 

Please see: https://arduinojson.org/v6/how-to/append-to-a-file/

HarborDesigns, 2021-11-19:
I tried using your sample code without SD card as you have in the example link which works but when I use same code in addition to SD card file open and close function it fails and resets the ESP32. 

```
void loop() { 
  File file = SD.open(filename, FILE_WRITE);
  if (!file) {
    Serial.println(F("Failed to create file"));
    return;
  }
  StaticJsonDocument<200> doc;
  doc["key"] = String("value") + counter++;
  serializeJson(doc, Serial);
  file.close();
  delay(1000);
}
```
I get this output in my Serial window:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1420
ho 0 tail 12 room 4
load:0x40078000,len:13540
load:0x40080400,len:3604
entry 0x400805f0
{"key":"value0"}
Stack smashing protect failure!

abort() was called at PC 0x400da934 on core 1

Backtrace:0x40082e99:0x3ffb25f00x4008901d:0x3ffb2610 0x4008d925:0x3ffb2630 0x400da934:0x3ffb26b0 0x400d1889:0x3ffb26d0 0x400d4ff5:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
```

bblanchon, 2021-11-21:
> Stack smashing protect failure!

This seems to be a stack corruption, but looking at this small piece of code, I cannot see a reason for it.  
`StaticJsonDocument<200>` is fairly small, it should not cause a stack overflow, but you can try with a `DynamicJsonDocument`.

Honestly, I don't think this problem comes from ArduinoJson.
Are you sure that your Arduino core for ESP32 is up to date?
Can you share an MCVE that I could run on **any** ESP32?

HarborDesigns, 2021-11-21:
> Are you sure that your Arduino core for ESP32 is up to date?

Yes, I am using ESP32 2.0.1, Arduino IDE 1.8.13 and ArduinoJSON library 6.18.4 

Here is the minimum code which fails on my end. I haven't tried VSPI pins of ESP32 but I assume it will do the same if you modify the pin definition.

Please note that the code works if I use the code inside loop in setup which means it will only calls once. The code inside main loop even works when I don't write the file on SD card instead I can simply print the output on my serial window. The code crashes as it I try to write the file infinite times. For eg. if I try to write file 10 times from the main loop or anywhere else it will work but it won't work if I want to keep on appending my file to store it on SD card.

```
#include <ArduinoJson.h>
#include <SD.h>
#include <FS.h>
#include <SPI.h>
#define SCK  14
#define MISO  12
#define MOSI  13
#define CS  15
SPIClass * hspi = NULL; 
const char *filename = "/testgb.json";  // <- SD library uses 8.3 filenames
int counter;
// Prints the content of a file to the Serial
void printFile(const char *filename) {
  // Open file for reading
  File file = SD.open(filename);
  if (!file) {
    Serial.println(F("Failed to read file"));
    return;
  } 
  // Extract each characters by one by one
  while (file.available()) {
    Serial.print((char)file.read());
  }
  Serial.println();
  // Close the file
  file.close();
}

void setup() {
  // Initialize serial port
  Serial.begin(115200);
  while (!Serial) continue;
  
  SPIClass spi = SPIClass(HSPI);
  spi.begin(SCK, MISO, MOSI, CS);
  if (!SD.begin(CS,spi,80000000)) {
    Serial.println("Card Mount Failed");
    return;
  } 
  SD.remove(filename); 
  File file = SD.open(filename, FILE_WRITE);
  if (!file) {
    Serial.println(F("Failed to create file"));
    return;
  }
  DynamicJsonDocument doc(512);
  doc["key"] = String("value") + counter++;
  serializeJson(doc, Serial);
  file.close();
//if you try to write the above to SD card it will work because it will write on SD card only once
}
void loop() { 
  File file = SD.open(filename, FILE_WRITE);          // comment this line to see it print output in serial window
  if (!file) {                                                              // comment this line to see it print output in serial window
    Serial.println(F("Failed to create file"));             // comment this line to see it print output in serial window
    return;                                                             // comment this line to see it print output in serial window
  }                                                                        // comment this line to see it print output in serial window
  DynamicJsonDocument doc(512);
  doc["key"] = String("value") + counter++;
  serializeJson(doc, Serial);
  file.close();                                                        // comment this line to see it print output in serial window
  delay(1000);
}
```

bblanchon, 2021-11-24:
Unfortunately, I cannot reproduce the crash on my ESP32.
If I could, I would have added [some traces](https://github.com/bblanchon/ArduinoTrace) in the different libraries to locate the exact line that causes the crash.
I would also have tried to revert to the 1.x branch of the ESP32 core.
Since I cannot perform these checks, I invite you to do them yourself.

I don't think I'll be able to help you more with this issue, and I really doubt it comes from ArduinoJson.
Please let me know if you make any progress; otherwise, I'll close the issue after 30 days of inactivity.
Thank you for your understanding.

HarborDesigns, 2021-11-24:
@bblanchon I figured out issue in my code. If you see my sample code which I shared you can I see I declared my SPI Class object in my setup which uses SD.begin. I removed this object to the main loop or local function which helps me call SD.open continuously since now everything is in local scope.

The updated sample code is as follow:

```
#include <ArduinoJson.h>
#include <SD.h>
#include <FS.h>
#include <SPI.h>
#define SCK  14
#define MISO  12
#define MOSI  13
#define CS  15
SPIClass * hspi = NULL; 
const char *filename = "/testgb.json";  // <- SD library uses 8.3 filenames
int counter;

void setup() {
  // Initialize serial port
  Serial.begin(115200);
  while (!Serial) continue;
}
void loop() { 
 SPIClass spi = SPIClass(HSPI);
 spi.begin(SCK, MISO, MOSI, CS);
 if (!SD.begin(CS,spi,80000000)) {
    Serial.println("Card Mount Failed");
    return;
  } 
  File file = SD.open(filename, FILE_WRITE);
  if (!file) {
    Serial.println(F("Failed to create file"));
    return;
  }
  DynamicJsonDocument doc(512);
  doc["key"] = String("value") + counter++;
  serializeJson(doc, Serial);
  file.close();
  Serial.println(F("Print test file..."));
  printFile(filename);   
  delay(1000);
}

// Prints the content of a file to the Serial
void printFile(const char *filename) {
  // Open file for reading
  File file = SD.open(filename);
  if (!file) {
    Serial.println(F("Failed to read file"));
    return;
  } 
  // Extract each characters by one by one
  while (file.available()) {
    Serial.print((char)file.read());
  }
  Serial.println();
  // Close the file
  file.close();
}
```

closed 2021-11-24 by HarborDesigns

bblanchon, 2021-11-24:
Thanks for the update, @HarborDesigns
I'm glad you finally find out the solution.

Anyway, thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: