bblanchon/ArduinoJson #1674: Question about globally allocated SpiRamJsonDocument.

zekageri, 2021-11-16:
**Question**
If i allocate a char array globally in external ram ( defined as SpiRamJsonDocument ), and i use it in a loop like this: 
`JsonObject zones = bigJSON.to<JsonObject>();` does it consume my internal ram as JsonObject or it continues to use the allocated memory space in the external ram?

**Describe the issue**  
 I'm allocating a big char array in external memory. ( 150k ) 

 ```C
    const size_t userProgJSON_Size = 150000;
    SpiRamJsonDocument userProgram(userProgJSON_Size);
```

This is a global char array.
 During setup, i put a big json object ( with multiple objects inside ).
I want to use this json object during runtime, in a separate task within a loop.

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32_WROVER_32 ( 4mb external ram, 16mb flash )
* Core/runtime: ESP32 core for Arduino
* IDE: VSCODE with PlatformIO extension.

**Reproduction**  

 ```C
    struct SpiRamAllocator {
        void* allocate(size_t size) {return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);}
        void deallocate(void* pointer) {heap_caps_free(pointer);}
        void* reallocate(void* ptr, size_t new_size) {return heap_caps_realloc(ptr, new_size, MALLOC_CAP_SPIRAM);}
    };
    using SpiRamJsonDocument = BasicJsonDocument<SpiRamAllocator>;

    const size_t userProgJSON_Size = 150000;
    SpiRamJsonDocument userProgram(userProgJSON_Size);

    static const inline void createUserProgTask(int Stack_Depth, int Core, int Priority){
        BaseType_t result = xTaskCreatePinnedToCore(userProgTask, "userProgTask", Stack_Depth, NULL, Priority, &userProgTask_Handle, Core);
        if(result == pdPASS){
            Serial.printf("User prog task created on core: %d with priority: %d and stack space: %d\n",Core,Priority,Stack_Depth);
        }else{
            Serial.println("User prog Task failed on create.");
        }
    }

    boolean userProgExists = false;

    void setup(){
       getUserProgram();
    }
    static const inline void getUserProgram(){
        File f = LITTLEFS.open(userProg,"r");
        if(!f){
            userProgExists = false;
            Serial.println("Failed to open user program file");
            return;
        }else{
            DeserializationError error = deserializeJson(userProgram, f);
            if (error) {
                Serial.print("User prog deserialization error. Code: ");
                Serial.println(error.c_str());
            }else{
                userProgExists = true;
                createUserProgTask(5000,1,1);
            }
            f.close();
        }
    }
    
    void userProgTask( void * parameter ){
        for ever {
            userProgRun();
            vTaskDelay(5000);
        }
    }

    static const inline void userProgRun(){
        if( userProgExists ) {
          JsonObject zones = userProgram.to<JsonObject>();
          for (JsonPair zone : zones) {
              JsonObject zoneObj = zone.value().to<JsonObject>();
              char zoneName = zoneObj["name"];
              Serial.println(zoneName);
          }
        }
    }

    void loop(){
       vTaskDelay(1000);
    }
```
**My json object looks something like this:**

```JSON
{
{
  "n_kw0btq7m_3u363tqc2pjbss5e4145l": {
    "name": "Test",
    "icon": "hsh-couch",
    "iconColor": "#0099eb",
    "id": "n_kw0btq7m_3u363tqc2pjbss5e4145l",
    "isGroup": false,
    "lights": {},
    "shaders":{},
},
{
  "f_hw0btq4m_3y363bqc2hjbss5e4145l": {
    "name": "Test Group",
    "icon": "hsh-group",
    "iconColor": "#0059eb",
    "id": "l_k760btq7m_4uy63vqc2njbus5e4145l",
    "isGroup": true,
    "lights": {},
    "shaders":{},
}
}
```
### Some left out words: 

**Before this idea i created a new Class for every object in the document at setup. The New keyword does allocate a class in my internal memory, so i ran out from it very quickly and i came out with that idea.  Is it good practices? Does my JsonObject in the loop consumes my internal memory or its just a reference to the array allocated in external memory? Should i deserialize and serialize my json object in every iteration instead? Thank you for your help. Sorry if i left out some important detail.**

zekageri, 2021-11-16:
With that solution, i got that serial message:  `User prog deserialization error. Code: NoMemory`
Maybe because i defined  `SpiRamJsonDocument userProgram(userProgJSON_Size);` globally?

zekageri, 2021-11-16:
### Should i do it like this instead?

```C
boolean userProgExists = false;
const size_t userProgJSON_Size = 150000;
char * userProgram;

static const inline void getUserProgram(){
    userProgram = (char *) ps_malloc (userProgJSON_Size * sizeof (char));
    File f = LITTLEFS.open(userProg,"r");
    if(!f){
        userProgExists = false;
        Serial.println("Failed to open user program file");
        return;
    }else{
        for (int i = 0; i < f.size(); i++){userProgram[i] = (char)f.read();}
        f.close();
    }
}

static const inline void userProgRun(){
    if(userProgExists){
        SpiRamJsonDocument usreZones(userProgJSON_Size);
        DeserializationError error = deserializeJson(usreZones, userProgram);
        if (!error) {
            JsonObject zones = usreZones.to<JsonObject>();
            for (JsonPair zone : zones) {
                JsonObject zoneObj = zone.value().to<JsonObject>();
                char zoneName = zoneObj["name"];
                Serial.println(zoneName);
            }
        }
    }
}

void userProgTask( void * parameter ){
    getUserProgram();
    for ever {
        userProgRun();
        vTaskDelay(5000);
    }
}
```

zekageri, 2021-11-16:
### Or should i open and deserialize and close the file in every iteration like this?

```C
static const inline void userProgRun(){
    if(LITTLEFS.exists(userProg)){
        File f = LITTLEFS.open(userProg,"r");
        if(!f){
            Serial.println("Failed to open user program file for read");
        }else{
            SpiRamJsonDocument usreZones(userProgJSON_Size);
            DeserializationError error = deserializeJson(usreZones, f);
            if (!error) {
                JsonObject zones = usreZones.to<JsonObject>();
                for (JsonPair zone : zones) {
                    JsonObject zoneObj = zone.value().to<JsonObject>();
                    
                    // Handle object variables...
                    char zoneName = zoneObj["name"];
                    Serial.println(zoneName);
                }
            }else{
                Serial.print("User prog deserialization error. Code: ");
                Serial.println(error.c_str());
            }
            f.close();
        }
    }
}

void userProgTask( void * parameter ){
    for ever {
        userProgRun();
        vTaskDelay(5000);
    }
}

```

zekageri, 2021-11-16:
**With my last approach, the documet being deserialized but i can't loop trought my object...**

```C
static const inline void userProgRun(){
    if(LITTLEFS.exists(userProg)){
        File f = LITTLEFS.open(userProg,"r");
        if(!f){
            Serial.println("Failed to open user program file for read");
        }else{
            SpiRamJsonDocument usreZones(userProgJSON_Size);
            DeserializationError error = deserializeJson(usreZones, f);
            if (!error) {
                Serial.println("Successfully deserialized user program file");
                // Program gets here ( prints the whole usreZones object ) but can't loop trought my object. Nothing happens beyond this point.
                JsonObject zones = usreZones.to<JsonObject>();
                for (JsonPair zone : zones) {
                    JsonObject zoneObj = zone.value().to<JsonObject>();

                    // Handle object variables...
                    char zoneName = zoneObj["name"];
                    Serial.println(zoneName);
                }
            }else{
                Serial.print("User prog deserialization error. Code: ");
                Serial.println(error.c_str());
            }
            f.close();
        }
    }else{
        Serial.println("User program file does not exist");
    }
}
```

zekageri, 2021-11-16:
### So i came up with this solution:

```C
void userProgTask( void * parameter ){
    for ever {
        userProgRun();
        vTaskDelay(1);
    }
}

static const inline void userProgRun(){
    if(LITTLEFS.exists(userProg)){
        File f = LITTLEFS.open(userProg,"r");
        if(!f){
            Serial.println("Failed to open user program file for read");
        }else{
            SpiRamJsonDocument userZones(userProgJSON_Size);
            DeserializationError error = deserializeJson(userZones, f);
            if (!error) {
                JsonObject zones = userZones.as<JsonObject>();

                for (JsonPair zoneRef : zones) {
                    const char* zoneID  = zoneRef.key().c_str();
                    JsonObject zone     = zones[zoneID];

                    boolean isGroup = zone["isGroup"].as<boolean>();
                    if(!isGroup){
                        processHandlerOne(zone["handlerOne"]);
                        processHandlerTwo(zone["handlerTwo"]);
                    }
                }
                
            }else{
                Serial.print("User prog deserialization error. Code: ");
                Serial.println(error.c_str());
            }
            f.close();
        }
    }
}
```

That way i must read the file in every iteration. But the problem is that if this JsonObject is updated by a user input ( which is a lot ) i must rewrite the file every update and read it again. This will eventually kill my FLASH.

**How could i make this SpiRamJsonDocument global and update this SpiRamJsonDocument instead of the file?**

zekageri, 2021-11-16:
Okay i ended up with this solution:

```C

const size_t userProgJSON_Size = 150000; // json size for now.
char * zonesBuffer; // Buffer to hold the updates json packet. Allocated in external memory at userProgTask start.
boolean canUpdateZones = false; // Indicator flag if we can update the json
static const inline void userProgRun(JsonObject zones);

// Processing the websocket data packets.
static const inline void processSocketData(){
  if(canProcessSocketData){
      canProcessSocketData = false;
      SpiRamJsonDocument doc(SOCKET_DATA_SIZE);
      DeserializationError error  =  deserializeJson(doc, socketData);
      currSocketBufferIndex       = 0;
      if( !error ){
          JsonObject packet = doc.as<JsonObject>();
          if( packet["type"] == "updateZones" ){
              String packetZones = packet["zones"]; // I can't belive i can not avoid String.
              packetZones.toCharArray(zonesBuffer, packetZones.length()+1);
              canUpdateZones  = true;
          }
     }
  }
} 

void userProgTask( void * parameter ){
    // Allocate memory space for updating the json buffer.
    zonesBuffer = (char *) ps_malloc (userProgJSON_Size * sizeof (char));
    if(LITTLEFS.exists(userProg)){
        File f = LITTLEFS.open(userProg,"r");
        if(!f){
            Serial.println("Failed to open user program file for read");
        }else{
            SpiRamJsonDocument userZones(userProgJSON_Size);
            DeserializationError error = deserializeJson(userZones, f);
            f.close();
            if (!error) {
                for ever {
                    JsonObject zones = userZones.as<JsonObject>();
                    if( canUpdateZones ){
                        canUpdateZones = false;
                        DeserializationError error = deserializeJson(userZones, zonesBuffer);
                        if (!error) {
                            zones = userZones.as<JsonObject>();
                        }
                    }
                    userProgRun(zones);
                    vTaskDelay(1);
                }
            }
        }
    }
}

// Running the actual program infinitely.
int zoneIndex = 0;
static const inline void userProgRun(JsonObject zones){
    JsonObject::iterator zoneRef = zones.begin();
    zoneRef += zoneIndex;
    const char* zoneID  = zoneRef->key().c_str();
    JsonObject zone     = zones[zoneID];

    boolean isGroup = zone["isGroup"].as<boolean>();
    if(!isGroup){
      // Handle components inside the json data.
       processHandlerOne( zone["handlerOne"] );
       processHandlerTwo( zone["handlerTwo"] );
    }
    
    zoneIndex++;
    if( zoneIndex >= zones.size() ){zoneIndex = 0;}
}
```

I still don't know if this is the correct approach. But it works for me for now.

bblanchon, 2021-11-17:
Hi @zekageri,

You cannot declare a global `SpiRamJsonDocument` because it would call `heap_caps_malloc()` before the external RAM is ready to use.

I added a warning in [How to use external RAM on ESP32?](https://arduinojson.org/v6/how-to/use-external-ram-on-esp32/)
I added this scenario in the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/#runtime/deserialization/nomemory/increased-capacity/basic/capacity-0/global)

Best regards,
Benoit

closed 2021-11-17 by bblanchon

zekageri, 2021-11-17:
### Can i create a class, pass a JsonObject deserialized from a file and use it like this?

```C
#define MAX_LIGHTS 20
class lightHandler {
    private:
        JsonObject light;
        void useLightObject(){
            const char* name = light["name"];
            Serial.println(name);
        }
    public:
        lightHandler(JsonObject light){
            lightHandler::light = light;
            useLightObject();
        }
};
lightHandler * lightHandlers[MAX_LIGHTS] EXT_RAM_ATTR;

static const inline void processUserProg(){
    if(LITTLEFS.exists(userProg)){
        File f = LITTLEFS.open(userProg,"r");
        if(!f){
            Serial.println("Failed to open user program file for read");
        }else{
            SpiRamJsonDocument userZones(userProgJSON_Size);
            DeserializationError error = deserializeJson(userZones, f);
            f.close();
            if (!error) {
                createdLights = 0;
                JsonObject zones = userZones.as<JsonObject>();
                for (JsonPair zoneRef : zones) {
                    const char* zoneID  = zoneRef.key().c_str();
                    JsonObject zone     = zones[zoneID];
                    boolean isGroup = zone["isGroup"].as<boolean>();
                    if(!isGroup){
                        JsonObject lights   = zones["lights"].as<JsonObject>();
                        for (JsonPair lightRef : lights) {
                            const char* lightID  = lightRef.key().c_str();
                            lightHandlers[createdLights] = new lightHandler( lights[lightID].as<JsonObject>() );
                            createdLights++;
                        }
                    }
                }
                    Serial.printf("Created %d light",createdLights);
            }
        }
    }
}
```

bblanchon, 2021-11-18:
No, you cannot because a `JsonObject` is a reference, and in this case it would point to a destructed `JsonDocument`.

The recommended practice is to extract the relevant values in dedicated class members.
In your case, it could be a `String name` member in `lightHandler`.
See [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/) for an example.

zekageri, 2021-11-19:
Thank you for all your help. I really appreciate it. I just experimenting with things