bblanchon/ArduinoJson #1676: Using FakeArduino for unit testing with platform io

tickietackie, 2021-11-18:
**Describe the issue**  
I'm trying to do the same as the guy here #548.
Using FakeArduino native to unit test with platform io, where ArduinoJson is deeply integrated.
I already tried the fix from the issue there, but I suppose sth. changed, that makes it not work anymore.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
<details><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at compile time</li><li>The error is not in the list</li></ol></details>

**Environment**  
Here is the environment that I'm using':
* Microconroller: NA
* Core/runtime: NA
* IDE: Platform IO and VS Code

**Further information**  
platform.ini file content:

```
[env:native]
platform = native
test_ignore = embedded
build_flags = -std=gnu++11
lib_deps = 
	ArduinoJson-6.18.5
	ArduinoFake
#lib_archive = false 
```

Compile output:

```
lib/ArduinoJson-6.18.5/src/ArduinoJson/Polyfills/pgmspace_generic.hpp:14:10: error: reinterpret_cast from 'const void' to 'const __FlashStringHelper *' is not allowed
  return reinterpret_cast<T>(pgm_read_ptr(p));
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
lib/ArduinoJson-6.18.5/src/ArduinoJson/Deserialization/DeserializationError.hpp:100:12: note: in instantiation of function template specialization 'ArduinoJson6185_F1::pgm_read<const __FlashStringHelper *>' requested here
    return ARDUINOJSON_READ_STATIC_ARRAY(const __FlashStringHelper*, messages,
           ^
lib/ArduinoJson-6.18.5/src/ArduinoJson/Polyfills/static_array.hpp:20:7: note: expanded from macro 'ARDUINOJSON_READ_STATIC_ARRAY'
      pgm_read<type>(name + index)
```

bblanchon, 2021-11-19:
Hi @tickietackie,

I couldn't find the source of ArduinoFake, but it seems that `pgm_read_ptr()` doesn't return the correct type: it should be `const void*`, but the error message says it's `const void`. 

This is a common mistake in cores that emulate the original `pgmspace.h` header.
It already poped up in #1381, #1442, #1519, #1536, and #1591.
The root of this error was fixed a year ago in arduino/ArduinoCore-API#118, but many clones are still using the old code.
I invite you to open a PR in the ArduinoFake repository.

In the meantime, you can work around this issue by disabling PROGMEM in ArduinoJson by setting [`ARDUINOJSON_ENABLE_PROGMEM`](https://arduinojson.org/v6/api/config/enable_progmem/) to&nbsp;`0`.

Best regards,
Benoit

tickietackie, 2021-11-20:
Thx, for the quick response. The workaround works as supposed.

closed 2021-11-20 by tickietackie

maxgerhardt, 2021-12-04:
For completness' sake: This was addresses in the FakeArduino library (https://github.com/FabioBatSilva/ArduinoFake/pull/21) and so one can reference the latest git version

```ini
[env:native]
platform = native
build_flags = -std=gnu++11
lib_deps =
     https://github.com/FabioBatSilva/ArduinoFake/archive/refs/heads/master.zip
     bblanchon/ArduinoJson @ ^6.18.5
```

and that makes that error go away.