bblanchon/ArduinoJson #1677: const char* array will be empty after deserialization even with read-only input classic mode.

zekageri, 2021-11-19:
### Here is my scenario:

- **IDE :**            VSCODE platformIO
- **Framework:** Arduino
- **CHIP:**           ESP32_WROVER ( 4mb psram 16mb flash )

**I got a big JSON string from websocket. This contains a json string in it's variable:**

```C
const size_t userProgJSON_Size = 150000;
const char * zonesBuffer;

void setup(){
 zonesBuffer = (char *) ps_malloc (userProgJSON_Size * sizeof (char));
}

// this will be called if some data available from socket.
void processSocketData(){
  SpiRamJsonDocument doc(SOCKET_DATA_SIZE);
        DeserializationError error  =  deserializeJson(doc, socketData);
        currSocketBufferIndex       = 0;
        if( !error ){
            JsonObject packet = doc.as<JsonObject>();if( packet["type"] == "updateZones" ){
                zonesBuffer = packet["zones"].as<String>().c_str();
                canUpdateZones  = true;
               // Here i can print zonesBuffer and it will contain my json string
            }
        }
}
```

**When i got the data, i set a flag to update some things from that const char array in the loop.**

```C
void loop(){
    processUpdate();
    saveUserProgram();
}

void processUpdate(){
  if( canUpdateZones ){
        canUpdateZones = false;

        SpiRamJsonDocument updatedZones(userProgJSON_Size);
        DeserializationError error = deserializeJson(updatedZones, (const char *)zonesBuffer);
        if (!error) {
            JsonObject zones = updatedZones.as<JsonObject>();
            for (JsonPair zoneRef : zones) {
                const char* zoneID  = zoneRef.key().c_str();
                JsonObject zone     = zones[zoneID];
                boolean isGroup     = zone["isGroup"].as<boolean>();
                if(!isGroup){
                   // Process the data in here. At this point the zonesBuffer is still there, i can process the data inside it no problem.
                }
            }
         }
            lastUpdateMs    = millis();
            canSaveProgram  = true;
         // Here i can still print zonesBuffer and it will contain my json string as before.
        // I set a flag here to save this json to the file system later if no updates comes meanwhile.
     }
}

static const inline void saveUserProgram(){
    if( ( canSaveProgram && millis() - lastUpdateMs >= updateIntervalMS ) || forceSaveUserProg){
        lastUpdateMs = millis();
        forceSaveUserProg = false;

        // When the program gets here the zonesBuffer is empty and i can no longer access it's content.
        if( zonesBuffer && !zonesBuffer[0] ){
            File f = LITTLEFS.open(userProg, "w");
            if( f ){
                canSaveProgram = false;
                f.print(zonesBuffer);
            }
            f.close();
        }else{
            Serial.println("User prog is empty for some fucking reason...");
        }
    }
}

```

**When the program gets to the save my json const char array ( which is allocated in PSRAM globally ) will be empty.
What is happening in here?**

_UI: I use this technic elsewhere in my program and its working fine, but not here._

bblanchon, 2021-11-20:
Hi @zekageri,

I stopped reading after seeing this line:

```c++
zonesBuffer = packet["zones"].as<String>().c_str();
```

This line is incorrect because it doesn't copy the string, it simply overrides the pointer.

One solution could be this:

```c++
memcpy(zonesBuffer, packet["zones"].c_str(), packet["zones"].size())
```

Best regards,
Benoit

zekageri, 2021-11-20:
Oh. I think I tried this but I got a runtime crash. I will look at it again. Thank you

zekageri, 2021-11-22:
```
src/headers/socket.h:101:53: error: 'ArduinoJson6180_91::enable_if<true, ArduinoJson6180_91::MemberProxy<ArduinoJson6180_91::ObjectRef, const char*> >::type {aka class 
ArduinoJson6180_91::MemberProxy<ArduinoJson6180_91::ObjectRef, const char*>}' has no member named 'c_str'
                 memcpy(zonesBuffer, packet["zones"].c_str(), packet["zones"].size())
```

zekageri, 2021-11-22:
**If i do this:**

`memcpy(zonesBuffer, packet["zones"].as<String>().c_str(), packet["zones"].size());`

I got this:

```
src/headers/socket.h:101:97: error: invalid conversion from 'const void*' to 'void*' [-fpermissive]
                 memcpy(zonesBuffer, packet["zones"].as<String>().c_str(), packet["zones"].size());
```

zekageri, 2021-11-22:
If i define my buffer as a char * it says incomplete input if i try to deserialize it.

```C
const size_t userProgJSON_Size = 150000;
char * zonesBuffer;

void setup(){
    zonesBuffer = (char *) ps_malloc (userProgJSON_Size * sizeof (char));
}

void loop(){
    if( packet["type"] == "updateZones" ){
        memcpy(zonesBuffer, packet["zones"].as<String>().c_str(), packet["zones"].size());
        canUpdateZones  = true;
    }
    
    if( canUpdateZones ){
        canUpdateZones = false;
        SpiRamJsonDocument updatedZones(userProgJSON_Size);
        DeserializationError error = deserializeJson(updatedZones, (const char *)zonesBuffer,userProgJSON_Size);
        if (!error) {
            // Process json....
        }else{
            Serial.println("Zones update error");
            Serial.println(error.c_str());
            // Incomplete input!!
        }
    }
}
```

bblanchon, 2021-11-22:
I'm sorry, this is what I meant:

```c++
memcpy(zonesBuffer, packet["zones"].as<const char*>(), packet["zones"].size());
```

zekageri, 2021-11-22:
> I'm sorry, this is what I meant:
> 
> ```c++
> memcpy(zonesBuffer, packet["zones"].as<const char*>(), packet["zones"].size());
> ```

### Results in a runtime error too.

**I must define my buffer as char* in order for the compiler to accept it.
And when the program gets there i got this error:**

```C
Backtrace: 0x4008c928:0x3ffb50f0 0x400dd099:0x3ffb5100 0x400dd997:0x3ffb5270 0x40090706:0x3ffb52c0
  #0  0x4008c928:0x3ffb50f0 in memcpy at /home/mak/e/p/newlib_old/newlib_xtensa-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:240
  #1  0x400dd099:0x3ffb5100 in processSocketData() at src/headers/socket.h:101      
  #2  0x400dd997:0x3ffb5270 in serverTask(void*) at src/headers/tasks.h:48
  #3  0x40090706:0x3ffb52c0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

**This is how it is:**

```C
const size_t userProgJSON_Size = 150000;
char * zonesBuffer;

static const inline void processSocketData(){
    if(canProcessSocketData){
        canProcessSocketData = false;
        SpiRamJsonDocument doc(SOCKET_DATA_SIZE);
        DeserializationError error  =  deserializeJson(doc, socketData);
        currSocketBufferIndex       = 0;
        if( !error ){
            JsonObject packet = doc.as<JsonObject>();if( packet["type"] == "updateZones" ){
                //zonesBuffer = packet["zones"].as<String>().c_str();
                memcpy(zonesBuffer, packet["zones"].as<const char*>(), packet["zones"].size());
                canUpdateZones  = true;
            }
        }
    }
}
```

bblanchon, 2021-11-24:
In the snippet above, you don't show how `zonesBuffer` is initialized.
I assumed you kept the line call to `ps_malloc` in `setup()`; is it still the case?
If so, please verify that it doesn't return NULL.

Another thing that could explain the crash is a NULL returned by `as<const char*>()`, which happens if `packet["zones"]` is not a string.
You can protect your program against this vulnerability like so:

```c++
JsonVariant zone = packet["zones"];
if (zone.is<const char*>()) {
  memcpy(zonesBuffer, zone.as<const char*>(), zone.size());
  ...
}
```

zekageri, 2021-11-24:
The initialization still happening in the setup(). Actually it happens in a separate task's setup.
With String it is working like a charm, but my current goal's whole point is to avaid Strings as much as possible.

Currently i'm using it like this but ofc iam loooosing ram:

```C
const size_t userProgJSON_Size = 150000;
const char * zonesBuffer;

void setup(){
    zonesBuffer = (char *) ps_malloc (userProgJSON_Size * sizeof (char));
    if( zonesBuffer == NULL ){
        // This line never gets printed. My PSram always initialized at this point.
        Serial.println("No psRam at this point.");
    }
}

if( packet["type"] == "updateZones" ){
    String zonesToUpdate = packet["zones"].as<String>();
    zonesBuffer     = zonesToUpdate.c_str();
    canUpdateZones  = true;
}
```

Can't belive that Strings make my life that easy and yet so destructive.

`packet["zones"]` is a string i'm sure of it. 
It comes from a JS client via websockets ( it is JSON.stringified at client side ) and i can print it on the ESP side without serialization or deserialization.

**Packet looks like this when it is an object at client side:**

```JS
var zonesProg = {
    "zoneONE": {},
    "zoneTWO": {},
};
var zonesBufferPacket = {
    type: "updateZones",
    zones: zonesProg,
};
```

**Sending like this:**

```JS
function updateZones(){
    socket.send( JSON.stringify(zonesBufferPacket) );
}
```

The ESP side is working fine too. I can extract the type from the packet and i got other packet handlers separated by " type " with nested json strings inside it.

zekageri, 2021-11-24:
I was wrong. With the String approach, it is copying it into the buffer but the content of the buffer gone after deserialization.
Iam desperate.

I'm doing it as before ( with String approach ):

```C
const size_t userProgJSON_Size = 150000;
const char * zonesBuffer;

void setup(){
    zonesBuffer = (char *) ps_malloc (userProgJSON_Size * sizeof (char));
    if( zonesBuffer == NULL ){
        // This line never gets printed. My PSram always initialized at this point.
        Serial.println("No psRam at this point.");
    }
}

void loop(){
    socketPacketHandler();
    updateZones();
    saveUserProgram();
}

void socketPacketHandler(){
  if(canProcessSocketData){
        canProcessSocketData = false;
        SpiRamJsonDocument doc(SOCKET_DATA_SIZE);
        DeserializationError error  =  deserializeJson(doc, socketData);
        currSocketBufferIndex       = 0;
        if( !error ){
            JsonObject packet = doc.as<JsonObject>();
            if( packet["type"] == "updateZones" ){
                String zonesToUpdate = packet["zones"].as<String>();
                zonesBuffer     = zonesToUpdate.c_str();
                canUpdateZones  = true;
            }
        }
  }else{
      char errorMsg[128];
      strcpy(errorMsg, "Socket packet parse error: ");
      strcpy(errorMsg, error.c_str());
      errorLog(errorMsg,"socket");
  }
}

static const inline void updateZones(){
    if( canUpdateZones ){
        canUpdateZones = false;
        SpiRamJsonDocument updatedZones(userProgJSON_Size);
        DeserializationError error = deserializeJson(updatedZones, (const char *)zonesBuffer,userProgJSON_Size);
        if (!error) {
            JsonObject zones = updatedZones.as<JsonObject>();
            // Data is good. I can extract it from zones object.

            lastUpdateMs    = millis();
            canSaveProgram  = true;
            Serial.println("Zones and components updated.");
        }else{
            Serial.println("Zones update error");
            Serial.println(error.c_str());
        }
     }
}

static const inline void saveUserProgram(){
    if( ( canSaveProgram && millis() - lastUpdateMs >= updateIntervalMS ) || forceSaveUserProg){
        lastUpdateMs = millis();
        forceSaveUserProg = false;
        if( zonesBuffer && !zonesBuffer[0] ){
            Serial.println("Saving zones...");
            Serial.println(zonesBuffer); // EMPTY IN HERE
            File f = LITTLEFS.open(userProg, "w");
            if( f ){
                canSaveProgram = false;
                f.print(zonesBuffer);
            }
            f.close();
        }else{
            Serial.println("User prog is empty for some fucking reason...");
        }
    }
}
```

When the program gets to the `saveUserProgram()` function, my buffer is empty.

zekageri, 2021-11-26:
### Okay i'm an idiot.

_The `packet["zones"]` was not a string at all. 
It was an object all along._

**I just had to do this:**

```C
if( packet["type"] == "updateZones" ){
    JsonObject zones = packet["zones"];
    serializeJson(zones, zonesBuffer, SOCKET_DATA_SIZE);
    canUpdateZones = true;
}
```

**Thank you for all your help once again. In the end it was your suggestion to check if it is really a char array or not. And i noticed that there was no cat claw surrounding the "string".** 

Problem solved for now. :D

closed 2021-11-26 by zekageri