bblanchon/ArduinoJson #1678: How to use JsonDocuments with std::unique_ptrs?

matth-x, 2021-11-21:
Hi Benoît,

Thank you for all the work on your library, it's really awesome and helps me a lot with my own library (if you're curious, you can have a look: [https://github.com/matth-x/ArduinoOcpp](https://github.com/matth-x/ArduinoOcpp)).

As your design allows to build the JSON documents as a composite, I took full advantage of it. To make the memory management still straightforward despite multiple sub-documents per document, I would like to go on using `unique_ptr`s because of its familiar semantics.

But the following way of using them seems problematic:

```c++
std::unique_ptr<DynamicJsonDocument> doc = factory->makeDoc1_uniquePtr();

//...

if (specialCase) {
    doc = factory->makeDoc2_uniquePtr(); //compiler error
}

```

This gives me the following compiler output:

<pre><code>
~\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h: In instantiation of 'void std::unique_ptr<_Tp, _Dp>::reset(std::unique_ptr<_Tp, _Dp>::pointer) [with _Tp = ArduinoJson6185_91::BasicJsonDocument<ArduinoJson6185_91::DefaultAllocator>; _Dp = std::default_delete<ArduinoJson6185_91::BasicJsonDocument<ArduinoJson6185_91::DefaultAllocator> >; std::unique_ptr<_Tp, _Dp>::pointer = ArduinoJson6185_91::BasicJsonDocument<ArduinoJson6185_91::DefaultAllocator>*]':
~\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:251:7: 
  required from 'std::unique_ptr<_Tp, _Dp>& std::unique_ptr<_Tp, _Dp>::operator=(std::unique_ptr<_Tp, _Dp>&&) [with _Tp = ArduinoJson6185_91::BasicJsonDocument<ArduinoJson6185_91::DefaultAllocator>; _Dp = std::default_delete<ArduinoJson6185_91::BasicJsonDocument<ArduinoJson6185_91::DefaultAllocator> >]'
src/mytest.cpp:102:10:   required from here
~\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:342:6: 
<b>error:** call of overloaded 'swap(ArduinoJson6185_91::BasicJsonDocument<ArduinoJson6185_91::DefaultAllocator>*&, ArduinoJson6185_91::BasicJsonDocument<ArduinoJson6185_91::DefaultAllocator>*&)' is ambiguous</b>
  swap(std::get<0>(_M_t), __p);
      ^
In file included from ~\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_pair.h:59:0,
                 from ~\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\stl_algobase.h:64,
                 from ~\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\deque:60,
                 from src/mytest.h:8,
                 from src/mytest.cpp:5:
~\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits\move.h:176:5: note: 
candidate: void std::swap(_Tp&, _Tp&) [with _Tp = ArduinoJson6185_91::BasicJsonDocument<ArduinoJson6185_91::DefaultAllocator>*] 
     swap(_Tp& __a, _Tp& __b)
     ^
In file included from .pio/libdeps/esp32-development-board/ArduinoJson/src/ArduinoJson/MsgPack/endianess.hpp:8:0,
                 from .pio/libdeps/esp32-development-board/ArduinoJson/src/ArduinoJson/MsgPack/MsgPackDeserializer.hpp:9,       
                 from .pio/libdeps/esp32-development-board/ArduinoJson/src/ArduinoJson.hpp:37,
                 from .pio/libdeps/esp32-development-board/ArduinoJson/src/ArduinoJson.h:9,
                 from src/mytest.h:10,
                 from src/mytest.cpp:5:
.pio/libdeps/esp32-development-board/ArduinoJson/src/ArduinoJson/Polyfills/utility.hpp:11:13: note: candidate: void ArduinoJson6185_91::swap(T&, T&) [with T = ArduinoJson6185_91::BasicJsonDocument<ArduinoJson6185_91::DefaultAllocator>*]
 inline void swap(T& a, T& b) {
             ^
*** [.pio\build\esp32-development-board\src\mytest.cpp.o] Error 1
</code></pre>

The problem seems to be due to the implementation of the `operator=` in the `unique_ptr` triggering a call to `swap` which could apply to both `std::swap` and `ARDUINOJSON_NAMESPACE::swap` as a consequence of [ADL](https://en.cppreference.com/w/cpp/language/adl).

What is the right way of using JsonDocuments with `unique_ptr`s? Have I mistaken something? Thank you in advance for tips!

bblanchon, 2021-11-22:
Hi @matth-x,

I just had a quick glance à ArduinoOcpp and this looks like a fantastic project; I'll add a link on arduinojson.org as soon as I can.

Honestly, I don't see the point of wrapping a `DynamicJsonDocument` in a `std::unique_ptr`, since this mechanics is already implemented in the class itself (it would make sense for a `StaticJsonDocument`, though).
However, this ambiguous call sounds like a bug, and I need to investigate further.

BTW, I think you might be interested in #1343 for creating composite documents.

Best regards,
Benoit

matth-x, 2021-11-23:
Hi Benoît,

Amazing, thank you for fixing it so quickly! And yes, I will follow #1343 and see when there will be updates. 

If you like, I would also write a reference for your website. After having been working with your project for 2 years I have seen more than enough to recommend it.

bblanchon, 2021-11-24:
By "reference", you mean "testimonial", right?
If so, you can write it here, or send me an email, as you prefer.

matth-x, 2021-12-01:
ArduinoJson is a good example for great engineering. If you rely heavily on JSON, then ArduinoJson will not only give you the tools to handle data processing efficiently, but it will also be a framework to build better products upon.

bblanchon, 2022-01-08:
The fix was release in [ArduinoJson 6.19.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.19.0).

commit cd11328e (2021-11-23): Fix `call of overloaded 'swap(...)' is ambiguous` (fixes #1678)
commit 994b7d35 (2021-11-23): Fix `call of overloaded 'swap(...)' is ambiguous` (fixes #1678)
commit 2df1bc7d (2021-11-23): Fix `call of overloaded 'swap(...)' is ambiguous` (fixes #1678)