bblanchon/ArduinoJson #1680: How to compile arduinojson using cmake

925486270, 2021-11-23:
**Describe the issue**  
Error message generated by compiling Cmakelist in arduinojson-6. X \ SRC directory

**Troubleshooter report**  
CMake Error at D:/CMake/share/cmake-3.22/Modules/WriteBasicConfigVersionFile.cmake:43 (message):
  No VERSION specified for WRITE_BASIC_CONFIG_VERSION_FILE()
Call Stack (most recent call first):
  D:/CMake/share/cmake-3.22/Modules/CMakePackageConfigHelpers.cmake:237 (write_basic_config_version_file)
  CMakeLists.txt:46 (write_basic_package_version_file)

**Environment**  
Here is the environment that I'm using':
CMake 3.22.0 Gui

bblanchon, 2021-11-24:
Hi @925486270,

Please see [How to use ArduinoJson with CMake?](https://arduinojson.org/v6/how-to/use-arduinojson-with-cmake/)
If you still get an error, please provide complete instructions to reproduce it.

Best regards,
Benoit

925486270, 2021-11-24:
Thank you very much. I have solved the problem. Thank you for your kindness

closed 2021-11-25 by bblanchon