bblanchon/ArduinoJson #1681: Stack dump on ESP8266 with using DynamicJsonDocument in version 6.18.5

oreillymj, 2021-11-24:
**Describe the bug**  
On the ESP8266, deserializing a String using a DynamicJsonDocument causes a crash on Free at the end of the parsing function in version 6.18.5 ( and possibly 6.18.4)

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
<details><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at run time</li><li>The issue concerns deserialization</li><li><code>deserializeJson()</code> returns <code>Ok</code></li><li>Program crashes after calling <code>deserializeJson()</code></li><li>Program calls neither <code>strcmp()</code> nor <code>strcpy()</code></li><li>Replacing pointer with a <code>String</code> doesn’t solve the issue</li></ol></details>

**Environment**  
Here is the environment that I used:
* Microcontroller: [e.g. ESP8266]
* Core/runtime: [e.g. ESP8266 core for Arduino v3.0.2]
* IDE: [e.g. Arduino IDE 1.8.16]

**Reproduction**  
Here is the JSON being parsed.
{
  "type": "actuation",
  "event": "relay",
  "status": "on",
  "uuid": "71ad8761-c9fe-4d92-9642-973bb869d121",
  "jobtype": 1,
  "epoch_duration": 120000,
  "start_epoch": "1637755140609",
  "end_epoch": "1637755260609"
}

Here is the suggested code from the assistant
StaticJsonDocument<384> doc;

DeserializationError error = deserializeJson(doc, input);

if (error) {
  Serial.print(F("deserializeJson() failed: "));
  Serial.println(error.f_str());
  return;
}

const char* type = doc["type"]; // "actuation"
const char* event = doc["event"]; // "relay"
const char* status = doc["status"]; // "on"
const char* uuid = doc["uuid"]; // "71ad8761-c9fe-4d92-9642-973bb869d121"
int jobtype = doc["jobtype"]; // 1
long epoch_duration = doc["epoch_duration"]; // 120000
const char* start_epoch = doc["start_epoch"]; // "1637755140609"
const char* end_epoch = doc["end_epoch"]; // "1637755260609"

I previously have code for version 5 which I ported to version 6 which starts as follows.

bool parseActuationJSON(String inputjson) {
  logln("parseActuationJSON()->start");

  //const size_t bufferSize = JSON_OBJECT_SIZE(8) + 384;

  // ArduinoJson 6
  const size_t bufferSize = 384; //
  //DynamicJsonDocument root(bufferSize);
  StaticJsonDocument<bufferSize> root;

  logln("Parsing received actuation message " + inputjson);
.......

The rest of this function is unchanged.
Changing the code to parse the JSON from a StaticJsonDocument to DynamicJsonDocument results in a core dump when the function exits.

Here is the top of the decoded exception report which seems to indicate a problem freeing memory.

Decoding stack results
0x40100ac7: umm_free_core(umm_heap_context_t*, void*) at Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 549
0x40217940: HardwareSerial::write(unsigned char const*, unsigned int) at Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/HardwareSerial.h line 193
0x40100ca6: free(void*) at Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 595
0x40206944: ArduinoJson6185_91::BasicJsonDocument ::freePool() at \Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/DynamicJsonDocument.hpp line 19
0x4020ad42: parseActuationJSON(String) at \Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp line 318

bblanchon, 2021-11-25:
Hi,

The behavior and the stack trace indicate that the destructor of `DynamicJsonDocument` calls `free()` with an invalid pointer. Yet, all the tests and sanitizers are passing, and no one complained about this.

I think this could be a stack corruption.
Do you store any big variables on the stack?

You elided many parts that could cause this crash.
Can you provide an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example)?

Best regards,
Benoit

oreillymj, 2021-11-25:
Changing between DynamicJsonDocument and StaticJsonDocument in the code below re-creates the crash I see on an ESP8266-01

#include <ArduinoJson.h>

#define DEBUG_MODE 1
#if DEBUG_MODE == 1
#define logln(x) Serial.println(x)
#define log(x) Serial.print(x)
#else
#define logln(x)
#define log(x)
#endif

unsigned long epoch_delay_millis = 0;
unsigned long epoch_delay_start_millis = 0;
String str_job_type = "unknown";
String epoch_start_time = "0";
String epoch_end_time = "0";
String str_jobuuid = "none";
long loop_count=0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);  //Start serial connection
}

void loop() {
  // put your main code here, to run repeatedly:
  String str_JSON = "{\"type\":\"actuation\",\"event\":\"relay\",\"status\":\"on\",\"uuid\":\"71ad8761-c9fe-4d92-9642-973bb869d121\",\"jobtype\": 1,\"epoch_duration\": 120000,\"start_epoch\":\"1637755140609\",\"end_epoch\":\"1637755260609\"}";
  loop_count = (loop_count + 1) % 10; 
  if (loop_count==0){
    parseActuationJSON(str_JSON);
  }
  yield();
  delay(500);
  yield();

  
}

bool parseActuationJSON(String inputjson) {
  logln("parseActuationJSON()->start");

  //const size_t bufferSize = JSON_OBJECT_SIZE(8) + 384; // ArduinoJson 5

  // ArduinoJson 6
  const size_t bufferSize = 384; //
  //DynamicJsonDocument root(bufferSize);
  StaticJsonDocument<bufferSize> root;

  logln("Parsing received actuation message " + inputjson);

  // ArduinoJson 6
  DeserializationError parse_error = deserializeJson(root, inputjson.c_str());

  if (parse_error) {
    logln("JSON Parsing failed with error " + String(parse_error.c_str()));
    return false;
  }

  bool found_keys = true;

  if (root.containsKey("type")) {
    found_keys = found_keys & true;
  } else {
    found_keys = false;
  }

  if (root.containsKey("event")) {
    found_keys = found_keys & true;
[... 127 lines cut ...]
      epoch_start_time = epoch_start_buffer;
      epoch_end_time = epoch_end_buffer;
      //Reset the counter when a new job arrives so back to back jobs are not affected.
      epoch_delay_start_millis = currentMillis;
      logln("parseActuationJSON->Current millis " + String(currentMillis));
      logln("parseActuationJSON->Updated actuation delay epoch start " + String(epoch_delay_start_millis));
      logln("parseActuationJSON->Actuation delay length " + String(epoch_delay_millis));
      //turnOnRelay();
      //relayack(1, str_devicemagic);
      yield();
      logln("parseActuationJSON->sent relay ON MQTT ACK");
    }else{
    //if (String(status_buffer) == "off") {
      epoch_start_time = "0";
      epoch_end_time = "0";
      //turnOffRelay();
      //relayack(0, str_devicemagic);
      yield();
      logln("parseActuationJSON->sent relay OFF MQTT ACK");
    }
  }

  
  logln("parseActuationJSON()->end");
}  //parseActuationJSON

bblanchon, 2021-11-26:
This program doesn't crash my ESP8266.
However, I did notice some big variables here:

```c++
char type_buffer[128];            // Max string size +1
char event_buffer[128];           // Max string size +1
char status_buffer[128];          // Max string size +1
char epoch_start_buffer[24];      // Max string size +1
char epoch_end_buffer[24];        // Max string size +1
char uuid_buffer[128];            // Max string size +1
```

Try to move them out of the stack; for example, move them to the global scope, or use `String`s.

oreillymj, 2021-11-26:
It definitely crashes once I change it from a StaticDocument to DynamicDocument. It doesn't crash as listed above.

````C++
  //DynamicJsonDocument root(bufferSize);
  StaticJsonDocument<bufferSize> root;
````

The variables are large I suppose but required to get the const char*  converted to Strings
I'd rather do this 
````C++
  String type = root["type"]; 
````

bblanchon, 2021-11-28:
I believe you.
The crash isn't systematic, which points again to a stack corruption.
Did you try to remove (or simply comment) the large buffer as I suggested?

oreillymj, 2021-11-29:
ok, so I adjusted the buffer sizes as suggested. But while testing the change and only wanted to see the out from the loop 4-5 times max to validate the behavior, I saw something odd. The function was looping inside itself.

I discovered this was because I had no final **return** statement in my code at the end of the function. It's bizarre that the IDE or compiling didn't flag this as an error, but it was obviously corrupting the return address on the stack.

Adding or removing
````C++
return true;
````
from the end of the code is enough to trigger the issue.

Looking through my entire codebase, this is the only function with a return type that was missing a final return value.
This is such a subtle bug, I likely would never have come across it without using your library.

closed 2021-11-29 by oreillymj

bblanchon, 2021-11-30:
Indeed, I had to insert a `return` statement to compile the code; I assumed you redacted some parts you didn't want to share.
However, it doesn't explain the "looping inside itself" behavior.
I guess you saw the logs of an older build since the new one couldn't complete.

Anyway, thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the library :wink:
Good luck with your project, @oreillymj!