bblanchon/ArduinoJson #1682: Convert JSONL to JSON for Datalogger Application

HarborDesigns, 2021-11-28:
**Issue**  
I am using ArduinoJson library to store my data on SD card as Datalogger. In my application I need to quickly write an array which consist of 1000 samples of data to my SD card(I have shown 10 samples in this example for simplicity). As per your recommendation on ['How to append file'](https://arduinojson.org/v6/how-to/append-to-a-file/), I am using JSONL. After 10 minutes of data collection in JSONL format, I need to convert JSONL to JSON array after 10 minutes so that my server can accept the data format. My goal is to take the JSONL data and convert it to JSON document which consist of big array of different objects separated by comma for eg. [{Dataset1},{Dataset2},...{DatasetN}]. I can't do this array structure because of buffering and parsing of JSON data to append the file which is time consuming.
**JSON L Data Structure File:**
```
[
  {
    "Mode": "1",
    "Timestamp": 285,
    "Debug Message": "Normal",
    "CH2_data": [
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418
    ]
  }
]
[
  {
    "Mode": "2",
    "Timestamp": 285,
    "Debug Message": "Normal",
    "CH2_data": [
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418
    ]
  }
]
```
**The above sample data structure needs to be converted to one big JSON array consisting of all the data objects collected in 10 mins and which are separated by comma.**

```
[
  {
    "Mode": "1",
    "Timestamp": 285,
    "Debug Message": "Normal",
    "CH2_data": [
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418
    ]
  }
,
  {
    "Mode": "2",
    "Timestamp": 285,
    "Debug Message": "Normal",
    "CH2_data": [
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418,
      416.6666418
    ]
  }
]
```

**Questions:**
1) Can you give an example to convert JSONL to JSON array.
2) Can you give me an example to print the deserialized JSON document. I am getting overloading error from my compiler. I am not sure if my syntax is correct.  

`'HardwareSerial::print(<unresolved overloaded function type>)'`

The error is pointed to this line where I am trying to print the JSON objects:
`Serial.print(doc_0["Mode"].as<const char*>);`

**My Deserialization code is as follows:**

```
  File file = SD.open(filename,FILE_READ); //open JSONL document to read 
  if (!file) {
    Serial.println(F("Failed to read file"));
    return;
  } 
  
  for(int a=0; a<4;a++){
    DynamicJsonDocument doc(28600);
    DeserializationError error = deserializeJson(doc, file);
    if (error) {
      Serial.print("deserializeJson() failed: ");
      Serial.println(error.c_str());
      return;
    }

    JsonObject doc_0 = doc[0];
    const char* doc_0_Mode = doc_0["Mode"]; // "2"
    unsigned long doc_0_Timestamp = doc_0["Timestamp"]; // 123
    const char* doc_0_Debug_Message = doc_0["Debug Message"]; // "Normal"
  JsonArray CH2_data = doc_0["CH2_data"];
  for (int b=0;b<4;b++){
    double CH2_data_0 = CH2_data[0]; 
    double CH2_data_1 = CH2_data[1]; 
    double CH2_data_2 = CH2_data[2]; 
    double CH2_data_3 = CH2_data[3]; 
    double CH2_data_4 = CH2_data[4];
    double CH2_data_5 = CH2_data[5];
    double CH2_data_6 = CH2_data[6]; 
    double CH2_data_7 = CH2_data[7]; 
    double CH2_data_8 = CH2_data[8];     
    double CH2_data_9 = CH2_data[9]; 
    }
  Serial.print(doc_0["Mode"].as<const char*>);
  Serial.print(" ");
  Serial.println(doc_0["Timestamp"].as<unsigned long>);
  Serial.print(" ");
  Serial.println(doc_0["Debug Message"].as<const char*>);
  Serial.print(" ");
  Serial.println(doc_0["CH2_data"].as<double>);
  
  }
  file.close();

```

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32
* Core/runtime: ESP32core for Arduino v2.0.1
* IDE: Arduino IDE 1.8.13
* ArduinoJSON library 6.18.4

bblanchon, 2021-11-29:
Hi @HarborDesigns,

To send the JSON data to the server, I recommend that you don't try to deserialize the file at all.
Instead, read the file line by line and insert a comma after each line except the last.

Assuming your JSONL file looks like this:

```text
{"Mode":"1","Timestamp":285,"Debug Message":"Normal","CH2_data":[416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418]}
{"Mode":"2","Timestamp":285,"Debug Message":"Normal","CH2_data":[416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418]}
```

You can "manually" recreate the array like that:

```json
[
{"Mode":"1","Timestamp":285,"Debug Message":"Normal","CH2_data":[416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418]},
{"Mode":"2","Timestamp":285,"Debug Message":"Normal","CH2_data":[416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418,416.6666418]}
]
```

You could implement this like so (not tested):

```c++
File file = SD.open(filename,FILE_READ);

client.println("[");

for(int line=0; line<4; line++) {
  // copy every character until the line break
  while (file.peek() != '\n')
    client.write(file.read()); 

  // skip line break
  while (isspace(file.peek())
    file.read();

  // append comma, except for last line
  if (line<3)
    client.println(",");
}

client.prinln("]");

file.close()
```

There are probably better ways to implement this functionality, but I think you get the idea.

Also, most implementations of `File` are very slow where we read or write one byte at a time like this, so you may need to use a `BufferedStream` (see [StreamUtils](https://github.com/bblanchon/ArduinoStreamUtils)).

Best regards,
Benoit

bblanchon, 2021-11-29:
PS: regarding the compilation issue, you simply forgot to add the parentheses to call `as<T>()`.

HarborDesigns, 2021-11-30:
@bblanchon 
Thank you for your response. The suggestions were helpful and I was able to move forward in my code but I still need some help.

I am now experimenting with Stream buffer. I was able to implement Deserialization & Serialization of my file and print it in my serial window for testing. I am getting runtime error from my IDE which is `deserializeJson() failed: IncompleteInput`. I can run everything fine with` ReadLoggingStream` and `WriteLoggingStream` but not when I use `ReadBufferingStream` and `WriteBufferingStream`. 

Questions:
1) Do you think my buffer length is not enough based on ArduinoJSON Assistant I need 3072 KB of dynamic memory. I am using double of that for individual read and write buffer cycle
2) Do I still need to allocate Dynamic memory for the JSON `DynamicJsonDocument doc(28600);`
3) Do you think while appending or adding a new data set object I need to use object name as 'item' instead of 'item1' 
  ` JsonObject item1 = doc.createNestedObject();`

Please see my snippet of code:
```
File file = SD.open(filename,FILE_READ);
  if (!file) {
    Serial.println(F("Failed to read file"));
    return;
  } 
  //ReadBufferingStream bufferedReadFile{file, 64};
  ReadLoggingStream bufferedReadFile(file, Serial);
  bufferedReadFile.setTimeout(10000);
  DynamicJsonDocument doc(28600);
  DeserializationError error = deserializeJson(doc, bufferedReadFile);
  if (error) {
    Serial.print("deserializeJson() failed: ");
    Serial.println(error.c_str());
    return;
  }
  for (JsonObject item : doc.as<JsonArray>()) {
    const char* Mode = item["Mode"]; // "1", "2", "3", "4"
    int Timestamp = item["Timestamp"]; // 285, 285, 285, 285
    const char* Debug_Message = item["Debug Message"]; // "Normal", "Normal", "Normal", "Normal"
    JsonArray CH2_data = item["CH2_data"];
    double CH2_data_0 = CH2_data[0]; // 416.6666418, 416.6666418, 416.6666418, 416.6666418
    double CH2_data_1 = CH2_data[1]; // 416.6666418, 416.6666418, 416.6666418, 416.6666418
    double CH2_data_2 = CH2_data[2]; // 416.6666418, 416.6666418, 416.6666418, 416.6666418
    double CH2_data_3 = CH2_data[3]; // 416.6666418, 416.6666418, 416.6666418, 416.6666418
    double CH2_data_4 = CH2_data[4]; // 416.6666418, 416.6666418, 416.6666418, 416.6666418
    double CH2_data_5 = CH2_data[5]; // 416.6666418, 416.6666418, 416.6666418, 416.6666418
    double CH2_data_6 = CH2_data[6]; // 416.6666418, 416.6666418, 416.6666418, 416.6666418
    double CH2_data_7 = CH2_data[7]; // 416.6666418, 416.6666418, 416.6666418, 416.6666418
    double CH2_data_8 = CH2_data[8]; // 416.6666418, 416.6666418, 416.6666418, 416.6666418
    double CH2_data_9 = CH2_data[9]; // 416.6666418, 416.6666418, 416.6666418, 416.6666418
    }
  file.close();
  Serial.println();
  
  JsonObject item1 = doc.createNestedObject();
  item1["Mode"] = "1";
  unsigned long time_stamp=millis();
  item1["Timestamp"] = time_stamp;
  item1["Debug Message"] = "Normal";
 
  JsonArray doc_0_CH2_data = item1.createNestedArray("CH2_data");
  for (int i=0;i< Total_Number_of_Samples;i++){
    doc_0_CH2_data[i]=converted_voltage_Pass[i];    
  }
   
  file = SD.open(filename, FILE_WRITE);
  //WriteBufferingStream bufferedWriteFile{file, 64};
  WriteLoggingStream bufferedWriteFile(file, Serial);
  bufferedWriteFile.setTimeout(10000);
  if (!file) {
    Serial.println(F("Failed to write file"));
    return;
  } 
  if (serializeJsonPretty(doc, bufferedWriteFile) == 0) {
      Serial.println(F("Failed to write to file"));
    }
  file.close(); 
```

bblanchon, 2021-12-01:
@HarborDesigns, I think you just forgot to call `bufferedWriteFile.flush()` before `file.close()`.

closed 2021-12-14 by bblanchon