bblanchon/ArduinoJson #1683: Deserialization of multi dimensional array

maxnowack, 2021-11-30:
I'm trying to deserialize a multi dimensional array, but it seems that the size of the array is always `0`. Can someone give me a quick example, how this can be done?
My json looks like the following:
````json
{
  "mapping": [
    [4, 1],
    [2, 1],
    [0, 0],
    [1, 0],
    [5, 0],
    [3, 0]
  ]
}
````

This is my code so far:

````c
JsonArray mappingArray = doc["mapping"].as<JsonArray>();
byte mapping[mappingArray.size()][2];
int i = 0;
for (JsonVariant value : mappingArray) {
	  mapping[i][0] = value[0];
	  mapping[i][1] = value[1];
	  i += 1;
}
````

bblanchon, 2021-11-30:
Hi @maxnowack,

Your program should work fine as you can see on [this demo](https://wandbox.org/permlink/8PFIcUNqLLdLEBtt).

Please share [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/)'s report.

Best regards,
Benoit

maxnowack, 2021-12-03:
Yes, you're right. The issue was that the input was invalid json. Thank you!

closed 2021-12-03 by maxnowack