bblanchon/ArduinoJson #1684: Boolean serialization is not showing true. Major bug

robocircuits, 2021-12-07:
I was using it to upload devices on state to internet. when the device is off it shows {"on":false} but when device is on is says {"on":}
**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  

**Environment**  
Here is the environment that I used:
* Microcontroller:  ESP8266
* Core/runtime: ESP8266 core for Arduino v3.0.2
* IDE: Arduino IDE 1.8.16

void button1()
{
  delayMicroseconds(100000);
  if(digitalRead(in1)==HIGH)                                     
  {
    buttonState1 =! buttonState1;
    Serial.println(buttonState1);
    digitalWrite(relay1,buttonState1);
    delayMicroseconds(150000);
    EEPROM.write(101,buttonState1);
    EEPROM.commit();
  }
}

if(buttonState1 != prevState1)
  {
    json.clear();
    doc["on"] = buttonState1;
    doc["s"] = "state";
    serializeJson(doc,json);
    publishDelegateTelemetry(delegateDeviceId[0],json);
    prevState1 = buttonState1;
    Serial.println(delegateDeviceId[0]);
    Serial.println(json);
  }

**output**    
{"on":,"s":"state"}
1
SA0013A
{"on":,"s":"state"}
0
SA0013A
{"on":false,"s":"state"}
1
SA0013A
{"on":,"s":"state"}
0
SA0013A
{"on":false,"s":"state"}
1
SA0013A
{"on":,"s":"state"}

bblanchon, 2021-12-07:
Hi @robocircuits,

This "major bug" title raised my attention, congratulations!

However, boolean serialization works fine as you can see in this online demo:   ​
https://wandbox.org/permlink/TDby7dX3DPzNzGpr

Can you check the type of `buttonState1`?
If you still believe there is a bug, please provide an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example).

Best regards,
Benoit

bblanchon, 2021-12-07:
PS: also please run the Troubleshooter and paste the report.

robocircuits, 2021-12-07:
I ran the troubleshooter but it didn't help

buttonState1 variable is a boolean type variable.

its urgent because it was working for me fine. but now code is same still its not working.

robocircuits, 2021-12-07:
[image]

bblanchon, 2021-12-09:
> I ran the troubleshooter but it didn't help

I know, but I need to see which choices you made so I can add relevant information at the right place.

Now, can you write an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example)?

closed 2021-12-19 by bblanchon