bblanchon/ArduinoJson #1686: Migrating from Version 5 to 6 I get InvalidInput on deserializeJson

juancgalvez, 2021-12-16:
I used to transmit via MQTT codes to me IR Remote micro controller and parse them without any issues. After upgrading to version 6 I have problems while parsing the JSON message.

This message used to work;

`{"protocol":"SAMSUNG","data": 0xE0E0E01F}`

That message used to work with version 5. But, with version 6 I get InvalidInput error on deserializeJson. I have to enclose the hex value inside quotes to make it work. Something like this:

`{"protocol":"SAMSUNG","data": "0xE0E0E01F"}`

 In JSON format, numeric values don't necessarily must be enclosed inside quotes.

bblanchon, 2021-12-16:
Hi @juancgalvez,

The JSON format doesn't support hexadecimal notation.

ArduinoJson 5 was less picky and didn't impose quotes around strings.
If anything, it's a bug in version 5 but not in version 6.

Best regards,
Benoit

juancgalvez, 2021-12-16:
You are right. I fixed he issue by including values in quotes.

Thank you and excuse me for reporting a false bug.

closed 2021-12-16 by juancgalvez

bblanchon, 2021-12-17:
You're welcome, @juancgalvez.
Thank you for using ArduinoJson!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: