bblanchon/ArduinoJson #1689: Tutorial to install and use arduino JSON on Atmel Studio For AVR

parthbhat13, 2021-12-18:
hi,

while surfing all the issues of older version and moreover reading all the needed document, unfortunately there was no straightforward explanation on how to setup and install the arduinoJson on atmel studio, many have asked the same question with different sort of errors all around, but there was not a single one stop solution where a simple written tutorial or anything that sort is available for proper error and warning free installation of the arduinoJson. 
i thought to do the writeup and post it after i was confident with it will work, but unfortunately tried all the versions, V5.11, V6.X but none of then would compile even after following all the threads and issues on the repo. 

so please some guidance would be really helpful. 

as i have posted this under the feature request and not the issue, i will still post the basic issue which i am facing with the ArduinoJson6.x with atmel studio 

`
cannot specify -o with -c, -S or -E with multiple files	jsonMain	avr-g++.exe	0
`

after searching more it seems like the makefile is not able to handle the multiple file types maybe? idk kinda lost here with the same. 
and yes i did find only one link which explained how to install for atmel studio but i feel its incomplete and not a lot of steps are mentioned [https://arduinojson.org/v5/faq/atmel-studio/](url)

finally, i will be really more than happy to do a quick writeup or even make a video for the community , with the installation instructions, prior it even works for me. 

cheers and thanks for all the efforts you take to make it work.

bblanchon, 2021-12-18:
Hi @parthbhat13,

The easiest way to use ArduinoJson with Atmel Studio is to download the single header distribution (e.g., [ArduinoJson-v6.18.5.h](https://github.com/bblanchon/ArduinoJson/releases/download/v6.18.5/ArduinoJson-v6.18.5.h)) and save it in your project folder among the source files.
There is no need to compile the library itself since it's a [header-only library](https://en.wikipedia.org/wiki/Header-only).

Best regards,
Benoit

parthbhat13, 2021-12-19:
Hey @bblanchon ,

Thank you very much for the very quick reply, honestly I was doing things wrong by adding all the source files to the code and then trying to make it work. I will make sure to read the docs properly next time, and here is a short unscripted [video ](https://youtu.be/tkUD7i2D8nU) I made for others to know on how the installation can be done and used. hope it helps others :)

cheers!
Parth Temkar

closed 2021-12-19 by parthbhat13