bblanchon/ArduinoJson #1691: DynamicJsonDocument serializeJson size problem

Mesihas, 2021-12-21:
**Describe the issue**  
DynamicJsonDocument serializeJson bigger than 4096 bytes just crashes the ESP32

**Error**  
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask)  

**Environment**  
ESP32
IDE: [Arduino IDE 1.8.16]
ArduinoJson 6

I am building the message at runtime and I was able to serialize using `StaticJsonDocument<2048> docItems `with no problem.
Then as the Json message I am building grew in size I started using DynamicJsonDocument `docItems(4096)`  which works fine but I need at least 16384 bytes and any size bigger than 4096 bytes just crash the ESP32.

ESP.getMaxAllocHeap()
113792
ESP.getFreeHeap()
217640

At the same time I am using another ` DynamicJsonDocument docMessage(20000); ` for Deserialization which works perfect.

bblanchon, 2021-12-21:
@Mesihas, please share the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/)'s report.

Mesihas, 2021-12-21:
Please try with a String instead of a char[]

and... It worked.  Sorry that I didn't know the  ArduinoJson Troubleshooter. 

Is this  char[] size limit related to Arduino environment?

THANKS

bblanchon, 2021-12-21:
You're welcome, @Mesihas!

By default, the size of the stack is limited to 8KB on ESP32, so `char output[10000]` is clearly too big.
If you don't know what I'm talking about, I recommend that you read the first chapter of my book [Mastering ArduinoJson](https://arduinojson.org/book/).

Best regards,
Benoit.

closed 2021-12-21 by bblanchon

Mesihas, 2021-12-21:
will do. Thanks