bblanchon/ArduinoJson #1692: "Arduino.h" is leaked everywhere that includes "ArduinoJson.h", even if all Arduino features are disabled

paulocsanz, 2021-12-23:
**Is your feature request related to a problem? Please describe.**  

We have a lower level layer that implements drivers for each device we support, `esp8266/Arduino` is one of these backends. When using this backend `ARDUINO` is always defined, but we do not use any `Arduino` types with `ArduinoJson` (we only use `std::string_view`).

The problem is that the higher level layer uses `ArduinoJson` and gets a `#include "Arduino.h"` with it. Causing conflicts in our platform agnostic code.

We use PlatformIO so we are kinda stuck in its ways of compiling dependencies, we don't want to fork the pio library, and it seems impractical using this git as the source while compiling the lib separately, without the flag, to then statically link it together with our code. But we are open to ideas.

**Describe the solution you'd like**  

A simple way to remove the `#include "Arduino.h"` from [/src/ArduinoJson/Configuration.hpp#L119](https://github.com/bblanchon/ArduinoJson/blob/6.x/src/ArduinoJson/Configuration.hpp#L119) without having to disable all Arduino features from the system. In a way that doesn't depend on a `undef`.

Maybe if any of the three Arduino features are disabled it could automatically ignore the include. Or adding an extra define, like DISABLE_ARDUINO (not necessarily the best idea). Also I had a bit of a hard time finding the include and the features that could be disabled. I wasn't able to find in a public doc, is there a link I missed?

I am willing to provide a PR.

bblanchon, 2021-12-25:
Hi @paulocsanz,

Thank you for reporting this issue; I didn't realize people could have `ARDUINO` defined without actually using any `Arduino.h`'s features.

For the record, this `#include` was added to address #1070 and #1071, where the user complained about the opposite issue.

I'll look at your PR in a moment.

Best regards,
Benoit

closed 2021-12-27 by bblanchon

bblanchon, 2022-01-08:
The fix was release in [ArduinoJson 6.19.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.19.0).

commit ec7ffa20 (2021-12-27): Avoid `Arduino.h` when all its features are disabled (fixes #1692)