bblanchon/ArduinoJson #1694: Incorrect numbers 1,2,4,5

radacina, 2021-12-24:
Well, this is unexpected:
ESP8266, ArduinoJson 6.18.5 on Platformio

```
DynamicJsonDocument aDoc(2048);
aDoc["itv"]="0123456789";
serializeJsonPretty(aDoc, Serial);
```
gives 
[image]

I tried StaticJsonDocument, ARDUINOJSON_DECODE_UNICODE 0, same result.
I'm new at this, am I missing something?

Troubleshooting report:
<details><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at run time</li><li>The issue concerns serialization</li><li>Output contains garbage</li><li>Program doesn’t call <code>deserializeJson()</code></li><li>Program calls <code>serializeJson(const JsonDocument&amp;, ...)</code></li><li>Program uses <code>StaticJsonDocument</code></li><li>Switching to <code>DynamicJsonDocument</code> doesn’t fixes the issue</li><li>Program uses <code>String</code></li><li>Removing <code>.c_str()</code> doesn’t fix the issue</li></ol></details>

radacina, 2021-12-24:
On the other hand, it looks like aDoc["itv"] still contains the good value so I presume something doesn't work as supposed in the serialization functions.

bblanchon, 2021-12-25:
Hi @radacina,

I don't think the library is to blame (see [this online demo](https://wandbox.org/permlink/pQ4uo5gmwSbD87J8))

I seems that you have a problem with your serial port.
Please try the following:

- [ ] reduce the baud rate
- [ ] serialize to a `String` then write the `String` to `Serial`

Can you notice any improvement?

Best regards,
Benoit

radacina, 2021-12-25:
After some trial and error, I found the culprit but I'm not sure I understand what's happening, how can the library be affected by this:
```
String sensorId="TrucMachin"+String(ESP.getChipId());
char * buff="";
...
//THIS: 
sensorId.toCharArray(buff,sensorId.length());
...
serializeJson(aDoc,Serial);
```
Changed it around a little in order not to use the toCharArray part, all is good now. 
Still, some question marks remain.

bblanchon, 2021-12-26:
Hi @radacina,

```c++
char * buff="";
```

This statement is incorrect and should raise a compiler warning.
It doesn't initialize a string as you might think but create a pointer to a string literal.
When the next line copies the characters of the string, it overrides variables in the text segment.

If you're not familiar with C++ memory management, I recommend you stick to `char[]` and `String`.
If you want to learn more, I invite you to read the first chapter of [my book](https://arduinojson.org/book/).

Best regards,
Benoit

closed 2021-12-26 by bblanchon