bblanchon/ArduinoJson #1696: SIM800 support

EliasA97, 2022-01-04:
Hello, first of all I want to tell you that your library works flawlessly and thank you for your work.
My problem is that I can't find any way to write to a stream using a gsm module like the SIM800L. I can only use this library in order to make a String and then, send the string. This consumes some memory on the arduino mega. Is it possible to make an example using this module?

bblanchon, 2022-01-05:
Hi @EliasA97,

Writing the JSON document to a temporary buffer is not necessarily a bad idea.
If you want to avoid the heap, replace the `String` with a char array.

If you do want to avoid the temporary buffer, you can write directly to the corresponding UART.
For example, if the SIM800L is connected to `Serial1`, you can write `serializeJson(doc, Serial1);`.

Best regards,
Benoit

EliasA97, 2022-01-05:
Hello Benoit,

Unfortunatelly I'm using a library with the SIM800 so I can't just Serial1.write as you suggest. The thing after my tests I don't get memory problems at this moment using String but I also found something weird.
The SIM800 library expects from me a json in a char format. By now what i did is converting the String to a char usign the .toCharArray command. The sketch consumed about 9% of the Mega's memory. When I replaced the String with a char, then the consumed memory got up to 15%. I don't know if it's good or bad, but I'll leave it that way, in order to get my code simpler.

Anyway, thank's for the help!

bblanchon, 2022-01-07:
Hi @EliasA97,

The char array is impacting the memory report at the end of the compilation phase because you declared it at the global scope.
Instead, you should move the declaration just above the call to `deserializeJson()`.
Indeed, a global variable consumes RAM for the whole lifetime of the program, whereas a local variable is only consuming RAM  only while the execution pointer is in the variable's scope.

Best regards,
Benoit

closed 2022-02-06 by bblanchon