bblanchon/ArduinoJson #1697: Gettting Null values from JsonDocument

VishBK, 2022-01-10:
This is my code:
```
    Serial.println("Parsing");
    char input[] = "{\"recenttracks\":{\"track\":[{\"artist\":{\"url\":\"https://www.last.fm/music/Daft+Punk\",\"name\":\"Daft Punk\",\"image\":[{\"size\":\"small\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/34s/2a96cbd8b46e442fc41c2b86b821562f.png\"},{\"size\":\"medium\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/64s/2a96cbd8b46e442fc41c2b86b821562f.png\"},{\"size\":\"large\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/174s/2a96cbd8b46e442fc41c2b86b821562f.png\"},{\"size\":\"extralarge\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/300x300/2a96cbd8b46e442fc41c2b86b821562f.png\"}],\"mbid\":\"\"},\"mbid\":\"069536c6-45f6-489f-93ae-916ab1396cbb\",\"name\":\"Castor\",\"image\":[{\"size\":\"small\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/34s/6bf3e1cd10aad29f0f3da398f22d3a23.jpg\"},{\"size\":\"medium\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/64s/6bf3e1cd10aad29f0f3da398f22d3a23.jpg\"},{\"size\":\"large\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/174s/6bf3e1cd10aad29f0f3da398f22d3a23.jpg\"},{\"size\":\"extralarge\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/300x300/6bf3e1cd10aad29f0f3da398f22d3a23.jpg\"}],\"streamable\":\"0\",\"album\":{\"mbid\":\"\",\"#text\":\"TRON: Legacy - The Complete Edition (Original Motion Picture Soundtrack)\"},\"url\":\"https://www.last.fm/music/Daft+Punk/_/Castor\",\"@attr\":{\"nowplaying\":\"true\"},\"loved\":\"0\"},{\"artist\":{\"url\":\"https://www.last.fm/music/Mac+Quayle\",\"name\":\"Mac Quayle\",\"image\":[{\"size\":\"small\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/34s/2a96cbd8b46e442fc41c2b86b821562f.png\"},{\"size\":\"medium\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/64s/2a96cbd8b46e442fc41c2b86b821562f.png\"},{\"size\":\"large\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/174s/2a96cbd8b46e442fc41c2b86b821562f.png\"},{\"size\":\"extralarge\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/300x300/2a96cbd8b46e442fc41c2b86b821562f.png\"}],\"mbid\":\"\"},\"date\":{\"uts\":\"1641801521\",\"#text\":\"10 Jan 2022, 07:58\"},\"mbid\":\"ac7937d9-7065-4f42-b715-b7b9195f80dc\",\"name\":\"1.1_4-hateurself.ra\",\"image\":[{\"size\":\"small\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/34s/fbe7813f8258abdeeaae93402d8a03fc.jpg\"},{\"size\":\"medium\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/64s/fbe7813f8258abdeeaae93402d8a03fc.jpg\"},{\"size\":\"large\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/174s/fbe7813f8258abdeeaae93402d8a03fc.jpg\"},{\"size\":\"extralarge\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/300x300/fbe7813f8258abdeeaae93402d8a03fc.jpg\"}],\"url\":\"https://www.last.fm/music/Mac+Quayle/_/1.1_4-hateurself.ra\",\"streamable\":\"0\",\"album\":{\"mbid\":\"44e399d5-30d5-4098-885d-a7331beb7b61\",\"#text\":\"Mr. Robot, Vol. 1 (Original Television Series Soundtrack)\"},\"loved\":\"0\"}],\"@attr\":{\"user\":\"VishBK\",\"totalPages\":\"48780\",\"page\":\"1\",\"perPage\":\"1\",\"total\":\"48780\"}}}";
    DynamicJsonDocument doc(4096);
    DeserializationError error = deserializeJson(doc, input);
    if (error) {
        Serial.print("deserializeJson() failed for LastFM: ");
        Serial.println(error.c_str());
    }
    JsonObject parsed = doc["recenttracks"]["track"][0];
    serializeJson(parsed, Serial);
```

and this is what `parsed` prints out:
```
{
  "artist": {
    "url": "https://www.last.fm/music/Daft+Punk",
    "name": "Daft Punk",
    "image": [
      {
        "size": "small",
        "#text": "https://lastfm.freetls.fastly.net/i/u/34s/2a96cbd8b46e442fc41c2b86b821562f.png"
      },
      {
        "size": "medium",
        "#text": "https://lastfm.freetls.fastly.net/i/u/64s/2a96cbd8b46e442fc41c2b86b821562f.png"
      },
      {
        "size": "large",
        "#text": "https://lastfm.freetls.fastly.net/i/u/174s/2a96cbd8b46e442fc41c2b86b821562f.png"
      },
      {
        "size": "extralarge",
        "#text": "https://lastfm.freetls.fastly.net/i/u/300x300/2a96cbd8b46e442fc41c2b86b821562f.png"
      }
    ],
    "mbid": ""
  },
  "mbid": "069536c6-45f6-489f-93ae-916ab1396cbb",
  "name": "Castor",
  "image": [
    {
      "size": "small",
      "#text": "https://lastfm.freetls.fastly.net/i/u/34s/6bf3e1cd10aad29f0f3da398f22d3a23.jpg"
    },
    {
      "size": "medium",
      "#text": "https://lastfm.freetls.fastly.net/i/u/64s/6bf3e1cd10aad29f0f3da398f22d3a23.jpg"
    },
    {
      "size": "large",
      "#text": "https://lastfm.freetls.fastly.net/i/u/174s/6bf3e1cd10aad29f0f3da398f22d3a23.jpg"
    },
    {
      "size": "extralarge",
      "#text": "https://lastfm.freetls.fastly.net/i/u/300x300/6bf3e1cd10aad29f0f3da398f22d3a23.jpg"
    }
  ],
  "streamable": "0",
  "album": {
    "mbid": "",
    "#text": "TRON: Legacy - The Complete Edition (Original Motion Picture Soundtrack)"
  },
  "url": "https://www.last.fm/music/Daft+Punk/_/Castor",
  "@attr": {
    "nowplaying": "true"
  },
  "loved": "0"
}
```
However, I cannot access anything else in the JSON, such as `parsed["image"][1]["#text"]`, it just returns null. I've checked the size with the assistant, and even the assistant's code is not working. Any ideas what the problem is?

bblanchon, 2022-01-11:
Hi,

I cannot reproduce your issue; see this online demo: https://wandbox.org/permlink/XCBQYgb6IZvKpbxd

Some ideas:

1. I noticed that your program doesn't stop if `deserializeJson()` returns an error; maybe you overlooked an error.
2. `input` is a pretty big variable (2700 bytes); maybe the stack overflows
3. ask to the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/)

Best regards,
Benoit

VishBK, 2022-01-12:
Thanks for the help, it's working again now. However, I'm having trouble using a filter here: [https://wokwi.com/arduino/projects/320552516072768082](https://wokwi.com/arduino/projects/320552516072768082)
`doc["recenttracks"]["track"][0]["image"][1]["#text"]` prints out fine without a filter, but when passing in the filter `doc` is empty.

VishBK, 2022-01-12:
This is what the assistant gives me by the way 
```c
  char input[] = "{\"recenttracks\":{\"track\":[{\"artist\":{\"url\":\"https://www.last.fm/music/Daft+Punk\",\"name\":\"Daft Punk\",\"image\":[{\"size\":\"small\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/34s/2a96cbd8b46e442fc41c2b86b821562f.png\"},{\"size\":\"medium\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/64s/2a96cbd8b46e442fc41c2b86b821562f.png\"},{\"size\":\"large\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/174s/2a96cbd8b46e442fc41c2b86b821562f.png\"},{\"size\":\"extralarge\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/300x300/2a96cbd8b46e442fc41c2b86b821562f.png\"}],\"mbid\":\"\"},\"mbid\":\"069536c6-45f6-489f-93ae-916ab1396cbb\",\"name\":\"Castor\",\"image\":[{\"size\":\"small\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/34s/6bf3e1cd10aad29f0f3da398f22d3a23.jpg\"},{\"size\":\"medium\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/64s/6bf3e1cd10aad29f0f3da398f22d3a23.jpg\"},{\"size\":\"large\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/174s/6bf3e1cd10aad29f0f3da398f22d3a23.jpg\"},{\"size\":\"extralarge\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/300x300/6bf3e1cd10aad29f0f3da398f22d3a23.jpg\"}],\"streamable\":\"0\",\"album\":{\"mbid\":\"\",\"#text\":\"TRON: Legacy - The Complete Edition (Original Motion Picture Soundtrack)\"},\"url\":\"https://www.last.fm/music/Daft+Punk/_/Castor\",\"@attr\":{\"nowplaying\":\"true\"},\"loved\":\"0\"},{\"artist\":{\"url\":\"https://www.last.fm/music/Mac+Quayle\",\"name\":\"Mac Quayle\",\"image\":[{\"size\":\"small\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/34s/2a96cbd8b46e442fc41c2b86b821562f.png\"},{\"size\":\"medium\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/64s/2a96cbd8b46e442fc41c2b86b821562f.png\"},{\"size\":\"large\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/174s/2a96cbd8b46e442fc41c2b86b821562f.png\"},{\"size\":\"extralarge\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/300x300/2a96cbd8b46e442fc41c2b86b821562f.png\"}],\"mbid\":\"\"},\"date\":{\"uts\":\"1641801521\",\"#text\":\"10 Jan 2022, 07:58\"},\"mbid\":\"ac7937d9-7065-4f42-b715-b7b9195f80dc\",\"name\":\"1.1_4-hateurself.ra\",\"image\":[{\"size\":\"small\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/34s/fbe7813f8258abdeeaae93402d8a03fc.jpg\"},{\"size\":\"medium\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/64s/fbe7813f8258abdeeaae93402d8a03fc.jpg\"},{\"size\":\"large\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/174s/fbe7813f8258abdeeaae93402d8a03fc.jpg\"},{\"size\":\"extralarge\",\"#text\":\"https://lastfm.freetls.fastly.net/i/u/300x300/fbe7813f8258abdeeaae93402d8a03fc.jpg\"}],\"url\":\"https://www.last.fm/music/Mac+Quayle/_/1.1_4-hateurself.ra\",\"streamable\":\"0\",\"album\":{\"mbid\":\"44e399d5-30d5-4098-885d-a7331beb7b61\",\"#text\":\"Mr. Robot, Vol. 1 (Original Television Series Soundtrack)\"},\"loved\":\"0\"}],\"@attr\":{\"user\":\"VishBK\",\"totalPages\":\"48780\",\"page\":\"1\",\"perPage\":\"1\",\"total\":\"48780\"}}}";

  StaticJsonDocument<128> filter;

  JsonObject filter_recenttracks_track_0 = filter["recenttracks"]["track"].createNestedObject();
  filter_recenttracks_track_0["image"][0]["#text"] = true;
  filter_recenttracks_track_0["@attr"]["nowplaying"] = true;

  StaticJsonDocument<512> doc;
  DeserializationError error = deserializeJson(doc, input, DeserializationOption::Filter(filter));

  if (error) {
    Serial.print("deserializeJson() failed: ");
    Serial.println(error.c_str());
    return;
  }
  serializeJson(doc, Serial);
  const char* nowplaying = doc["recenttracks"]["track"][0]["@attr"]["nowplaying"];
  Serial.println(nowplaying);
```

bblanchon, 2022-01-13:
Your program seems to work fine with ArduinoJson 6.18.5; this is a regression in 6.19.0.
I'll fix this bug as soon as possible.
Thank you very much for reporting this issue 👍

closed 2022-01-13 by bblanchon

bblanchon, 2022-01-14:
The fix was published in [ArduinoJson 6.19.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.19.1)

commit 11310f40 (2022-01-13): Fix filter not working in zero-copy mode (fixes #1697)
commit 21b2c765 (2022-01-13): Fix filter not working in zero-copy mode (fixes #1697)