bblanchon/ArduinoJson #1699: Not an issue, parsing JsonArray with no JsonObject in it

nodoubtman, 2022-01-16:
Bonjour Benoit.
 Comment vas-tu ?

J'ai une question, j'ai un tableau mais ne contenant pas de JsonObject. je peux avec mes propres moyens, aller chercher les valeurs, mais LA BONNE façon de faire, il faudrait...

alors mon code:

MON TABLEAU:

`{
  "rss": [
    "Prise d’otage dans une synagogue au Texas | Un otage libere, les negociations continuent",
    " Les forces de l’ordre americaines menaient samedi soir des negociations « tendues » avec un homme qui a pris plusieurs personnes en otages dans une synagogue de Colleyville, une ville du Texas aux Etats-Unis, disant etre le « frere » d’une Pakistanaise condamnee pour terrorisme. ",
    "Eruption d’un volcan aux iles Tonga | Alerte au tsunami des Etats-Unis au Japon",
    " L’eruption spectaculaire d’un volcan sous-marin des iles Tonga, dans l’ocean Pacifique sud, a provoque un tsunami samedi qui a atteint le Japon avec des vagues de 1,20 m et place le Chili et la cote ouest des Etats-Unis en alerte. ",
    "Visa annule | L’audience de Novak Djokovic en cours",
    " L’ultime bataille judiciaire entre le numero un mondial de tennis Novak Djokovic et le gouvernement australien, qui soutient que le Serbe, non vaccine contre la COVID-19, constitue un « risque sanitaire » et entend l’expulser du pays, a demarre dimanche devant la Cour federale d’Australie a Melbourne. ",
    "Arizona | Donald Trump tiendra un premier rassemblement depuis le 6 janvier",
    " Des milliers de sympathisants de Donald Trump se sont reunis samedi dans l’Arizona pour un grand rassemblement ou l’ex-president americain a prevu d’attaquer son successeur Joe Biden et de repeter ses allegations non prouvees de fraude lors des elections de 2020. ",
    "Assassinat de Jovenel Moise | Un ex-senateur haitien arrete en Jamaique",
    " L’ancien senateur haitien John Joel Joseph, recherche par les autorites de Haiti dans l’enquete sur l’assassinat du president Jovenel Moise, a ete arrete en Jamaique, a-t-on appris samedi de source policiere jamaicaine. ",
    "Un incendie majeur ravage un immeuble de Saint-Jean-sur-Richelieu",
    " Un immeuble de 24 logements a ete devore par les flammes samedi soir sur la rue Lesieur, a Saint-Jean-sur-Richelieu. "
  ]
}`

`
#define MAX_RSS_2  5
typedef struct {
  //String rssTitre;
  //String rssDescr;
  char rssTitre[150];
  char rssDescr[450];
  byte nbArticles;
} rss2Struct;
rss2Struct rss2Datas[MAX_RSS_2];
// rss2Datas[i].rssTitre
// rss2Datas[i].rssDescr
// rss2Datas[0].nbArticles

bool estPair(int chiffre) {
  if ( chiffre & 1)
    return false;

  return true;
}

void Remplir_Tableau(){

 byte monCompteur = 0;

    //for (JsonObject elem : doc["rss"].as<JsonArray>()) {

    for (int i = 0; i < rss2Datas[0].nbArticles; i++) {
      if (!estPair(i)) {
        const char* rss_titre = rss[i] | "titre";
        strcpy( rss2Datas[i].rssTitre, rss_titre );
        rss2Datas[i].rssTitre[strlen(rss2Datas[i].rssTitre)] = '\0';
        Serial.printf("\n%s\n", rss2Datas[i].rssTitre);
      }
      else {
        const char* rss_descr = rss[i] | "descr";
        strcpy( rss2Datas[i].rssDescr, rss_descr );
        rss2Datas[i].rssDescr[strlen(rss2Datas[i].rssDescr)] = '\0';
        Serial.println( rss2Datas[i].rssDescr );
      }

      yield();

    }  // for (int i = 0; i < rss2Datas[0].nbArticles; i++)

    /*
        Serial.println(); Serial.println();
        serializeJsonPretty(doc, Serial);
        Serial.println(); Serial.println();
    */

    delay(1000);
    doc.clear();

    yield();
}

`

Alors, je te remercie beaucoup.
:)

Bonne nuit ici.
A+!
Marc

bblanchon, 2022-01-17:
Hi Marc,

I'm doing good, thanks for asking.

At first sight, this program looks about right, except that:

1. you should use `strlcpy()` instead of `strcpy()` to prevent overflows,
2. adding the null-terminator is redundant with `strcpy()` and `strlcpy()`,
3. the first `yield()` could introduce a race condition,
4. `doc.clear()` is probably useless,
5. the second `yield()` is useless.

If you need an in-depth code review, please [contact me via codementor](https://www.codementor.io/@bblanchon), I'll be happy to help you.

Best regards,
Benoit

nodoubtman, 2022-01-17:
Merci bien Benoit. Tu as perdu ton français ? ;)

bblanchon, 2022-01-18:
L'Anglais permet à plus de personnes de trouver et lire la réponse.

nodoubtman, 2022-01-18:
Ok

closed 2022-01-18 by nodoubtman