bblanchon/ArduinoJson #1700: float serialization not accurate

Burnyfighter, 2022-01-16:
Hello, 
I'm trying to run this small program on esp8266 to fill a json document:
`
float myValue=21.81
StaticJsonDocument<1024> jsonLoad;
JsonArray jsonData=jsonLoad.createNestedArray("data");
JsonObject data_0=jsonData.createNestedObject();
data_0["key"]=myValue;
`
While sending this this is what I have:
{"data":[{"key":20.80999947}]}
Everywhere else where I work with 'myValue' float, caculus are good (as long as I can check) only issue appears while sending it to JSON document.

Any help is appreciated ;-)

bblanchon, 2022-01-17:
Hi @Burnyfighter,

This probably comes from the limited precision of `float`.
You'd have much better precision with `double`.
See this online demo: https://wandbox.org/permlink/TdrKhBRgNmpXr8JS

Best regards,
Benoit

Burnyfighter, 2022-01-18:
Thanks so much Benoit for your reply.
Will test it !

Merci Benoit de prendre le temps pour les novices ;-)

closed 2022-01-18 by Burnyfighter