bblanchon/ArduinoJson #1702: to<T> vs  as<T>  and  JsonObject vs JsonVariant

ali80, 2022-01-20:
## JsonObject vs JsonVariant
I'm confused about actual difference of `JsonObject` and  `JsonVariant` and when to use each, The documentation says:

> In ArduinoJson, an object is a collection of key-value pairs. A JsonObject is a reference to this object, but a JsonDocument owns the data

> JsonVariant is a reference to a value in a JsonDocument. It supports all types allowed by JSON.

## `to<T>` vs `as<T>`
The documentation says
> JsonVariant::as<T>() casts the value pointed by the JsonVariant to the specified type. Unlike JsonVariant::to<T>(), this function doesn’t modify the value pointed by the JsonVariant.

I get what cast is, my question is in what situations I need that cast and when do I need to use `to<T>`

bblanchon, 2022-01-21:
Hi @ali80,

### `JsonObject` vs `JsonVariant`

Both [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/) and [`JsonVariant`](https://arduinojson.org/v6/api/jsonvariant/) are references to values stored in the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/).

The difference is that [`JsonVariant`](https://arduinojson.org/v6/api/jsonvariant/) can refer to any supported type (integer, float, string, array, object...), whereas [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/) can only refer to an object. The advantage of [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/) over [`JsonVariant`](https://arduinojson.org/v6/api/jsonvariant/) is that it supports operations specific to objects, such as [enumerating key-value pairs](https://arduinojson.org/v6/api/jsonobject/begin_end/). 

### `as<T>()` vs `to<T>()`

[`JsonVariant::as<T>()`](https://arduinojson.org/v6/api/jsonvariant/as/) and [`JsonVariant::to<T>()`](https://arduinojson.org/v6/api/jsonvariant/to/) look similar but are very different: the former reads the value, whereas the latter changes the reference.

Suppose the [`JsonVariant`](https://arduinojson.org/v6/api/jsonvariant/) refers to an object:

* `JsonVariant::as<JsonObject>()` returns a [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/) pointing to this object.
* `JsonVariant::to<JsonObject>()`:
    1. creates a new object in the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/),
    2. makes the [`JsonVariant`](https://arduinojson.org/v6/api/jsonvariant/) point to the new object,
    3. returns a [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/) pointing to it.

[`JsonVariant::to<T>()`](https://arduinojson.org/v6/api/jsonvariant/to/) allows **creating an empty object or an empty array**, and it's probably its only legitimate use. Calling [`JsonVariant::to<T>()`](https://arduinojson.org/v6/api/jsonvariant/to/) with a `T` other than [`JsonArray`](https://arduinojson.org/v6/api/jsonarray/) or [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/) works but has no practical use.

[`JsonVariant::as<T>()`](https://arduinojson.org/v6/api/jsonvariant/as/), on the other hand, allows **solving situations where implicit casts don't work**; for example, when you call a function with several matching overloads (like `Serial.print()`) or when you use type deduction with the `auto` keyword.

Remark that [`JsonVariantConst`](https://arduinojson.org/v6/api/jsonvariantconst/), which is a read-only version of [`JsonVariant`](https://arduinojson.org/v6/api/jsonvariant/), supports [`as<T>()`](https://arduinojson.org/v6/api/jsonvariantconst/as/) but not `to<T>()`.

Let me know if you have any other questions.

Best regards,
Benoit

ali80, 2022-01-23:
Hi @bblanchon 
many thanks for your awesome work, sorry for late response, I got sick for a few days.
your explanations answers all my questions
here is my simplified use case, I wanna serialize/deserialize and array of `Config` objects. below is a sample of what I did, is there any recommendations to improve it or do you see any problems in my usage.
```json
{
  "data": [
    {
      "id": 1,
      "name": "module1"
    },
    {
      "id": 2,
      "name": "module2"
    }
  ],
  "_v": 1
}
```
 this serialization/deserialization is used several places and so I've added `toJson` and `fromJson` to the `Config` class itself.
```cpp
struct Config {
    Config() { memset(this, 0, sizeof(Config)); }
    int64_t id;
    char name[32] = "";
    // many more fields ...

  void toJson(const JsonObject& obj)
    {
        if (id)
            obj[_KEY_id] = id;
        if (name)
            obj[_KEY_name] = name;
   }

  void fromJson(const JsonObject& obj)
    {
        if (obj.containsKey(_KEY_id)) {
            id = obj[key];
        }
        const char* nameptr = obj[_KEY_name];
        if (name)
            strncpys(name, nameptr, sizeof(name));
  }
}

void toJsonAll(const JsonVariant& root)
    {
        JsonVariant obj = root.createNestedObject("data");
        JsonArray array = obj.to<JsonArray>();
        for (int32_t i = 0; i < configs.length(); i++) {
            auto el = array.addElement();
            configs[i].toJson(el.to<JsonObject>());
        }
    }
void fromJsonAll(const JsonVariant& obj)
    {
        JsonArray arr = obj["data"];
        int i=0;
        for (const JsonObject& v : arr) { 
              configs[i].fromJson(v);
              i++;
        }
    }

int32_t file_load()
    {
        DynamicJsonDocument doc(2048);
        int32_t ret = 0;
        if (FS1.exists(_DEFS_CONFIGSPATH)) {
            File file = FS1.open(_DEFS_CONFIGSPATH, "r"); // read
            DeserializationError error = deserializeJson(doc, file);
            if (error) {
                file.close();
                log_c(1, MODULE "Deserialize Error %s", error.c_str());
                ret = 1;
            } else {
                fromJsonAll(doc.as<JsonObject>());
                file.close();
                ret = 0;
            }
        } else {
            log_c(2, MODULE "File not found");
            ret = 3;
        }
     }
   return ret;
}
int32_t file_save()
    {
        DynamicJsonDocument doc(2048);
        toJsonAll(doc.to<JsonObject>());
        File file = FS1.open(_DEFS_CONFIGSPATH, "w+"); // write
        serializeJson(doc, file);
        file.close();
        log_c(3, MODULE "Saved");
        return 0;
    }
```

bblanchon, 2022-01-24:
Hi @ali80,

For simple configurations, see [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/).
For complex configurations, use [custom converters](https://arduinojson.org/news/2021/05/04/version-6-18-0/), as shown in the case studies of [Mastering ArduinoJson](https://arduinojson.org/book/).

Best regards,
Benoit

closed 2022-01-24 by ali80