bblanchon/ArduinoJson #1704: Cannot add camera image buffer to JSON string in NodeMCU

pradeep22kumar, 2022-01-29:
I want to add the camera image buffer string data to json string but i could not able to do it. Here is the string 
`0xFF 0xD8 0xFF 0xE0 0x00 0x10 0x4A 0x46 0x49 0x46 0x00 0x01 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xDB 0x00 0x43 0x00 0xC 0x08 0x09 0xB 0x09 0x08 0xC 0xB 0xA 0xB 0xE 0xD 0xC 0xE 0x12 0x1E 0x14 0x12 0x11 0x11 0x12 0x25 0x1A 0x1C 0x16 0x1E 0x2C 0x26 0x2E 0x2D 0x2B 0x26 0x2A 0x29 0x30 0x36 0x45 0x3B 0x30 0x33 0x41 0x34 0x29 0x2A 0x3C 0x52 0x3D 0x41 0x47 0x4A 0x4D 0x4E 0x4D 0x2F 0x3A 0x55 0x5B 0x54 0x4B 0x5A 0x45 0x4C 0x4D 0x4A 0xFF 0xDB 0x00 0x43 0x01 0xD 0xE 0xE 0x12 0x10 0x12 0x23 0x14 0x14 0x23 0x4A 0x32 0x2A 0x32 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0xFF 0xC4 0x00 0x1F 0x00 0x00 0x01 0x05 0x01 0x01 0x01 0x01 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x02 0x03 0x04 0x05 0x06 0x07 0x08 0x09 0xA 0xB 0xFF 0xC4 0x00 0xB5 0x10 0x00 0x02 0x01 0x03 0x03 0x02 0x04 0x03 0x05 0x05 0x04 0x04 0x00 0x00 0x01 0x7D 0x01 0x02 0x03 0x00 0x04 0x11 0x05 0x12 0x21 0x31 0x41 0x06 0x13 0x51 0x61 0x07 0x22 0x71 0x14 0x32 0x81 0x91 0xA1 0x08 0x23 0x42 0xB1 0xC1 0x15 0x52 0xD1 0xF0 0x24 0x33 0x62 0x72 0x82 0x09 0xA 0x16 0x17 0x18 0x19 0x1A 0x25 0x26 0x27 0x28 0x29 0x2A 0x34 0x35 0x36 0x37 0x38 0x39 0x3A 0x43 0x44 0x45 0x46 0x47 0x48 0x49 0x4A 0x53 0x54 0x55 0x56 0x57 0x58 0x59 0x5A 0x63 0x64 0x65 0x66 0x67 0x68 0x69 0x6A 0x73 0x74 0x75 0x76 0x77 0x78 0x79 0x7A 0x83 0x84 0x85 0x86 0x87 0x88 0x89 0x8A 0x92 0x93 0x94 0x95 0x96 0x97 0x98 0x99 0x9A 0xA2 0xA3 0xA4 0xA5 0xA6 0xA7 0xA8 0xA9 0xAA 0xB2 0xB3 0xB4 0xB5 0xB6 0xB7 0xB8 0xB9 0xBA 0xC2 0xC3 0xC4 0xC5 0xC6 0xC7 0xC8 0xC9 0xCA 0xD2 0xD3 0xD4 0xD5 0xD6 0xD7 0xD8 0xD9 0xDA 0xE1 0xE2 0xE3 0xE4 0xE5 0xE6 0xE7 0xE8 0xE9 0xEA 0xF1 0xF2 0xF3 0xF4 0xF5 0xF6 0xF7 0xF8 0xF9 0xFA 0xFF 0xC4 0x00 0x1F 0x01 0x00 0x03 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x02 0x03 0x04 0x05 0x06 0x07 0x08 0x09 0xA 0xB 0xFF 0xC4 0x00 0xB5 0x11 0x00 0x02 0x01 0x02 0x04 0x04 0x03 0x04 0x07 0x05 0x04 0x04 0x00 0x01 0x02 0x77 0x00 0x01 0x02 0x03 0x11 0x04 0x05 0x21 0x31 0x06 0x12 0x41 0x51 0x07 0x61 0x71 0x13 0x22 0x32 0x81 0x08 0x14 0x42 0x91 0xA1 0xB1 0xC1 0x09 0x23 0x33 0x52 0xF0 0x15 0x62 0x72 0xD1 0xA 0x16 0x24 0x34 0xE1 0x25 0xF1 0x17 0x18 0x19 0x1A 0x26 0x27 0x28 0x29 0x2A 0x35 0x36 0x37 0x38 0x39 0x3A 0x43 0x44 0x45 0x46 0x47 0x48 0x49 0x4A 0x53 0x54 0x55 0x56 0x57 0x58 0x59 0x5A 0x63 0x64 0x65 0x66 0x67 0x68 0x69 0x6A 0x73 0x74 0x75 0x76 0x77 0x78 0x79 0x7A 0x82 0x83 0x84 0x85 0x86 0x87 0x88 0x89 0x8A 0x92 0x93 0x94 0x95 0x96 0x97 0x98 0x99 0x9A 0xA2 0xA3 0xA4 0xA5 0xA6 0xA7 0xA8 0xA9 0xAA 0xB2 0xB3 0xB4 0xB5 0xB6 0xB7 0xB8 0xB9 0xBA 0xC2 0xC3 0xC4 0xC5 0xC6 0xC7 0xC8 0xC9 0xCA 0xD2 0xD3 0xD4 0xD5 0xD6 0xD7 0xD8 0xD9 0xDA 0xE2 0xE3 0xE4 0xE5 0xE6 0xE7 0xE8 0xE9 0xEA 0xF2 0xF3 0xF4 0xF5 0xF6 0xF7 0xF8 0xF9 0xFA 0xFF 0xC0 0x00 0x11 0x08 0x00 0xF0 0x01 0x40 0x03 0x01 0x21 0x00 0x02 0x11 0x01 0x03 0x11 0x01 0xFF 0xDA 0x00 0xC 0x03 0x01 0x00 0x02 0x11 0x03 0x11 0x00 0x3F 0x00 0xE0 0xAA 0x29 0xFF 0x00 0xD5 0x9A 0x00 0xA2 0x2A 0x58 0x07 0xEF 0x28 0xEA 0x22 0x6B 0xA5 0xC9 0x5A 0x07 0x4A 0x5D 0x06 0x3D 0xBE 0xE5 0x40 0x68 0x6F 0x51 0x85 0x3E 0xA2 0x40 0x32 0x51 0x48 0x89 0xEB 0x4F 0x9B 0x40 0x24 0xA6 0x33 0xD2 0x5A 0x83 0x23 0xCE 0x4D 0x4B 0x1A 0x77 0x35 0x62 0x5B 0x92 0x8E 0xB5 0x38 0x1E 0x57 0xBB 0x52 0x88 0xD8 0xD2 0x73 0x45 0x5E 0xFA 0x88 0x28 0xA6 0x02 0xD2 0xD3 0x01 0x68 0xA4 0x02 0xD1 0x4C 0x02 0x93 0x34 0x98 0x84 0xDF 0x4C 0x69 0x6A 0x44 0x42 0xF3 0x54 0x12 0x49 0xBF 0xF0 0xA5 0xD4 0x76 0x22 0xA2 0x80 0xB8 0x56 0x8C 0x12 0xC4 0x96 0xE9 0xB5 0x7F 0x79 0xFC 0x54 0xC9 0x90 0xD4 0xBB 0x60 0x6A 0xDC 0x77 0x74 0xEE 0x4B 0x44 0x35 0x1C 0xDF 0xEA 0xCD 0x26 0x6A 0x51 0x51 0x52 0x41 0xFE 0xB4 0x50 0x05 0x9B 0x8E 0xD5 0x5C 0x35 0x20 0x2C 0x7F 0x05 0x42 0x69 0x32 0x86 0xD4 0x95 0x23 0xE 0xF4 0x74 0xA5 0x6D 0x2C 0x04 0x6E 0xF4 0xCE 0xB5 0x69 0x10 0x4A 0x89 0x8A 0x95 0x57 0x3D 0x2A 0x59 0x5B 0x13 0xF1 0x10 0xFF 0x00 0x6A 0x99 0x5A 0xA4 0x21 0x69 0x68 0xD0 0x34 0xA 0x29 0x20 0x62 0x8A 0x5A 0x62 0xA 0x33 0x55 0xD4 0x03 0xB5 0x26 0x68 0x01 0x99 0xA6 0x96 0xA4 0x22 0x36 0x6A 0x89 0x9E 0x90 0xC8 0x89 0xCD 0x25 0x20 0x24 0xF9 0x71 0xF3 0xA 0x12 0x35 0x60 0x79 0xE7 0xD2 0x8E 0xA2 0x2D 0xC7 0x64 0x9D 0xF3 0x56 0x7C 0x95 0xC6 0x00 0xAB 0xB7 0x72 0x08 0x5E 0xDC 0x62 0xA2 0xF2 0xB6 0xF4 0xA8 0x68 0x09 0xEA 0x39 0x7F 0xD5 0xB5 0x23 0x4D 0x4A 0x09 0xD6 0xA4 0x1F 0x2C 0x8B 0x40 0x16 0xAE 0x3E 0xE8 0xA8 0xCC 0x61 0x40 0xC5 0x27 0xB0 0x12 0x2F 0x4F 0xC2 0xA1 0x34 0x99 0x48 0x6D 0x48 0xBD 0x29 0x31 0x85 0x42 0xE6 0x84 0x26 0x37 0xA9 0xA9 0x91 0x71 0x56 0xF6 0x24 0x94 0x29 0x73 0x81 0x53 0x70 0x9C 0x2F 0x5A 0x51 0x45 0xC 0xA5 0xE9 0x57 0xB1 0x22 0xD2 0xD3 0x63 0xA 0x4A 0x91 0xB 0x45 0x50 0x03 0x53 0x68 0x60 0xC4 0x2D 0x4C 0x26 0x80 0x1A 0x4D 0x33 0x35 0x22 0xB0 0xC3 0x4D 0xDB 0x43 0x43 0xB8 0xE1 0x1D 0x4A 0x23 0xA1 0x08 0x6C 0xF1 0x61 0x77 0x54 0xD6 0xF6 0xC1 0xE 0xEE 0xF4 0xC4 0xCB 0x79 0xA7 0x66 0xAC 0x81 0x7A 0xD2 0x14 0xA5 0xD0 0x45 0x7A 0x8E 0x6F 0xF5 0x66 0xA1 0xB3 0x52 0x8C 0x7F 0x7A 0x9E 0xC3 0x8A 0x9E 0xA3 0x2C 0x93 0xBA 0xDD 0x4D 0x1F 0xC0 0xB5 0x32 0x18 0xB1 0xD4 0x4D 0x4F 0xB0 0xC6 0x53 0xD6 0x9B 0x18 0xEA 0x85 0x97 0x2D 0xC5 0x4A 0xEE 0x26 0x48 0xAA 0x14 0x53 0xD1 0x37 0x51 0xB8 0xF6 0x27 0xFB 0xAB 0xB5 0x69 0xB5 0x64 0x8B 0x45 0x50 0x05 0x2D 0x31 0x05 0x15 0x25 0xB 0x45 0x3B 0x88 0x43 0x51 0x1A 0x3D 0x04 0x21 0xA6 0x9A 0x60 0x25 0x25 0x4D 0xC0 0x4C 0x66 0x9C 0x16 0x98 0x12 0x05 0xA9 0x40 0xA0 0x44 0x52 0xF3 0x32 0x2F 0xA5 0x59 0xE9 0xD2 0x9A 0x24 0x29 0xC2 0x80 0x17 0x34 0xEC 0xD2 0xB1 0x2F 0x62 0xB5 0x47 0x37 0xDC 0x34 0x9A 0x35 0x28 0x29 0xE6 0xA6 0x15 0x2C 0x07 0x2F 0xDC 0x65 0xA5 0x52 0x7A 0x53 0x90 0x12 0xC7 0x51 0xC8 0x31 0x9A 0x9B 0xEC 0x55 0xC8 0xAA 0x44 0xA7 0x21 0x8E 0xA2 0xB3 0x45 0xF 0x44 0xDC 0x7D 0xAA 0x4C 0xE3 0x85 0xE9 0x57 0x15 0x62 0x44 0xA5 0xAB 0x24 0x5A 0x28 0x40 0x14 0x53 0x6C 0x02 0x8A 0x00 0x5A 0x28 0xB8 0x98 0xB4 0xD6 0x5C 0xD3 0x19 0x16 0xDA 0x69 0xA4 0x20 0xA4 0xA1 0x00 0xA1 0x69 0xF8 0xA4 0x2B 0x8F 0x5C 0x52 0xBB 0xED 0x15 0x57 0xB2 0xB 0x15 0x62 0x2D 0x24 0x84 0xE7 0x9A 0xB8 0x32 0x05 0x4C 0x44 0x3A 0x9D 0x54 0xC4 0x25 0x39 0x4D 0x4E 0x84 0x90 0xD4 0x53 0xFF 0x00 0xAB 0x6A 0xD 0x4A 0x09 0x53 0x27 0xA5 0x4B 0x01 0x5B 0x82 0xF 0xAD 0x49 0x02 0xF3 0x9A 0x60 0x3D 0x38 0xA4 0x9F 0xEF 0xD2 0x7F 0x09 0x44 0x14 0xE4 0xEB 0x50 0xEE 0x31 0xF4 0xE4 0x5C 0x9E 0x68 0x1B 0x25 0x3E 0xDD 0x29 0x2B 0x5B 0x10 0x2D 0x2E 0x28 0xD0 0x00 0x52 0xD0 0x01 0x4B 0x45 0x80 0x4A 0x29 0x80 0x52 0xD1 0xB0 0xB 0x49 0x43 0x10 0x11 0x9A 0x84 0xAD 0x30 0x1B 0xDA 0x96 0xA4 0x05 0xA3 0x70 0x14 0xC0 0x43 0x30 0xED 0x4D 0x7C 0xBF 0xD2 0x95 0xC0 0x7C 0x03 0x6F 0x6A 0x9E 0x81 0x00 0xA5 0xAA 0x24 0x29 0x73 0x52 0xF6 0x11 0x1D 0x45 0x37 0xDC 0x6A 0xD 0x7A 0x94 0x23 0xFB 0xD5 0x2A 0xF0 0xD5 0x2C 0x44 0x93 0x8F 0x97 0x35 0x24 0x3D 0x28 0xE8 0x31 0xDD 0xCD 0x13 0xFF 0x00 0x4A 0x3E 0xC8 0xCA 0xF4 0x2F 0x5A 0x37 0x19 0x25 0x39 0x7A 0xD4 0x6C 0xC0 0x92 0x81 0x5B 0x22 0x47 0x51 0x48 0x02 0x96 0x80 0xA 0x29 0x80 0x51 0x4A 0xE2 0x61 0x46 0x6A 0x9A 0x01 0x68 0xCD 0x20 0x1B 0xBA 0x93 0x34 0x03 0x23 0x3C 0xD2 0x60 0xD2 0x00 0x09 0x49 0x22 0x63 0xA0 0xA6 0x05 0x72 0x30 0x6A 0x58 0xDC 0x80 0x45 0x47 0xA8 0x13 0x42 0xC7 0x15 0x35 0x54 0x50 0x30 0xA5 0xCD 0x04 0x8B 0xDA 0x8A 0x64 0xB1 0x95 0x1C 0xDF 0x70 0xD2 0x66 0x97 0x28 0x45 0xF7 0x85 0x4C 0xC3 0x06 0xA2 0x43 0x48 0x91 0xF9 0x86 0x96 0x1F 0xBB 0x40 0xF 0xFE 0x33 0x4B 0x28 0xE1 0x68 0xE8 0x32 0xAE 0x29 0x68 0x02 0x51 0x45 0x4A 0x65 0x12 0x53 0xAB 0x6D 0x08 0x1D 0x49 0x8A 0x40 0x2D 0x14 0x30 0x16 0x8A 0x2E 0x03 0x68 0xA3 0x51 0x30 0xE6 0x8C 0x1A 0x2E 0x02 0x91 0x49 0xC5 0x30 0xA 0x31 0x43 0x00 0xDB 0x4A 0x05 0x00 0x18 0xA0 0x8A 0x57 0x24 0x6B 0x20 0x6E 0xA2 0x8F 0x2D 0x7B 0x53 0x1A 0x17 0x1E 0x94 0xB4 0x99 0x21 0x4B 0x47 0x40 0x02 0x68 0x14 0x26 0x1A 0x01 0xA8 0x67 0xFF 0x00 0x56 0x69 0x96 0x50 0x1F 0x2B 0xD5 0xA9 0x39 0xC3 0x54 0xB0 0x1B 0x9F 0x90 0x8A 0x92 0x23 0x80 0x6A 0x46 0x38 0x7D 0xFA 0x7C 0xBC 0xA0 0xA3 0xA0 0xCA 0xA4 0x73 0x45 0x3B 0x20 0x1E 0xB4 0xEA 0xCC 0x64 0x99 0xE2 0x97 0x35 0xAA 0xD4 0x56 0x1D 0x45 0x31 0xB 0xCD 0x14 0x58 0x03 0x14 0xEC 0x50 0x02 0x51 0x40 0x05 0x21 0xCD 0x1E 0x60 0x03 0x9A 0x31 0x54 0x21 0x71 0x4B 0x48 0x10 0x51 0x4C 0x04 0xA2 0xA6 0xDA 0x88 0x33 0x45 0x00 0x25 0x14 0x6C 0x20 0xA2 0x81 0x05 0x14 0x00 0xB5 0x05 0xC7 0x [... cut ...]
        stream_root["image_data"] = String("No Image Present");
    }
    
    stream_root["temperature"] = temp_obj;
    if(tag.length() == 0){
      stream_root["rfid"] = "";
    }
    else
    {
        stream_root["rfid"] = tag.c_str();
    }
}`
After Conversion my output is 
`{
  "uuid": "d4bff49-9a70-4a5f-a787-4018714fb0f1",
  "employee": false,
  "timestamp": "Sat 2022-01-29 17:42:12 GMT",
  "image_data": "⸮⸮⸮⸮",
  "temperature": 38,
  "rfid": ""
}`
In Image_data key value is not copied correctly. Kindly help to resolve this.

bblanchon, 2022-01-31:
Hi @pradeep22kumar,

Because you passed a `const char*` to the `JsonDocument`, it treats the string as a zero-terminated string and, therefore, stops at the first NUL character. That explains why you can only see 4 characters in the Serial Monitor.

To fix this you need to pass a string type that supports NUL.
For example, assuming `temp_buf` is an `std::string`, you can write:

    stream_root["image_data"] = temp_buf;

Alternatively, you could write:

    stream_root["image_data"] = JsonString(temp_buf.c_str(), cam_buff_len);

Note that the Arduino `String` class does **not** support NUL, so it is not suitable for this application.

Best regards,
Benoit

pradeep22kumar, 2022-01-31:
std::string is not available in the Arduino String Class. So what should i do ? On giving this JsonString i am getting compilation error . Is this function available in your Arduino json. Please let me know how to finish this issue ASAP.  I am running out of time.

bblanchon, 2022-02-02:
Hi @pradeep22kumar,

As I said, Arduino's `String` class doesn't support NUL characters, so `temp_buf` is probably truncated.
If so, you'll have to use another storage type, such as `std::string` or a `char` buffer.

You said that `std::string` isn't supported on Arduino, but this statement is only partially true.
Indeed many Arduino cores, such as ESP8266 and ESP32, support it.
Of course, I cannot tell if it's supported in your case because you didn't mention the kind of hardware that you use, even though the issue template clearly recommended doing so.

[`JsonString`](https://arduinojson.org/v6/api/jsonstring/) is indeed a class from ArduinoJson.
Its capabilities were significantly enhanced in [version 6.19](https://arduinojson.org/news/2022/01/08/arduinojson-6-19-0/), so it's crucial that you use a recent version of the library.
You need to upgrade to 6.19 to support NUL in strings anyway.

Regarding the compilation error, I cannot help you if you don't tell me the content of the message.
I understand that you are in a hurry, but I cannot help you if you don't give all the required information.

Best regards,
Benoit

pradeep22kumar, 2022-02-07:
Hi @bblanchon ,

I have changed my library to ArduinoJson 6.19.1 in my hardware platform NodeMCU12E. Declared the buffer type as std::string camera_buffer; in the code. 

`
len = fifo_len();
while(len--)
  {
    temp = myCAM.read_fifo();
     camera_buffer.push_back(temp);  
  }` 
Again forming with JON String is below
 
  ` stream_root["uuid"] = uuid.c_str();
    stream_root["employee"] = (new_card_present>0)?true:false;
    stream_root["timestamp"] = buf;
    stream_root["image_data"] = JsonString(camera_buffer.c_str());
    stream_root["rfid"] = tag.c_str();
`
Still the issue is persisting and correct data i am not seeing in json string format. 
`{
   "uuid": "d4bff49-9a70-4a5f-a787-4018714fb0f1",
   "employee": false,
   "timestamp": "Mon 2022-02-07 04:19:11 GMT",
   "image_data": "⸮⸮⸮",
   "temperature": 31.29000092,
   "rfid": ""
}`
Input string for camera buffer is
`0xD8 0xFF 0xE0 0x00 0x10 0x4A 0x46 0x49 0x46 0x00 0x01 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xDB 0x00 0x43 0x00 0xC 0x08 0x09 0xB 0x09 0x08 0xC 0xB 0xA 0xB 0xE 0xD 0xC 0xE 0x12 0x1E 0x14 0x12 0x11 0x11 0x12 0x25 0x1A 0x1C 0x16 0x1E 0x2C 0x26 0x2E 0x2D 0x2B 0x26 0x2A 0x29 0x30 0x36 0x45 0x3B 0x30 0x33 0x41 0x34 0x29 0x2A 0x3C 0x52 0x3D 0x41 0x47 0x4A 0x4D 0x4E 0x4D 0x2F 0x3A 0x55 0x5B 0x54 0x4B 0x5A 0x45 0x4C 0x4D 0x4A 0xFF 0xDB 0x00 0x43 0x01 0xD 0xE 0xE 0x12 0x10 0x12 0x23 0x14 0x14 0x23 0x4A 0x32 0x2A 0x32 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0x4A 0xFF 0xC4 0x00 0x1F 0x00 0x00 0x01 0x05 0x01 0x01 0x01 0x01 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x02 0x03 0x04 0x05 0x06 0x07 0x08 0x09 0xA 0xB 0xFF 0xC4 0x00 0xB5 0x10 0x00 0x02 0x01 0x03 0x03 0x02 0x04 0x03 0x05 0x05 0x04 0x04 0x00 0x00 0x01 0x7D 0x01 0x02 0x03 0x00 0x04 0x11 0x05 0x12 0x21 0x31 0x41 0x06 0x13 0x51 0x61 0x07 0x22 0x71 0x14 0x32 0x81 0x91 0xA1 0x08 0x23 0x42 0xB1 0xC1 0x15 0x52 0xD1 0xF0 0x24 0x33 0x62 0x72 0x82 0x09 0xA 0x16 0x17 0x18 0x19 0x1A 0x25 0x26 0x27 0x28 0x29 0x2A 0x34 0x35 0x36 0x37 0x38 0x39 0x3A 0x43 0x44 0x45 0x46 0x47 0x48 0x49 0x4A 0x53 0x54 0x55 0x56 0x57 0x58 0x59 0x5A 0x63 0x64 0x65 0x66 0x67 0x68 0x69 0x6A 0x73 0x74 0x75 0x76 0x77 0x78 0x79 0x7A 0x83 0x84 0x85 0x86 0x87 0x88 0x89 0x8A 0x92 0x93 0x94 0x95 0x96 0x97 0x98 0x99 0x9A 0xA2 0xA3 0xA4 0xA5 0xA6 0xA7 0xA8 0xA9 0xAA 0xB2 0xB3 0xB4 0xB5 0xB6 0xB7 0xB8 0xB9 0xBA 0xC2 0xC3 0xC4 0xC5 0xC6 0xC7 0xC8 0xC9 0xCA 0xD2 0xD3 0xD4 0xD5 0xD6 0xD7 0xD8 0xD9 0xDA 0xE1 0xE2 0xE3 0xE4 0xE5 0xE6 0xE7 0xE8 0xE9 0xEA 0xF1 0xF2 0xF3 0xF4 0xF5 0xF6 0xF7 0xF8 0xF9 0xFA 0xFF 0xC4 0x00 0x1F 0x01 0x00 0x03 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x02 0x03 0x04 0x05 0x06 0x07 0x08 0x09 0xA 0xB 0xFF 0xC4 0x00 0xB5 0x11 0x00 0x02 0x01 0x02 0x04 0x04 0x03 0x04 0x07 0x05 0x04 0x04 0x00 0x01 0x02 0x77 0x00 0x01 0x02 0x03 0x11 0x04 0x05 0x21 0x31 0x06 0x12 0x41 0x51 0x07 0x61 0x71 0x13 0x22 0x32 0x81 0x08 0x14 0x42 0x91 0xA1 0xB1 0xC1 0x09 0x23 0x33 0x52 0xF0 0x15 0x62 0x72 0xD1 0xA 0x16 0x24 0x34 0xE1 0x25 0xF1 0x17 0x18 0x19 0x1A 0x26 0x27 0x28 0x29 0x2A 0x35 0x36 0x37 0x38 0x39 0x3A 0x43 0x44 0x45 0x46 0x47 0x48 0x49 0x4A 0x53 0x54 0x55 0x56 0x57 0x58 0x59 0x5A 0x63 0x64 0x65 0x66 0x67 0x68 0x69 0x6A 0x73 0x74 0x75 0x76 0x77 0x78 0x79 0x7A 0x82 0x83 0x84 0x85 0x86 0x87 0x88 0x89 0x8A 0x92 0x93 0x94 0x95 0x96 0x97 0x98 0x99 0x9A 0xA2 0xA3 0xA4 0xA5 0xA6 0xA7 0xA8 0xA9 0xAA 0xB2 0xB3 0xB4 0xB5 0xB6 0xB7 0xB8 0xB9 0xBA 0xC2 0xC3 0xC4 0xC5 0xC6 0xC7 0xC8 0xC9 0xCA 0xD2 0xD3 0xD4 0xD5 0xD6 0xD7 0xD8 0xD9 0xDA 0xE2 0xE3 0xE4 0xE5 0xE6 0xE7 0xE8 0xE9 0xEA 0xF2 0xF3 0xF4 0xF5 0xF6 0xF7 0xF8 0xF9 0xFA 0xFF 0xC0 0x00 0x11 0x08 0x00 0xF0 0x01 0x40 0x03 0x01 0x21 0x00 0x02 0x11 0x01 0x03 0x11 0x01 0xFF 0xDA 0x00 0xC 0x03 0x01 0x00 0x02 0x11 0x03 0x11 0x00 0x3F 0x00 0xE1 0x29 0x33 0xEF 0x58 0x94 0x19 0xA2 0x95 0xD8 0x83 0x34 0x6E 0xA7 0xCC 0x3D 0xC3 0x26 0x97 0x34 0x5D 0x80 0xBB 0x8D 0x4F 0x6E 0x49 0x7C 0x56 0x97 0xB3 0x02 0x1C 0xFB 0xD1 0x93 0x53 0x79 0xC 0x5C 0x9A 0x50 0x4D 0x3E 0x80 0x58 0x88 0x61 0xC 0x8D 0xD3 0xB0 0xA6 0xF9 0xCF 0x9F 0x95 0xB1 0xF4 0xAA 0xE6 0x01 0xF7 0x4C 0xC4 0xAE 0xE3 0xF3 0x63 0x9A 0x83 0x9A 0x1B 0xD4 0x04 0xC9 0xA5 0xC9 0xA9 0x63 0x16 0x96 0x8E 0x66 0x01 0x93 0x4F 0x52 0x6A 0xC4 0xCD 0xFD 0x29 0x71 0x1E 0x6A 0xD5 0x6E 0x21 0x33 0x49 0x9C 0x55 0x80 0xC6 0x7C 0x54 0x32 0x4F 0x8A 0x9B 0x85 0x8A 0x72 0x5E 0xB5 0x54 0x92 0xF6 0x53 0xD1 0x8D 0x63 0x29 0x95 0xCA 0x8A 0xEF 0x71 0x2F 0xF7 0xDB 0xF3 0xA9 0x61 0x91 0x88 0xC9 0x62 0x73 0x52 0xA7 0x71 0x32 0x6D 0xC7 0xD6 0x8C 0x9F 0x5A 0xD7 0x63 0x26 0xB5 0xC 0x9A 0x4C 0xFB 0xD3 0xBB 0x00 0xC9 0xF5 0xA3 0x9A 0x4D 0xD9 0x12 0xCC 0xBE 0x68 0xC1 0xAE 0x26 0xD9 0xD0 0x27 0x34 0xB8 0x34 0xAE 0x21 0x39 0xA3 0x9A 0x2F 0x21 0xE8 0x26 0xF 0xAD 0x3B 0x93 0xD2 0xAB 0xDE 0x06 0x4C 0xB6 0xAF 0xB7 0x2C 0xD8 0xA9 0x6C 0xD7 0x17 0x03 0x9C 0xD6 0xEA 0x9C 0xAF 0xAB 0x33 0xB9 0xC 0xF1 0xEC 0x7F 0x94 0xF1 0x4C 0xF9 0xAB 0x37 0x7E 0xE5 0xB 0xF8 0xD2 0xE0 0xD2 0xD4 0x64 0xF8 0x6F 0xB2 0xAE 0x3B 0x35 0x10 0x2E 0x64 0xE7 0xB7 0x34 0xEE 0xD0 0x11 0xB1 0x66 0x62 0x49 0xA4 0xE6 0xA6 0xEC 0xAB 0xB 0xCF 0xAD 0x2E 0xD 0x3B 0xB0 0xB0 0x73 0x46 0xD 0x3B 0xBB 0x00 0xB8 0x3E 0xB5 0x2C 0x40 0xEE 0xA6 0x9B 0x06 0x8E 0x8E 0xC5 0x71 0x05 0x49 0x5D 0x04 0xA1 0xA7 0x35 0x1B 0x1A 0xAB 0xB2 0x8A 0xEE 0x4D 0x53 0x99 0xCF 0xAD 0x65 0x26 0xC [... cut ...] x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xAA `

Let me know what is the problem from my end why key value is not appending after changing the library .

pradeep22kumar, 2022-02-08:
Hi @bblanchon ,

After Converting my Camera buffer to JsonString like this 
`
    stream_root["uuid"] = uuid.c_str();
    stream_root["employee"] = (new_card_present>0)?true:false;
    stream_root["timestamp"] = buf;
    stream_root["image_data"] = JsonString(camera_buffer.c_str(),cam_buff_len);
    stream_root["rfid"] = tag.c_str();
`
Null character in the string gets converted into 

`{
   "uuid": "d4bff49-9a70-4a5f-a787-4018714fb0f1",
   "employee": false,
   "timestamp": "Tue 2022-02-08 03:11:15 GMT",
   "image_data": "⸮⸮⸮\u0000JFIF\u0000\u0000\u0000\u0000\u0000\u0000\u0000⸮⸮\u0000C\u0000\f\b\t\t\b\f\n\r\f%,&.-+&*)06E;03A4)*<R=AGJMNM/:U[TKZELMJ⸮⸮\u0000C\r##J2*2JJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJJ⸮⸮\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\b\t\n⸮⸮\u0000⸮\u0000\u0000\u0000}\u0000!1AQa\"q2⸮⸮⸮\b#B⸮⸮R⸮⸮$3br⸮\t\n%&'()*456789:CDEFGHIJSTUVWXYZcdefghijstuvwxyz⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\b\t\n⸮⸮\u0000⸮\u0000\u0000w\u0000!1AQaq\"2⸮\bB⸮⸮⸮⸮\t#3R⸮br⸮\n$4⸮%⸮&'()*56789:CDEFGHIJSTUVWXYZcdefghijstuvwxyz⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮\u0000\b\u0000⸮@!\u0000⸮⸮\u0000\f\u0000\u0000?\u0000⸮rԙ⸮vca7R⸮v\r\rK⸮⸮⸮⸮v⸮z⸮⸮ۨ⸮⸮@qK⸮⸮⸮⸮⸮⸮5<⸮.⸮\\⸮s⸮⸮⸮u+⸮⸮⸮4n⸮⸮⸮n⸮⸮N⸮$⸮⸮⸮bL0櫘VVp⸮⸮+⸮=F^⸮⸮6zaj9⸮⸮KSwT⸮⸮M⸮>v+&&i3J⸮74⸮⸮⸮⸮⸮!jnh⸮V'Ͻ⸮A⸮ZZ⸮⸮2i⸮⸮f⸮⸮c\nu+⸮⸮⸮j1h⸮⸮~h⸮i⸮7`⸮⸮⸮\fњ⸮⸮-@U51zlCK⸮⸮w⸮v!7SC⸮⸮9⸮⸮⸮⸮Tꉰ⸮⸮sO⸮bRf⸮F⸮HZ⸮⸮⸮⸮%\u0000OA⸮\n\\⸮HR⸮vP⸮⸮⸮⸮;4⸮0⸮⸮⸮4⸮⸮*⸮~4⸮⸮\u0000R⸮⸮Hh⸮⸮⸮N⸮⸮:⸮ҁXm%\r⸮⸮)]⸮m4ѩ6⸮i?6⸮⸮JZ⸮⸮⸮y⸮⸮⸮⸮⸮⸮⸮h4 <Q⸮.1CsRP⸮$NisGB⸮⸮(jD⸮⸮3AWQ@⸮⸮sAHZL⸮⸮4⸮⸮⸮⸮⸮IL]⸮⸮⸮CO⸮M⸮2⸮⸮⸮-L⸮⸮⸮7&⸮a\r⸮⸮⸮uP\r⸮&i\u0000⸮⸮⸮4⸮\f⸮i⸮⸮⸮\"g⸮⸮`jl⸮!4v⸮⸮ih`⸮AAE⸮⸮⸮⸮)M\u0000⸮⸮@6⸮⸮0⸮⸮⸮4⸮!⸮⸮n>⸮X.7>⸮⸮⸮!3I@h7⸮JB\nni⸮⸮⸮!)(⸮⸮c'⸮⸮⸮⸮\\f⸮⸮Z(⸮0\n(⸮sE!⸮⸮⸮⸮0⸮⸮ ⸮⸮)⸮⸮;⸮⸮⸮⸮Hu⸮@M&h⸮⸮74Đ⸮⸮⸮+⸮⸮Ͻ%+⸮⸮⸮⸮v&%!⸮⸮⸮⸮f⸮⸮vQG0⸮%%=!⸮!*B⸮⸮S⸮⸮wb⸮~4⸮⸮⸮⸮--\t⸮⸮⸮⸮!⸮)h⸮⸮b⸮⸮R⸮U⸮⸮֏Ƌ⸮%%\u0000?ZN⸮⸮M⸮⸮⸮\\Cy⸮⸮O⸮vҎh⸮7⸮D⸮F)⸮bb⸮⸮w⸮⸮G0⸮%⸮I.i*⸮+\u0000⸮4⸮ ⸮R⸮E⸮:R⸮⸮⸮⸮⸮⸮⸮b⸮Ԏ⸮⸮`(⸮⸮⸮⸮⸮B⸮⸮M⸮Z1E⸮JJ5\u0000&⸮⸮&h⸮p⸮(⸮bJ%⸮q\r⸮⸮>⸮⸮⸮؏⸮⸮j.⸮⸮[⸮~⸮z⸮=mr⸮W⸮H~⸮'⸮K⸮Ѩ⸮⸮⸮i⸮⸮⸮Q⸮iH⸮⸮⸮P⸮⸮K⸮⸮⸮pƗ⸮R⸮⸮⸮Ѩ⸮⸮⸮⸮⸮⸮+⸮4>⸮⸮{ӳ\u0000⸮i;⸮`⸮?OƋA?1⸮OT!⸮⸮⸮?Z51⸮G⸮F⸮⸮o⸮G⸮KP⸮OƓ⸮Q\t⸮⸮~4j11⸮G⸮OQ3{D⸮⸮⸮⸮⸮8⸮-E⸮kRjR⸮⸮⸮A⸮⸮⸮iX⸮ndf⸮MM⸮D\rO⸮⸮)⸮⸮0YisC⸮\u0000⸮Sr⸮⸮⸮⸮s⸮⸮⸮N⸮QE⸮⸮E.mB⸮Q⸮G3\r6⸮⸮f⸮5w'⸮f⸮⸮E⸮a⸮⸮J⸮&}(⸮;⸮⸮⸮XJCN⸮Bf⸮⸮⸮⸮ ⸮⸮⸮⸮Z⸮.⸮⸮⸮⸮⸮⸮4⸮V⸮⸮⸮4j]МӁ4Y⸮Z⸮⸮Qi\u0000⸮8⸮R⸮a⸮⸮9⸮⸮⸮R⸮K⸮⸮⸮⸮⸮⸮O⸮@;⸮^iـS⸮Y⸮⸮sI⸮+0(⸮⸮⸮⸮⸮7⸮\\⸮4⸮Ni⸮z⸮sHsK⸮⸮q⸮⸮9⸮⸮⸮7棚5⸮⸮h⸮1⸮'8⸮⸮}⸮⸮D⸮⸮⸮l⸮6⸮⸮⸮⸮⸮R⸮⸮ri⸮+⸮⸮zt⸮)g#\t⸮\u0000z⸮⸮⸮CN⸮)⸮%⸮4⸮⸮⸮a⸮⸮($.⸮.⸮x⸮ޗ⸮⸮ޗq⸮⸮Ǡ⸮⸮⸮⸮\r⸮K⸮⸮⸮>⸮⸮⸮⸮⸮*⸮⸮⸮⸮⸮⸮ѯpD⸮⸮⸮n⸮4⸮⸮⸮⸮⸮⸮⸮Ϲ⸮܆⸮;⸮ZN}⸮⸮!??=⸮\\N}⸮4{⸮⸮WCri3B⸮p⸮\f⸮SMϹ⸮P⸮BO⸮&O⸮a⸮⸮⸮⸮4{⸮⸮ܟZ⸮<SL⸮⸮⸮⸮K⸮񽹧⸮6W⸮gp⸮95o⸮kO⸮=)⸮⸮⸮d⸮S⸮⸮⸮⸮*⸮⸮⸮ij]⸮f⸮j⸮PP\nQ⸮⸮⸮⸮/4⸮⸮;⸮0⸮h⸮b⸮N⸮\t⸮J m;m=⸮⸮]⸮b⸮rD4⸮@\r⸮O⸮XLSps֢⸮`⸮pj⸮⸮&⸮7⸮⸮⸮j⸮'4⸮⸮@9\"⸮wI⸮T2t⸮⸮$⸮⸮7⸮⸮!⸮*;⸮.⸮i;⸮b⸮M⸮Л⸮ZL⸮⸮⸮O⸮.M!K⸮⸮S⸮⸮(&⸮⸮\f⸮⸮I⸮ßZw4⸮1⸮ӗ⸮L.X⸮\n~c\f⸮]⸮T=@L⸮Z2i⸮⸮9⸮ ⸮&)⸮.&⸮JV⸮HsVi*u⸮i⸮⸮o5\"K⸮⸮Z⸮K#o⸮'5f⸮⸮D⸮c⸮3f⸮P+⸮⸮樢j(⸮⸮⸮ZZC\nwzwiih⸮O^⸮:⸮8⸮⸮⸮⸮S@⸮⸮1F⸮d⸮IH⸮⸮SM⸮4⸮U⸮\r⸮֚jD4⸮MQ#M ⸮>⸮ЇI&4⸮⸮)\bB⸮-E⸮`GIL⸮⸮%+⸮⸮^⸮⸮⸮⸮⸮⸮⸮Ǡ⸮⸮N⸮⸮jH⸮ҿ⸮qN⸮N⸮ ⸮-1)i⸮⸮N⸮⸮6⸮⸮⸮⸮⸮O^⸮⸮CM⸮⸮⸮z\f⸮⸮⸮⸮&⸮M2⸮⸮⸮⸮J#rF⸮⸮L⸮q⸮⸮⸮@⸮S3L⸮⸮2ri(⸮\t⸮⸮K⸮z.M⸮ޝ⸮zWe⸮K⸮zw`(⸮⸮⸮Թ6⸮⸮ޔzWv⸮ԑ⸮⸮sE⸮\\~4⸮;⸮⸮i⸮1⸮⸮G⸮K⸮⸮⸮ZOƑ#o⸮T⸮⸮⸮i?Bci⸮⸮OT?i⸮Ҹ⸮⸮ޚZ⸮⸮\f-⸮u⸮\\6⸮⸮5i⸮H⸮⸮LϽ3Fh⸮⸮⸮⸮⸮⸮⸮⸮`旚⸮X9⸮sN⸮R⸮1⸮\n:⸮T⸮⸮⸮<⸮⸮⸮⸮iE@⸮R⸮⸮E;⸮)?.i⸮⸮\\⸮\b⸮⸮0⸮⸮⸮EP⸮HEHXe6⸮⸮⸮⸮⸮2⸮W0⸮G⸮a⸮⸮QN⸮⸮.'⸮%\u0000⸮⸮R⸮F(⸮f⸮⸮iE⸮@⸮N⸮=J\rH⸮⸮,⸮⸮⸮⸮⸮N⸮⸮⸮⸮⸮q⸮6⸮z⸮R⸮aM⸮C⸮j⸮i⸮⸮@0⸮j⸮⸮⸮⸮⸮('BjeR⸮Ĥ⸮⸮)⸮⸮⸮⸮⸮⸮K⸮⸮-+⸮;⸮⸮⸮⸮iiu\r⸮ihs⸮⸮b⸮⸮2⸮-H⸮/⸮RE1⸮⸮⸮⸮⸮q\r⸮⸮H⸮⸮6⸮⸮4⸮)⸮q⸮⸮@⸮⸮^⸮⸮⸮Ul⸮k@⸮`S6i3K⸮Rһ$)٣Q⸮is@\\\\⸮\r6\bp⸮T⸮⸮֯ڟ⸮⸮⸮\"⸮KR⸮⸮4⸮⸮⸮W\n3R⸮\fd⸮⸮|Tp⸮E0%⸮⸮⸮@⸮⸮\tL⸮W⸮⸮⸮׹,a⸮G⸮1⸮⸮⸮\"⸮nIU⸮\r2⸮⸮%%QAE\u0000M⸮)⸮bQH⸮⸮⸮\u0000/4⸮4⸮1piy⸮$8S⸮⸮⸮6⸮ٓȢ@⸮c4⸮5⸮⸮⸮8G4\n⸮⸮- ⸮jG⸮}⸮s@Xi⸮ӻ⸮⸮4⸮,i⸮M⸮⸮⸮A⸮⸮!⸮I⸮⸮է⸮⸮⸮⸮1(⸮⸮]Ɠ'֟3hϽ+⸮⸮⸮⸮٧⸮1w{⸮⸮ޗ0⸮⸮;uE⸮\u00004⸮⸮⸮⸮(cz⸮⸮Ϛ9⸮⸮⸮}⸮٬⸮⸮⸮ތ⸮⸮⸮⸮⸮⸮N⸮!ozL⸮⸮⸮u&⸮w\t⸮ޗu={⸮3HO⸮⸮⸮s⸮N䉚i>⸮⸮⸮⸮⸮5z⸮⸮7>⸮^^\r\t⸮M⸮.;i3E⸮⸮⸮0jy⸮'4⸮S⸮}jn!Fi⸮Ӹ⸮4⸮hb⸮p⸮H⸮l∉)l⸮4⸮j⸮⸮s@Ť怰⸮~4⸮'4s@⸮⸮j⸮@/4⸮4\\C9⸮⸮Xo⸮HsK⸮⸮⸮m⸮aa⸮⸮dPI[⸮ַ(JJ.⸮⸮%1⸮/4⸮Jw4⸮Ө`/4PP⸮⸮ӆh⸮⸮⸮ju⸮F⸮(*Z⸮⸮E=⸮%⸮4⸮⸮ɥ⸮[⸮⸮⸮⸮\u0000/⸮I⸮⸮;⸮iڒ[⸮⸮S⸮⸮4⸮mF⸮⸮#4Ӛ⸮@⸮Aޡ⸮W())⸮%&h'4\n⸮\\⸮⸮-⸮E\u0000-<C⸮d5+0йf⸮!⸮B⸮⸮I⸮ \n)X.74f⸮\t⸮(\u0000⸮⸮Xw⸮⸮\r⸮6⸮⸮⸮4⸮\r⸮Fj6⸮܋uDj⸮4P2J)⸮0⸮3Fi]⸮⸮S⸮搄⸮Sm⸮⸮\\⸮⸮⸮44奨⸮V⸮⸮⸮M⸮;\"[&⸮⸮⸮4j⸮3Sv\r⸮⸮⸮a⸮⸮*y⸮h%⸮]CA3ElZ⸮⸮qh6⸮WlZ\f⸮⸮⸮⸮,⸮⸮wb⸮\f5Х.⸮dBM35|⸮bRS⸮c$⸮⸮ĥ⸮@)h)⸮⸮Ţ⸮c⸮8P⸮M⸮⸮⸮⸮⸮i⸮E0\nJ⸮\n\r I⸮5\u00004⸮@-% \nJ%⸮⸮H⸮CH⸮i⸮⸮⸮⸮⸮⸮⸮⸮EQOQ\tKKQ⸮/⸮KP\nZlB⸮⸮4⸮QO⸮2⸮⸮⸮⸮R⸮=\bi⸮⸮@i⸮⸮⸮⸮:w⸮⸮~4⸮⸮⸮⸮IH⸮⸮⸮\r⸮i?z⸮RQ⸮⸮G⸮L⸮\r%-F74⸮⸮F⸮M5vT&⸮⸮⸮⸮S⸮C⸮irh⸮d⸮!y⸮⸮.0ɥɤ⸮⸮⸮⸮p⸮!⸮⸮iݪ=⸮⸮j⸮hً⸮9>i2j\u00009⸮&⸮\r⸮4⸮L⸮⸮l\bL⸮41u\f⸮L⸮!2h⸮\t⸮ɤɪ3Mɤ$'4⸮h3⸮⸮⸮V2:J⸮⸮⸮⸮\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000a",
   "temperature": 34.29000092,
   "rfid": ""
}
[... cut ...] 0000\u0000\u0000\u0000a","temperature":34.29000092,"rfid":""}
`
My doubt is how i can retrieve back the original data from server some of the new characters is added for null characters as i saw in the string.
After Sending this data i am getting HTTP post response as :

`HTTP Code : 400`

`HTTP Payload: {"statusCode":400,"code":"FST_ERR_CTP_INVALID_CONTENT_LENGTH","error":"Bad Request","message":"Request body size did not match Content-Length"}`

let me know what is problem for the above issue.

bblanchon, 2022-02-11:
Hi @pradeep22kumar,

`\u0000` is the correct way to encode a NUL in JSON.
Your program is probably sending the wrong value for the `Content-Length` header.
You should use the value returned by [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/) or [measureJson()](https://arduinojson.org/v6/api/json/measurejson/).

Best regards,
Benoit

pradeep22kumar, 2022-02-11:
Hi @bblanchon ,

I will try to decode the data in server  as the same way \u0000.
Function to send the data to server
`
int post_data_to_server(){
   
    memset(json_str,0,sizeof(json_str));
    if (WiFi.status() == WL_CONNECTED)
	{ 
	  //Check WiFi connection status
      Serial.println(serializeJson(stream_root,json_str));
           
	  //Declare object of class HTTPClient
      HTTPClient http;                    
      http.begin(client,"http://pradeep-timestamp.glitch.me/check-emp");      //Specify request destination   "application/json"
      http.addHeader("Content-Type", "application/json");                     //Specify content-type header
      
      int httpCode = http.POST(json_str);                                     //Send the request
      
      //Get the response payload
      String payload = http.getString();                  
      Serial.print("HTTP Code : ");  
      Serial.println(httpCode);   
      Serial.print("HTTP Payload: ");
      Serial.println(payload);    
   
      http.end();  //Close connection
      
      if((payload == "false" && httpCode == 200) || (payload == "true" && httpCode == 200) ){
        Serial.println("values Correct");
        return 1;
      }
      else
       return 0;
      
  }
  else {
        Serial.println("Error in WiFi connection");
  }
  
  return 2;
}
`
When i send the below content to the above function i am not mentioning any header length in **stream_root**  object
`
{
  "uuid": "d4bff49-9a70-4a5f-a787-4018714fb0f1",
  "employee": false,
  "timestamp": "Fri 2022-02-11 08:43:33 GMT",
  "image_data": "No Image Present",
  "temperature": 31.29000092,
  "rfid": ""
}
`
I got the response as
```
HTTP Code : 200
HTTP Payload: true
```
If i send the data with image content only i am getting error as 
`
HTTP Payload: {"statusCode":400,"code":"FST_ERR_CTP_INVALID_CONTENT_LENGTH","error":"Bad Request","message":"Request body size did not match Content-Length"}
`
That's why i am confused that why the content length giving problem here.

bblanchon, 2022-02-17:
Hi @pradeep22kumar,

I don't know what's wrong, but I can tell you that ArduinoJson is not the cause.

I suggest you try to perform the request "manually", without HTTPClient.
You can find an example in the [serialization tutorial](https://arduinojson.org/v6/doc/serialization/) and in #1711.

Best regards,
Benoit

closed 2022-02-17 by bblanchon