bblanchon/ArduinoJson #1705: trying to deserialize unsuccesfully

truecodebender, 2022-01-30:
When trying to de serialize at run time  I get the message "deserializeJson() failed: EmptyInput".

<details open=""><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at run time</li><li>The issue concerns deserialization</li><li><code>deserializeJson()</code> returns <code>EmptyInput</code></li><li>Input comes neither from an HTTP response, nor a file, nor a stream</li></ol></details>

The part of code that deserialization fails is here:

`void callback(char payload[], unsigned int msg_size) {
  Serial.print("Message arrived ");
  for (int i=0;i<msg_size;i++) {
  Serial.print((payload[i]));
  }
  DeserializationError error = deserializeJson(doc, payload, msg_size);
  
  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return;
  }
  
  const char* command = doc["cmd"];
  Serial.print("command ");
  Serial.println(command);

  //rest of code

}
`

The printout from serial ide is here:

`Message arrived {"cmd":"demo","seq":"1"}deserializeJson() failed: EmptyInput`

The array of chars is coming from a I2C connection. Can you provide any debugging hints? Thank you

bblanchon, 2022-01-30:
Hi @truecodebender,

I wonder if there could be a NUL at the beginning of the JSON string.
Can you print `msg_size` to verify that it is equal to 24.

Best regards,
Benoit

truecodebender, 2022-01-31:
Hi Benoit, 
Yes you are right I looked and filtered for nulls and now it deserializeJson successfully! Thank you!

`while (0 < Wire.available()) { // loop through all but the last
    char c = Wire.read(); // receive byte as a character
    if (c == '\0') {
      //ignore 
    } else {
      //char c = Wire.read();
      Serial.print(c);          // print the character
      buff[i] = c;
      buff[i+1] = '\0';         // Add a NULL after each character
      i++;
    }`

closed 2022-01-31 by bblanchon