bblanchon/ArduinoJson #1707: Raspberry Pi Pico RP2040 fail to compile

cotestatnt, 2022-02-04:
**Describe the bug**  
Raspberry Pi Pico RP2040 fail to compile due to error in `pgmspace.hpp`

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
<details><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at compile time</li><li>The error is not in the list</li></ol></details>

**Environment**  
Here is the environment that I used:
* Raspberry Pi Pico RP2040
* Arduino Mbed OS RP2040 and Raspberry Pi Pico/RP2040
* Arduino IDE 1.8.16

**Reproduction**  
Tested with **StringExample.ino** and with others example

**Compiler output**    
```
...src/ArduinoJson/Polyfills/pgmspace.hpp: In function 'uint32_t pgm_read_dword(ArduinoJson6190_F1::pgm_p)':
...src/ArduinoJson/Polyfills/pgmspace.hpp:102:3: error: cannot convert 'ArduinoJson6190_F1::pgm_p' to 'const void*'
  102 |   memcpy_P(&result, p, 4);
```

Forcing the undef of **memcpy_P** before including ArduinoJson.h I can compile and it works as expected, but if I do the same in one of my library which has ArduinoJson as dependency it doesn't work. I have to edit directly the `pgmspace.hpp` (I suppose due to the Arduino compiler).

```
#undef memcpy_P
#include <ArduinoJson.h>
  .....

```

bblanchon, 2022-02-06:
Hi @cotestatnt,

Thank you very much for reporting this bug :+1:

I fixed it in the main branch.
If you don't want to check out the main branch, you can define [ARDUINOJSON_ENABLE_PROGMEM](https://arduinojson.org/v6/api/config/enable_progmem/) to `0`.

Best regards,
Benoit

closed 2022-02-06 by bblanchon

bblanchon, 2022-02-14:
The fix was published in [ArduinoJson 6.19.2](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.19.2)

commit 702f8c2e (2022-02-06): Fix `cannot convert 'pgm_p' to 'const void*'` (fixes #1707)