bblanchon/ArduinoJson #1708: Filter by value

metingul, 2022-02-08:
Hello,
I want to filter result by value (like database query), the sample json content below. for example I want to get userID for ESP8266 (I know userName but I want to fetch userID for that value.)

thank you.

```
[
  {
    "userID": 1,
    "userName": "Arduino",
    "eMail": "arduino@arduinojson.org"
  },
  {
    "userID": 2,
    "userName": "ESP8266",
    "eMail": "esp@arduinojson.org"
  },
  {
    "userID": 3,
    "userName": "ESP32",
    "eMail": "esp32@arduinojson.org"
  }
]
```

metingul, 2022-02-08:
I solved the problem with code below; but may be there is easyway method / function.

```
char* input = "[{\"userID\":1,\"userName\":\"Arduino\",\"eMail\":\"arduino@arduinojson.org\"},{\"userID\":2,\"userName\":\"ESP8266\",\"eMail\":\"esp@arduinojson.org\"},{\"userID\":3,\"userName\":\"ESP32\",\"eMail\":\"esp32@arduinojson.org\"}]";
// size_t inputLength; (optional)

StaticJsonDocument<256> doc;

DeserializationError error = deserializeJson(doc, input);

if (error) {
  Serial.print(F("deserializeJson() failed: "));
  Serial.println(error.f_str());
  return;
}

for (JsonObject item : doc.as<JsonArray>()) {

  int userID = item["userID"]; // 1, 2, 3
  String userName = item["userName"]; // "Arduino", "ESP8266", "ESP32"
  const char* eMail = item["eMail"]; // "arduino@arduinojson.org", "esp@arduinojson.org", ...

  if (userName == "ESP8266") {
  Serial.print("UserID Found:");
  Serial.println(userID);    
  }

}
```

bblanchon, 2022-02-09:
Hi @metingul,

Enumerating the objects in the objects is the way to go.

You could avoid a copy by using a `JsonString` instead of a `String` (requires [ArduinoJson 6.19](https://arduinojson.org/news/2022/01/08/arduinojson-6-19-0/)).

Best regards,
Benoit

closed 2022-02-14 by bblanchon