bblanchon/ArduinoJson #1711: serializeJson with httpClient.POST function

ElcRcp, 2022-02-10:
Hello,
I'm using ESP32-Wrover with Arduino IDE 1.8.19
I used to send JSON messages as shown in examples, which prepare a JSON document and print it to a string then post the string. As example (this is from an old code uses json 5, Im using json 6 now);
  
```
  http.begin(Send_Address);  
  http.addHeader("Content-Type","application/json");
  Jdoc["data"]   =Message;
  Jdoc.printTo(postMessage);
  httpCode = http.POST(postMessage);
  Response = http.getString();
```

Now I'm creating a very large json file which cannot be printed into a String because there is not enough RAM for it and I was thinking if I can use serializeJson function directly with http.POST something like ;
```
  http.begin(Send_Address);  
  http.addHeader("Content-Type","application/json");
  Jdoc["data"]   =Message;
  serializeJson(Jdoc,httpcode=http.POST(Jdoc));  // I know this is wrong but I dont know how to use it
  Response = http.getString();
```
I couldn't find any example refers to this and I don't know if it is possible nor I have any idea how to do workaround this. Can anybody guide me here please?

bblanchon, 2022-02-11:
Hi @ElcRcp,

Unfortunately, HTTPClient doesn't let us do that.
As a workaround, you can use `WiFiClient` directly, as demonstrated in the [Serialization tutorial](https://arduinojson.org/v6/doc/serialization/) (section 4.5.4).

Best regards,
Benoit

ElcRcp, 2022-02-11:
Thanks @bblanchon 
It was very helpful, using and combining the tutorial and a few other examples I managed to send very large json files and keep track of server response. It is a case spesific and simple solution but I'm leaving my code here for future referance.
`
String midString(String str, String start_sig, String finish_sig){
  int locStart = str.indexOf(start_sig);
  if (locStart==-1) {return "";}
  locStart += start_sig.length();
  int locFinish = str.indexOf(finish_sig, locStart);
  if (locFinish==-1) {return "";}
  return str.substring(locStart, locFinish);
}

int postJsonData(int ret_httpcode, String ret_serverResponse, JsonObject inp_json, const char * inp_hostname, const uint16_t inp_port, String inp_endPoint, WiFiClient client)
{
  if(client.connect(inp_hostname,inp_port))
  {
    // Preparing "POST command to endpoint
    String postUrl ="POST ";
    postUrl+=inp_endPoint;
    postUrl+=" HTTP/1.1\r\n";

    // Preparing HTTP headers
    postUrl+="Host: ";
    postUrl+=inp_hostname;
    postUrl+="\r\n";
    postUrl+="Connection: close\r\n";
    postUrl+="Content-Length: ";
    postUrl+=measureJson(inp_json);
    postUrl+="\r\n";
    postUrl+="Content-Type: application/json\r\n\r\n";
    // Sending headers and post together
    client.print(postUrl);
    
    // Sending Json data
    serializeJson(inp_json,client);
    unsigned long timeout = millis();

    // Get Client Response
    while (client.available() == 0){
      if (millis() - timeout > 3000){
        //Serial.println(">>> Client Timeout !");
        client.stop();
        return -2;
      }
    }
    while(client.available()){
      ret_serverResponse+= client.readStringUntil('\r');
    }
    int ret_httpCode=midString(ret_serverResponse, " ", " ").toInt();
    client.stop();
  }
  else
  {
    //Serial.println("Cannot connect to the host");
    return -1;
  }
  return 1;
}
`

bblanchon, 2022-02-12:
I'm glad you could make it work!

If I were you, however, I would not use the `String` class as much, because it tends to fragment the stack a lot.
Instead, I would use the `Print` interface as in the example, most likely with a `WriteBufferingStream`  (see [How to improve (de)serialization speed?](https://arduinojson.org/v6/how-to/improve-speed/)).

Moreover, your program is taking `String` arguments by copy, which is terribly inefficient.
I recommend that you have a look at "The Missing C++" chapter in [Mastering ArduinoJson](https://arduinojson.org/book/), it will give you a better understanding of how this works.
Alternatively, you can contact me on [Codementor](https://www.codementor.io/@bblanchon) so I can review your code and help you improve it.

closed 2022-02-12 by bblanchon

ElcRcp, 2022-02-16:
Thanks again you were very helpful again, I improved the code a bit upon your suggestion. I think it is enough for my usage so I'll leave it there. 

```
int parseHttpCode(char * inp_str){
  char *firstChar = inp_str+strcspn(inp_str," ");
  char *lastChar = firstChar+strcspn(*firstChar ? firstChar + 1 : firstChar, " ");
  char result[sizeof inp_str];
  memcpy (result, firstChar + 1, lastChar - firstChar);
  result[lastChar - firstChar] = '\0';
  return atol(result);
}

int postJsonData(char * ret_serverResponse, JsonObject inp_json, const char * inp_hostname, const uint16_t inp_port, char * inp_endPoint, WiFiClient client)
{
  if(client.connect(inp_hostname,inp_port))
  {
    // Sending "POST command to endpoint
    client.print("POST ");
    client.print(inp_endPoint);
    client.print(" HTTP/1.1\r\n");

    // Sending HTTP headers
    client.print("Host: ");
    client.print(inp_hostname);
    client.print("\r\n");
    client.print("Connection: close\r\n");
    client.print("Content-Length: ");
    client.print(measureJson(inp_json));
    client.print("\r\n");
    client.print("Content-Type: application/json\r\n\r\n");
    // Sending headers and post together
    
    // Sending Json data
    WriteBufferingStream serializeToHttpBuffer(client, 256);
    serializeJson(inp_json,serializeToHttpBuffer);
    serializeToHttpBuffer.flush();
    unsigned long timeout = millis();

    // Get Client Response
    while (client.available() == 0){
      if (millis() - timeout > 3000){
        Serial.println(">>> Client Timeout !");
        client.stop();
        return -2;
      }
    }
    while(client.available()){
      strcat(ret_serverResponse,(client.readStringUntil('\r')).c_str());
    }
    return parseHttpCode(ret_serverResponse);
    client.stop();
  }
  else
  {
    Serial.println("Cannot connect to the host");
    return -1;
  }
}
```

bblanchon, 2022-02-17:
That's much better!
However, you could improve the performance by using the `WriteBufferingStream` to send the headers as well.
In fact, there should be no call to `client.print()` but only to `serializeToHttpBuffer.print()`.

BTW, your program doesn't call `client.stop()` on success.
Again, [contact me](https://www.codementor.io/@bblanchon) if you want a code review.