bblanchon/ArduinoJson #1712: reusing a json document - JsonDocument.clear() not clearing pool

proddy, 2022-02-12:
I'm finding the `JsonDocument.clear()` function doesn't always reset and clear the pool. I've read https://arduinojson.org/v6/how-to/reuse-a-json-document/ and I suspect it's because my `json` object is dangling after each clear() but wanted to report it here anyway since it used to work < 6.19.0. 

**Environment** 

ESP8266, ESP32, PlatformIO, also standalone with G++ and gcc

**Reproduction**  

Copy into wandbox:

```c++
#include <iostream>

#include "ArduinoJson.h"

using namespace std;

int main() { 
  DynamicJsonDocument doc(1024);
  JsonObject json = doc.to<JsonObject>(); // I need the JsonObject to pass to other functions later
    
  for (int i = 1; i < 5; i++) {
      doc.clear();
      printf("Adding object %d:\n", i);
      printf(" Before: ");
      serializeJson(json, cout);
      cout<<endl;
      
      json["name"] = "name";
      json["index"] = i;
      
      printf(" After: ");
      serializeJson(json, cout);
      cout<<endl;
  }
   
}
```

Expected output:

```
Adding object 1:
 Before: {}
 After: {"name":"name","index":1}
Adding object 2:
 Before: {}
 After: {"name":"name","index":2}
Adding object 3:
 Before: {}
 After: {"name":"name","index":3}
Adding object 4:
 Before: {}
 After: {"name":"name","index":4}
```

Actual output:

```
Adding object 1:
 Before: {}
 After: {"name":"name","index":1}
Adding object 2:
 Before: {"name":"name","index":1}
 After: {"name":"name","index":2}
Adding object 3:
 Before: {"name":"name","index":2}
 After: {"name":"name","index":3}
Adding object 4:
 Before: {"name":"name","index":3}
 After: {"name":"name","index":4}
```

(the JsonObject json still has data after the JsonDocument clear() )

bblanchon, 2022-02-13:
Hi @proddy,

This is the expected behavior, and I think the documentation is pretty clear about it:

> every reference acquired before clearing the JsonDocument is invalidated. In other words, you must not use a JsonArray, JsonObject, or JsonVariant created before calling JsonDocument::clear().

If you're surprised (or disappointed) by this behavior, think of [std::vector::clear()](https://www.cplusplus.com/reference/vector/vector/clear/): "All iterators, pointers and references related to this container are invalidated." This is a known issue in C and C++ collections; there is nothing we can do about it.

To fix you program, you just have to move `JsonObject json = doc.to<JsonObject>()` inside the loop.

Best regards,
Benoit

bblanchon, 2022-02-13:
PS: your program gives the same output with 6.19.1, 6.18.5, 6.17.3, 6.16.1, 6.15.2, and 6.14.1:
https://wandbox.org/permlink/mpgRuSKdZTEV8aPH
https://wandbox.org/permlink/ojmV9U9BE7tO459z
https://wandbox.org/permlink/Zcw2gGnEdQfULZZt
https://wandbox.org/permlink/JuuJdN22qd0cSxZx
https://wandbox.org/permlink/o4ybNCyae8iA9k3E
https://wandbox.org/permlink/KMkYBMQ02UHxkitF

proddy, 2022-02-13:
thanks, makes sense.

closed 2022-02-13 by proddy