bblanchon/ArduinoJson #1714: Encoding a StaticJSONDocument to a msgpack fails

beniroquai, 2022-02-24:
**Describe the issue**  
After creating a static json document that summarizes data which needs to be send to a remote server I want to pack it using msgpack. Surprisingly, I found out that ArduinoJson can do it too. Great! However, the encoded printout does not suggest that it works correctly. The code is relatively straight forward and fails in the function `void execute(String server_url, String model_name) `

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  

<details><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at run time</li><li>The issue concerns serialization</li><li>Output contains garbage</li><li>Program doesn’t call <code>deserializeJson()</code></li><li>Program calls <code>serializeJson(const JsonDocument&amp;, ...)</code></li><li>Program uses <code>StaticJsonDocument</code></li><li>Switching to <code>DynamicJsonDocument</code> doesn’t fixes the issue</li><li>Program uses <code>String</code></li><li>Removing <code>.c_str()</code> doesn’t fix the issue</li></ol></details>

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32
* Core/runtime: ESP32 core for Arduino v3.0.2
* IDE: Arduino IDE 1.8.16

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
#include <Arduino.h>
#include <WiFi.h>
#include <WebServer.h>
#include <ArduinoJson.h>
#include <FreeRTOS.h>
#include <MsgPack.h>
#include <HTTPClient.h>
#include <WiFiMulti.h>

const char *SSID = "ATTNVezUEM";
const char *PWD = "t8bfmze5a#id";

// Web server running on port 80
WebServer server(80);

// JSON data buffer
StaticJsonDocument<7000> jsonDocument;
//char buffer[4500];

String server_url = "https://ai.imjoy.io";
String model_name = "cellpose-python";

void connectToWiFi() {
  Serial.print("Connecting to ");
  Serial.println(SSID);

  WiFi.begin(SSID, PWD);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
    // we can even make the ESP32 to sleep
  }

  Serial.print("Connected. IP: ");
  Serial.println(WiFi.localIP());
}

void get_config(String server_url, String model_name) {
  HTTPClient http;
  String subpath = "/public/services/triton-client/get_config?model_name=";
  String serverPath = server_url + subpath + model_name;

  // Your Domain name with URL path or IP address with path
  http.begin(serverPath.c_str());

  // Send HTTP GET request
  int httpResponseCode = http.GET();

  if (httpResponseCode > 0) {
    Serial.print("HTTP Response code: ");
    Serial.println(httpResponseCode);
    String payload = http.getString();
    //Serial.println(payload);
    deserializeJson(jsonDocument, payload);
    serializeJsonPretty(jsonDocument, Serial);
  }
  else {
    Serial.print("Error code: ");
    Serial.println(httpResponseCode);
  }
  // Free resources
  http.end();
}

void execute(String server_url, String model_name) {
  /*
    Execute a model on the trition server.
    The supported kwargs are consistent with pyotritonclient
    https://github.com/oeway/pyotritonclient/blob/bc655a20fabc4611bbf3c12fb15439c8fc8ee9f5/pyotritonclient/__init__.py#L40-L50
  */

  int H = 25;
  int W = 25;
  uint16_t current_frame[H][W] = { 0 };

  jsonDocument.clear();
  jsonDocument["_rtype"] = "ndarray";
  jsonDocument["_rdtype"] = "float32";
  jsonDocument["model_name"] =model_name;
  JsonArray _rshape = jsonDocument.createNestedArray("_rshape");
  _rshape.add(1);
  _rshape.add(W);
  _rshape.add(H);

  serializeJsonPretty(jsonDocument, Serial);

  // Encode the arguments as msgpack
  serializeMsgPack(jsonDocument, Serial);    

    /*Next Step: Send msgpack 
    HTTPClient http;
    
    // Your Domain name with URL path or IP address with path
    String subpath = "/public/services/triton-client/execute";
    String serverPath = server_url + subpath + model_name;

    // Your Domain name with URL path or IP address with path
    http.begin(serverPath.c_str());
    http.addHeader("Content-Type", "application/msgpack");
*/
}

  

void setup() {
  Serial.begin(115200);

  connectToWiFi();

  get_config(server_url, model_name);
  execute(server_url, model_name);
}

void loop() {
  server.handleClient();
}
```

**Program output**  

Actual output:

```
Connecting to ATTNVezUEM
......Connected. IP: 192.168.1.252
Error code: -11
{
  "_rtype": "ndarray",
  "_rdtype": "float32",
  "model_name": "cellpose-python",
  "_rshape": [
    1,
    25,
    25
  ]
}⸮⸮_rtype⸮ndarray⸮_rdtype⸮float32⸮model_name⸮cellpose-python⸮_rshape⸮
```

bblanchon, 2022-02-25:
Hi @beniroquai,

> the encoded printout does not suggest that it works correctly

What do you mean? 
`⸮⸮_rtype⸮ndarray⸮_rdtype⸮float32⸮model_name⸮cellpose-python⸮_rshape⸮���` looks good to me.

Best regards,
Benoit

beniroquai, 2022-02-25:
Hey @bblanchon, I was expecting something more `Hex`-ic. :D 
So I could go on and try sending this to a remote server over HTTP and decode it there? 

Thanks!

bblanchon, 2022-02-28:
Hi @beniroquai,

You can send MessagePack the same way you would do for JSON.
Please follow the instructions from the [serialization tutorial](https://arduinojson.org/v6/doc/serialization/); it shows how to perform an HTTP POST on pages 125+.

Best regards,
Benoit

beniroquai, 2022-03-01:
Thanks a lot! I follow your tutorials with greatest interest! 
Amazing how much effort you put into this! Plus one for open-source projects! Thanks!! 

One more question: Do you have a guess what'S the best way to "pack" a 2D array (camera frame) into the `msgpack`? I read something about buffer [here](https://github.com/bblanchon/ArduinoJson/issues/1704). Not sure if this is the right solution?

beniroquai, 2022-03-01:
Quick update with (untested) code:

```cpp

  int WIDTH = 1600;
  int HEIGHT = 1200;
  camera_fb_t *frame_buffer = esp_camera_fb_get();
  if (!frame_buffer)
    return false;

  // compute the required size
  const size_t CAPACITY = JSON_ARRAY_SIZE(WIDTH*HEIGT);

  // allocate the memory for the document
  StaticJsonDocument<CAPACITY> doc;

  // create an empty array
  JsonArray frame = doc.to<JsonArray>();

  uint16_t current_frame[WIDTH * HEIGHT] = { 0 };

  // down-sample image in blocks
  for (uint32_t i = 0; i <  WIDTH * HEIGHT; i++) {
    const uint8_t pixel = frame_buffer->buf[i];
    // average pixels in block (accumulate)
    current_frame[i] += pixel;
  }

  copyArray(current_frame, frame);

  serializeJsonPretty(jsonDocument, Serial);

  // Encode the arguments as msgpack
  serializeMsgPack(jsonDocument, Serial);
```

Do you think this goes in the right direction?

closed 2022-03-23 by bblanchon