bblanchon/ArduinoJson #1716: Esp8266 (D1 Mini) Will stop functioning correct after a few ArduinoJson operation. Need to erase the flash to work again

hungyip, 2022-02-27:
**Describe the issue**  
I use my esp8266 to control some LED strip. I use ArduinoJson to serialize and deserialize Json to be able to communicate with the board and also be able to use the Json to save settings into the board (by LittleFS).

I found that after some settings read and settings write, the program will stop working. the AP of the board will no longer work and I need to erase the flash and reprogram to make it work again. My Json buffer size is 500 and use Static memory. The program tend to crash faster if I increase the buffer size to 1000. It can happen as soon as the first time the program read the config file and deserialize the config if i set the buffer size to 1000. Have tried to troubleshoot for more than a day but have no clue. Not likely a hardware issue as I have tried different boards and it give me same problem.

Sometimes restarting the board will give me No Memory error upon deserializeJson(), but it doesn't always happen. I am sure the Json is well below the buffer size.

**Troubleshooter report**  
<details><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at run time</li><li>The issue concerns deserialization</li><li>The program doesn’t check the error</li><li>The program crashes</li><li>Program crashes after calling <code>deserializeJson()</code></li><li>Program calls neither <code>strcmp()</code> nor <code>strcpy()</code></li><li>Replacing pointer with a <code>String</code> doesn’t solve the issue</li></ol></details>

**Environment**  
<details><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at run time</li><li>The issue concerns deserialization</li><li><code>deserializeJson()</code> returns <code>NoMemory</code></li><li>Increasing the capacity of the <code>JsonDocument</code> doesn’t fixes the issue</li><li>The <code>JsonDocument</code> is a <code>StaticJsonDocument</code></li><li>Changing the Assistant settings doesn’t solve the issue</li></ol></details>

bblanchon, 2022-02-28:
Hi @hungyip,

It's difficult to tell without looking at your code, but the symptoms could suggest a stack overflow.
Can you try with a `DynamicJsonDocument`?

Also, make sure that the core for ESP8266 is up to date.

Best regards,
Benoit

hungyip, 2022-02-28:
I spent some more time debugging the problem. 

Seems like my json_buffer_size is too small. From arduinojson assistant it states that 384 is enough. I made it 500 to be safe. Turns out 500 will give me NoMemory problem but 600 would not. 

Seems like it's working now. 1000 will crash (not sure why but I will investigate more when I have time, my program is quite small in size and thought 1000 would do fine). 

Thank you so much for the help and I will report back after testing for two more days.

Regards,

Raymond

hungyip, 2022-03-23:
I am still working to understand the problem.

Just one question here. In my code I use uint32_t to store color values, and these color value can be big (0xFFFFFFFF). Using ArduinoJson to deserialize the data, does it matter if I use #define ARDUINOJSON_USE_LONG_LONG 1 or not?

bblanchon, 2022-03-28:
[ARDUINOJSON_USE_LONG_LONG](https://arduinojson.org/v6/api/config/use_long_long/) is for 64-bit integers and it's set to `1` by default since 6.19.

closed 2022-04-13 by bblanchon