bblanchon/ArduinoJson #1717: Define instance methods for custom converters

MatiasMFM2001, 2022-02-28:
**Is your feature request related to a problem? Please describe.**  
About [existing solution](https://arduinojson.org/news/2021/05/04/version-6-18-0/), I don't like we have to define "free" functions inside the `ARDUINOJSON_NAMESPACE`

**Describe the solution you'd like**  
I suggest a more OOP way to achieve this: Create interfaces, so our classes will implement its methods
```cpp
// Alternative names: JSONable, Jsonable, Json-able
class ConvertableToJson {
    public:
        bool toJson(JsonVariant dst) = 0;
};
```
```cpp
class ConvertableFromJson {
    public:
        void fromJson(VariantConstRef src) = 0;  // This is a setter
        bool checkJson(VariantConstRef src) = 0;
};
```
We will now have a `JsonVariant::set(const ConvertableToJson& value)` overload. And second interface will be used by `JsonVariant::is<T>()`, `JsonVariant::as<T>()`, `JsonVariant::to<T>()` methods

While I was writing this, I realised `ConvertableFromJson` doesn't make sense. It requires our class to be default-constructible, or use an existing instance and create new ones from it (`fromJson()` would now return `T`)

**Describe alternatives you've considered**  
Existing solution doesn't have the problems `ConvertableFromJson` has. We should keep using that

**Additional context**  
Goodbye!

bblanchon, 2022-02-28:
Hi @MatiasMFM2001,

This feature requires virtual functions, which are forbidden by the design constraints for ArduinoJson 6.

The good news is that you can implement it in your application without altering the library.

Here is a draft:

```c++
class ConvertableToJson {
 public:
  virtual void toJson(JsonVariant dst) const = 0;
};

void convertToJson(const ConvertableToJson& src, JsonVariant dst) {
  src.toJson(dst);
}
```

And:

```c++
class ConvertableFromJson {
 public:
   virtual void fromJson(JsonVariantConst src) = 0;
   virtual bool checkJson(JsonVariantConst) const { return true; }
};

void convertFromJson(JsonVariantConst src, ConvertableFromJson& dst) {
  dst.fromJson(src);
}

bool canConvertFromJson(JsonVariantConst src, const ConvertableFromJson& dst) {
  return dst.checkJson(src);
}
```

Best regards,
Benoit

MatiasMFM2001, 2022-02-28:
> This feature requires virtual functions, which are forbidden by the design constraints for ArduinoJson 6.

That's true! I forgot about `virtual` keyboard

> The good news is that you can implement it in your application without altering the library.
> Here is a draft: [Code]

Awesome, and thank you! That brings the best of both solutions
I'll close this issue for now. I can re-open it later if I have some trouble

Goodbye!

closed 2022-02-28 by MatiasMFM2001