bblanchon/ArduinoJson #1718: serializeMsgPack is seemingly outputting larger message than serializeJson

andreasqlight, 2022-02-28:
**Describe the issue**  
I'm trying to shave off some size of the data I'm trying to send over MQTT and I see that ArduinoJson has msgpack support. When pasting in the message: `{"s":"0","m":{"0":[260,0,0,227,0,0,0,0],"1":[33,0,0,0,0,0,0,0],"2":[24,0,0,0,0,0,0,0],"3":[0,0,0,0,0,0,0,0],"4":[0,0,0,0,0,0,0,0],"5":[0,0,0,0,0,0,0,0],"6":[0,0,0,0,0,0,0]}}` on the msgpack tester page, I get a return of 96 bytes when it does its conversion, which is a 50% reduction. 

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
Doesn't seem like the troubleshooter have any options to select for msgpack? 

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32
* Core/runtime: Arduinoespressif32 1.0.6
* IDE: VSCODE 1.64.2

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
        DynamicJsonDocument doc(1024);       
        
        doc["s"] = "0";
        doc["m"]["0"][0] = meterData[0].lastReading;
        doc["m"]["0"][1] = meterData[0].powerDraw;
        doc["m"]["0"][2] = meterData[0].current;
        doc["m"]["0"][3] = meterData[0].voltage;       
        doc["m"]["0"][4] = meterData[0].current_l2;            
        doc["m"]["0"][5] = meterData[0].current_l3;    
        doc["m"]["0"][6] = meterData[0].voltage_l2;
        doc["m"]["0"][7] = meterData[0].voltage_l3;
        doc["m"]["1"][0] = meterData[1].lastReading;
        doc["m"]["1"][1] = meterData[1].powerDraw;
        doc["m"]["1"][2] = meterData[1].current;
        doc["m"]["1"][3] = meterData[1].voltage;
        doc["m"]["1"][4] = meterData[1].current_l2;            
        doc["m"]["1"][5] = meterData[1].current_l3;    
        doc["m"]["1"][6] = meterData[1].voltage_l2;
        doc["m"]["1"][7] = meterData[1].voltage_l3;
        doc["m"]["2"][0] = meterData[2].lastReading;
        doc["m"]["2"][1] = meterData[2].powerDraw;
        doc["m"]["2"][2] = meterData[2].current;
        doc["m"]["2"][3] = meterData[2].voltage;    
        doc["m"]["2"][4] = meterData[2].current_l2;            
        doc["m"]["2"][5] = meterData[2].current_l3;    
        doc["m"]["2"][6] = meterData[2].voltage_l2;
        doc["m"]["2"][7] = meterData[2].voltage_l3;
        doc["m"]["3"][0] = meterData[3].lastReading;
        doc["m"]["3"][1] = meterData[3].powerDraw;
        doc["m"]["3"][2] = meterData[3].current;
        doc["m"]["3"][3] = meterData[3].voltage;      
        doc["m"]["3"][4] = meterData[3].current_l2;            
        doc["m"]["3"][5] = meterData[3].current_l3;    
        doc["m"]["3"][6] = meterData[3].voltage_l2;
        doc["m"]["3"][7] = meterData[3].voltage_l3;
        doc["m"]["4"][0] = meterData[4].lastReading;
        doc["m"]["4"][1] = meterData[4].powerDraw;
        doc["m"]["4"][2] = meterData[4].current;
        doc["m"]["4"][3] = meterData[4].voltage;     
        doc["m"]["4"][4] = meterData[4].current_l2;            
        doc["m"]["4"][5] = meterData[4].current_l3;    
        doc["m"]["4"][6] = meterData[4].voltage_l2;
        doc["m"]["4"][7] = meterData[4].voltage_l3;
        doc["m"]["5"][0] = meterData[5].lastReading;
        doc["m"]["5"][1] = meterData[5].powerDraw;
        doc["m"]["5"][2] = meterData[5].current;
        doc["m"]["5"][3] = meterData[5].voltage;      
        doc["m"]["5"][4] = meterData[5].current_l2;            
        doc["m"]["5"][5] = meterData[5].current_l3;    
        doc["m"]["5"][6] = meterData[5].voltage_l2;
        doc["m"]["5"][7] = meterData[5].voltage_l3;
        doc["m"]["6"][0] = meterData[6].lastReading;
        doc["m"]["6"][1] = meterData[6].powerDraw;
        doc["m"]["6"][2] = meterData[6].current;
        doc["m"]["6"][3] = meterData[6].voltage;     
        doc["m"]["6"][4] = meterData[6].current_l2;            
        doc["m"]["6"][5] = meterData[6].current_l3;    
        doc["m"]["6"][6] = meterData[6].voltage_l2;
        doc["m"]["6"][7] = meterData[6].voltage_l3;
        doc["m"]["7"][0] = meterData[7].lastReading;
        doc["m"]["7"][1] = meterData[7].powerDraw;
        doc["m"]["7"][2] = meterData[7].current;
        doc["m"]["7"][3] = meterData[7].voltage;
        doc["m"]["7"][4] = meterData[7].current_l2;            
        doc["m"]["7"][5] = meterData[7].current_l3;    
        doc["m"]["7"][6] = meterData[7].voltage_l2;
        doc["m"]["7"][7] = meterData[7].voltage_l3;
        

        char buffer[1024];
        int bytesWrittenJson = serializeJson(doc, buffer);        
        
        Serial.print("Bytes written json: ");
        Serial.println(bytesWrittenJson);

        char data[1024];
        int bytesWritten = serializeMsgPack(doc, data);

        Serial.print("Bytes written msg pack: ");
        Serial.println(bytesWritten);   
        
        for(int i = 0; i<bytesWritten; i++){
            Serial.printf("%02X ",data[i]);
        }
        Serial.println("");

        Serial.print("Buffer json: ");
        Serial.println(buffer);
        Serial.print("Buffer msgpack: ");
        Serial.println(data);
```

**Program output**  

Output:

```
Bytes written json: 179
Bytes written msg pack: 304
82 A1 73 A1 30 A1 6D 87 A1 30 98 CA 43 82 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 43 63 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 A1 31 98 CA 42 04 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 A1 32 98 CA 41 C0 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 A1 33 98 CA 43 34 00 00 CA 00 00 00 00 CA 
00 00 00 00 CA 43 63 66 66 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 A1 34 98 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 A1 35 98 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 A1 36 97 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 00 00 CA 00 00 
00 00
Buffer json: {"s":"0","m":{"0":[260,0,0,227,0,0,0,0],"1":[33,0,0,0,0,0,0,0],"2":[24,0,0,0,0,0,0,0],"3":[180,0,0,227.4,0,0,0,0],"4":[0,0,0,0,0,0,0,0],"5":[0,0,0,0,0,0,0,0],"6":[0,0,0,0,0,0,0]}}
Buffer msgpack: ��s�0�m��0��C�
```

Maybe I'm reading this wrong, as it should be an array of HEX values, thus the size output from serializeMsgPack does not completely correspond to the actual size. However, when I send it over asyncMQTTclient as a char array, on the other end the printed size is the same. Am I doing something wrong to get a bigger msgPack than the serialized JSON?

If you copy the printout of the output and put it into a msgPack to JSON converter, it is valid msgPack, but if you use the JSON to msgPack tool on msgpack.org the hex array is seemingly a lot smaller.

bblanchon, 2022-02-28:
Hi @andreasqlight,

Thank you very much for reporting this issue 👍 

In the output, I can see: `CA 00 00 00 00`, which is `0.0f` in MsgPack.
As you can see, it takes 5 bytes instead of 2 bytes in JSON (for `0,`).

I think this behavior is incorrect.
ArduinoJson should see that the value is integral and emit a simple `00` instead, even if it means changing the value type.
After all, it already converts floats to integers when it writes `0` instead of `0.0`, so it would be consistent with the JSON serialization.

Hopefully, this change won't break any existing code.

Best regards,
Benoit

closed 2022-03-01 by bblanchon

bblanchon, 2022-03-08:
The fix was published in [ArduinoJson 6.19.3](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.19.3).

commit e3e375f5 (2022-02-28): MessagePack: serialize round floats as integers (fixes #1718)