bblanchon/ArduinoJson #1720: [Docs] An URL in a v6 page points to a v5 one

MatiasMFM2001, 2022-03-01:
**Describe the bug**  
Webpage "[How to reduce memory usage?](https://arduinojson.org/v6/issues/incomplete-output/)", section "What if my microcontroller is too small?", item "How to reduce memory usage?". The URL brings me to [this page](https://arduinojson.org/v5/faq/how-to-reduce-memory-usage/), instead of [this one](https://arduinojson.org/v6/how-to/reduce-memory-usage/)

I don't know if there are more URLs affected - I've just spotted that one. Would it be possible to scan them all with some automated tool?

**Environment**  
Here is the environment that I used:
* OS: GNU/Linux, Debian 10
* CPU Arch: x86_64
* Browser 1: Firefox 91.6.0esr 64-bit
* Browser 2: Chrome 98.0.4758.102 64-bit

**Reproduction**
Just click the URL

Goodbye!

bblanchon, 2022-03-01:
Hi @MatiasMFM2001,

Thank you very much for reporting this problem.
The page should be fixed now.

Best regards,
Benoit

closed 2022-03-01 by bblanchon

MatiasMFM2001, 2022-03-01:
You're welcome! Yup, It's fixed